lwfinger/rtl8723bu #1: Wi-Fi promiscuous mode

arthur-eberhardt, 2015-02-11:
Hello,

I would like to get all packets from the Wi-Fi interface with a RT8723BU dongle. 
I'm able to create a virtual interface in monitor mode with this command:
iw phy phy0 interface add moni0 type monitor

Then, I'm launching a python script that uses libpcap to capture all Wi-Fi packet and not only packet with my MAC address. But, I'm not able to get those packets from this interface. Do you know if this driver is able to receive those packets?

Thanks for your help,

Arthur

arthur-eberhardt, 2015-02-11:
I have more information on this issue. I'm not able to switch the main interface to monitor mode. When I type this command:

sudo iwconfig wlan1 mode monitor

I get this:

Error for wireless request "Set Mode" (8B06) :
    SET failed on device wlan1 ; Invalid argument.

That's weird because when I type "iw phy", it says that monitor mode is supported. Do you know if monitor mode is supported by the chipset or driver?

Thank you

lwfinger, 2015-02-11:
I just pushed changes to build with CONFIG_IOCTL_CFG80211 enabled. Try setting up monitor mode with 'iw' commands.

arthur-eberhardt, 2015-02-12:
Thank you, I just compiled and load the module but it still doesn't work.

When I type "iw phy", I get this:

Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
		 * P2P-client
		 * P2P-GO
	software interface modes (can always be added):
		 * monitor
	interface combinations are not supported
	Supported commands:
		 * new_interface
		 * set_interface
		 * new_key
		 * new_beacon
		 * new_station
		 * set_bss
		 * join_ibss
		 * set_pmksa
		 * del_pmksa
		 * flush_pmksa
		 * remain_on_channel
		 * action
		 * set_channel
		 * connect
		 * disconnect

It seems that monitor mode is supported, but when I type "sudo iw --debug dev wlan1 set monitor none", I get this:

-- Debug: Sent Message:
--------------------------   BEGIN NETLINK MESSAGE ---------------------------
  [HEADER] 16 octets
    .nlmsg_len = 44
    .nlmsg_type = 26 <0x1a>
    .nlmsg_flags = 5 <REQUEST,ACK>
    .nlmsg_seq = 1423727805
    .nlmsg_pid = 13617
  [PAYLOAD] 28 octets
    06 00 00 00 08 00 03 00 13 00 00 00 08 00 05 00 06 00 ..................
    00 00 08 00 17 00 04 00 00 00                         ..........
---------------------------  END NETLINK MESSAGE   ---------------------------
-- Debug: Received Message:
--------------------------   BEGIN NETLINK MESSAGE ---------------------------
  [HEADER] 16 octets
    .nlmsg_len = 64
    .nlmsg_type = 2 <ERROR>
    .nlmsg_flags = 0 <>
    .nlmsg_seq = 1423727805
    .nlmsg_pid = 13617
  [ERRORMSG] 20 octets
    .error = -95 "Operation not supported"
  [ORIGINAL MESSAGE] 16 octets
    .nlmsg_len = 16
    .nlmsg_type = 26 <0x1a>
    .nlmsg_flags = 5 <REQUEST,ACK>
    .nlmsg_seq = 1423727805
    .nlmsg_pid = 13617
---------------------------  END NETLINK MESSAGE   ---------------------------
command failed: Operation not supported (-95)

Does this really mean that monitor mode is not supported or is this just some misconfiguration on my side?

psanford, 2015-06-08:
I'm also interested in getting monitor mode working on this hardware. I adding a monitor interface to the phy as per [these instructions][1] but that also did not work.

Do you have any suggestions on how I might debug this further?

[1]: https://sandilands.info/sgordon/capturing-wifi-in-monitor-mode-with-iw

lwfinger, 2015-06-08:
A little more info about what fails would be helpful.
What happens when you issue the following command?

sudo iw phy phy0 interface add mon0 type monitor

psanford, 2015-06-08:
Here's the steps I've been following:

```
# insmod 8723bu.ko
```

dmsg output:
```
RTL871X: module init start
RTL871X: rtl8723bu v4.3.6.11_12942.20141204_BTCOEX20140507-4E40
RTL871X: rtl8723bu BT-Coex version = BTCOEX20140507-4E40
RTL871X:
usb_endpoint_descriptor(0):
RTL871X: bLength=7
RTL871X: bDescriptorType=5
RTL871X: bEndpointAddress=84
RTL871X: wMaxPacketSize=512
RTL871X: bInterval=0
RTL871X: RT_usb_endpoint_is_bulk_in = 4
RTL871X:
usb_endpoint_descriptor(1):
RTL871X: bLength=7
RTL871X: bDescriptorType=5
RTL871X: bEndpointAddress=5
RTL871X: wMaxPacketSize=512
RTL871X: bInterval=0
RTL871X: RT_usb_endpoint_is_bulk_out = 5
RTL871X:
usb_endpoint_descriptor(2):
RTL871X: bLength=7
RTL871X: bDescriptorType=5
RTL871X: bEndpointAddress=6
RTL871X: wMaxPacketSize=512
RTL871X: bInterval=0
RTL871X: RT_usb_endpoint_is_bulk_out = 6
RTL871X:
usb_endpoint_descriptor(3):
RTL871X: bLength=7
RTL871X: bDescriptorType=5
RTL871X: bEndpointAddress=87
RTL871X: wMaxPacketSize=64
RTL871X: bInterval=3
RTL871X: RT_usb_endpoint_is_int_in = 7, Interval = 3
RTL871X:
usb_endpoint_descriptor(4):
RTL871X: bLength=7
RTL871X: bDescriptorType=5
RTL871X: bEndpointAddress=8
RTL871X: wMaxPacketSize=512
RTL871X: bInterval=0
RTL871X: RT_usb_endpoint_is_bulk_out = 8
RTL871X:
usb_endpoint_descriptor(5):
RTL871X: bLength=7
RTL871X: bDescriptorType=5
RTL871X: bEndpointAddress=9
RTL871X: wMaxPacketSize=512
RTL871X: bInterval=0
RTL871X: RT_usb_endpoint_is_bulk_out = 9
RTL871X: nr_endpoint=6, in_num=2, out_num=4

RTL871X: USB_SPEED_HIGH
RTL871X: CHIP TYPE: RTL8723BU
RTL871X: register rtw_netdev_ops to netdev_ops
RTL871X: rtw_wdev_alloc(padapter=f08cc000)
RTL871X: Chip Version Info: CHIP_8723B_Normal_Chip_TSMC_D_CUT_1T1R_RomVer(0)
RTL871X: RF_Type is 3!!
RTL871X: _ConfigChipOutEP_8723 OutEpQueueSel(0x07), OutEpNumber(4)
RTL871X: EEPROM type is E-FUSE
RTL871X: ====> _ReadAdapterInfo8723BU
RTL871X: Boot from EFUSE, Autoload OK !
RTL871X: hal_EfuseSwitchToBank: Efuse switch bank to 0
RTL871X: hal_ReadEFuse_WiFi: data end at address=0x82
RTL871X: Efuse Realmap:

29 81 03 7C 01 08 21 00 40 07 05 35 10 00 00 00
26 28 28 28 28 28 28 28 28 28 28 02 FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF 29 2A 2A 2B 2B 2B
2B 2B 2C 2C 2C 02 FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF 20 1C 1F 00 00 00 FF FF
FF 28 20 11 00 00 00 FF 00 FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
DA 0B 20 B7 E7 47 03 54 E4 BD E7 76 B4 09 03 52
65 61 6C 74 65 6B 16 03 38 30 32 2E 31 31 6E 20
57 4C 41 4E 20 41 64 61 70 74 65 72 00 FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF 0F FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF FF
RTL871X: EEPROM VID = 0x bda
RTL871X: EEPROM PID = 0xb720
RTL871X: Hal_EfuseParseBTCoexistInfo_8723B: Enable BT-coex, ant_num=1
RTL871X: InitAdapterVariablesByPROM_8723BU(): REPLACEMENT = 0
RTL871X: <==== _ReadAdapterInfo8723BU in 350 ms
RTL871X: init_channel_set ChannelPlan ID 20 Chan num:13
RTL871X: Init_ODM_ComInfo_8723b(): fab_ver=0 cut_ver=0
RTL871X: can't get autopm:
RTL871X: rtw_macaddr_cfg MAC Address  = 54:e4:bd:e7:76:b4
RTL871X: bDriverStopped:1, bSurpriseRemoved:0, bup:0, hw_init_completed:0
RTL871X: rtw_ndev_init(wlan0)
RTL871X: _rtw_drv_register_netdev, MAC Address (if1) = 54:e4:bd:e7:76:b4
usbcore: registered new interface driver rtl8723bu
RTL871X: module init ret=0

```

```
# iw phy phy0 interface add mon0 type monitor
# echo $?
0
```

dmsg output:
```
RTL871X: cfg80211_rtw_add_virtual_intf(wlan0) wiphy:phy0, name:mon0, type:6
RTL871X: cfg80211_rtw_add_virtual_intf(wlan0) ndev:c7cd4800, ret:0
```

```
# ifconfig mon0 up
```

dmsg output:
```
RTL871X: rtw_cfg80211_monitor_if_open
```

```
# tcpdump -i mon0 -n
tcpdump: WARNING: mon0: no IPv4 address assigned
tcpdump: verbose output suppressed, use -v or -vv for full protocol decode
listening on mon0, link-type IEEE802_11_RADIO (802.11 plus radiotap header), capture size 65535 bytes
```

dmsg output:
```
device mon0 entered promiscuous mode
``` 

The tcpdump doesn't see any packets (I'm in an environment with a lot of active clients so it should definitely see traffic).

shineyear, 2015-11-23:
i have the same issue , anyone can tell me how to fix it?

Encr1pt3d, 2017-04-30:
Hi friends, I hae the same problem:

**# ifconfig** 
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.202.128  netmask 255.255.255.0  broadcast 192.168.202.255
        inet6 fe80::20c:29ff:fe15:efe9  prefixlen 64  scopeid 0x20<link>
        ether 00:0c:29:15:ef:e9  txqueuelen 1000  (Ethernet)
        RX packets 1039  bytes 1057099 (1.0 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 126  bytes 9662 (9.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 22  bytes 1270 (1.2 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 22  bytes 1270 (1.2 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

**# iwconfig**
lo        no wireless extensions.

eth0      no wireless extensions.

wlan0     IEEE 802.11  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:on

**# lsusb** 
Bus 002 Device 002: ID 13b1:0039 Linksys AE1200 802.11bgn Wireless Adapter [Broadcom BCM43235]
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 001 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

# iw dev
phy#0
	Interface wlan0
		ifindex 3
		wdev 0x1
		addr c0:c1:c0:6d:61:22
		type managed
		channel 1 (2412 MHz), width: 20 MHz, center1: 2412 MHz

# iw phy phy0 info
Wiphy phy0
	max # scan SSIDs: 10
	max scan IEs length: 2048 bytes
	max # sched scan SSIDs: 0
	max # match sets: 0
	max # scan plans: 1
	max scan plan interval: -1
	max scan plan iterations: 0
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Device supports roaming.
	--- Supported Ciphers:---
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP-128 (00-0f-ac:4)
	Available Antennas: TX 0 RX 0
	---Supported interface modes:--- **

jurek4321, 2020-05-11:
I created a fork with monitor mode support https://github.com/jurek4321/rtl8723bu-monitor-mode