lwfinger/rtl8723bu #2: RTL8723BU BT driver

anikol, 2015-02-27:
Thank you very much for this driver. But is there any known Bluetooth driver for RTL8723BU in existence?

lwfinger, 2015-02-27:
I do not know. What USB ID does lsusb show?

anikol, 2015-02-27:
0bda:b720 Realtek Semiconductor Corp. 
I also tried a BT driver from rtl8723au_bt repository. It makes the module recognized but fails to bring the interface up.

sudo hciconfig hci1 -a
hci1:	Type: BR/EDR  Bus: USB
	BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
	DOWN 
	RX bytes:0 acl:0 sco:0 events:0 errors:0
	TX bytes:3 acl:0 sco:0 commands:1 errors:0

sudo hciconfig hci1 up
Can't init device hci1: Input/output error (5)

I am running Ubuntu 14.04 x32

uname -r
3.13.0-46-generic

lwfinger, 2015-02-28:
I added that USB ID to the tables. Do a fresh 'git pull', and switch to the kernel branch with 'git checkout kernel'. Then make and 'sudo make install'. Finally, remove any existing drivers with 'sudo modprobe -rv btusb', 'sudo modprobe -rv rtk_btusb', and 'sudo modprobe -v btusb'. Finally, post any dmesg output that mentions btusb.

anikol, 2015-03-02:
It still fails to bring the interface up.
sudo hciconfig hci0 up
Can't init device hci0: Input/output error (5)

Here is dmesg after checking out and compiling the kernel branch:
[    7.048543] rtk_btusb: module verification failed: signature and/or  required key missing - tainting kernel
[    7.048643] rtk_btusb: Realtek Bluetooth USB driver ver 0.8
[    7.048657] rtk_btusb: btusb_probe 1==========can_wakeup=1	 flag2=0
[    7.048659] rtk_btusb: wakeup_disable==========can_wakeup=1	 flag2=0
[    7.048659] rtk_btusb: patch_add
[    7.048899] rtk_btusb: btusb_open start pm_usage_cnt(0x1)
[    7.048900] rtk_btusb: btusb_open hdev->promisc ==0
[    7.048901] rtk_btusb: download_patch start
[    7.048911] usbcore: registered new interface driver rtk_btusb
[    7.049857] rtk_btusb: check_fw_version : read_ver_rsp->lmp_subver = 0x8723
[    7.049858] rtk_btusb: Rtk patch end 1
[    7.049864] rtk_btusb: btusb_open end  pm_usage_cnt(0x0)
[   17.061204] rtk_btusb: btusb_flush add delay 
[   17.071144] rtk_btusb: btusb_close

Also it stops seeing the adapter if I remove rtk_btusb by 'sudo modprobe -rv rtk_btusb'.

lwfinger, 2015-03-02:
I have reverted that change. I did not have much time when I added it, and failed to look carefully enough. In addition, I thought that change was only applied to the kernel branch.

bomellberg, 2015-05-25:
Is the rtl8723au_bt driver valid for 2.6.33? I'm asking because I can't seem to get scanning to work. The driver loads fine and brings up hci0

~ # hciconfig -a
hci0:   Type: USB
        BD Address: 5C:F3:70:17:E4:1B ACL MTU: 820:8 SCO MTU: 255:16
        UP RUNNING PSCAN ISCAN
        RX bytes:415 acl:0 sco:0 events:22 errors:0
        TX bytes:578 acl:0 sco:0 commands:21 errors:0
        Features: 0xff 0xff 0xff 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        Link policy: RSWITCH HOLD SNIFF PARK
        Link mode: SLAVE ACCEPT
        Name: 'LUCAS3 (0)'
        Class: 0x3e0100
        Service Classes: Networking, Rendering, Capturing
        Device Class: Computer, Uncategorized
        HCI Ver:  (0x6) HCI Rev: 0xb LMP Ver:  (0x6) LMP Subver: 0x8723
        Manufacturer: not assigned (93)

Scanning doesn't find anything. Also, the device is not visible to any other BT equipment, despite PSCAN ISCAN.

lwfinger, 2015-05-25:
I have no RTL8723BU hardware; however, as far as I know, the driver works on all USB-based BT hardware. There is a different driver for SDIO hardware (RTL8723BS).

dp4qb, 2015-11-06:
Hello.
Thank you for the drivers, but I've got an issue and I hope you can hint me where I'm wrong. 
I've tried to compile this module on Raspberry Pi2 (ARMv7 architecture), and I can't make BT work -- hcitool does not see it. WiFi works, though.
I've thought it might be that _this_ driver does not provide BT functions (although some lines in Makefile made me think it does)? I've followed your posts above and compiled rtl8723au_bt, but no WiFi/BT interface are visible with it.

Kernel 4.1.7+v7.
Sorry for a dumb questions here, but I'm not very familiar with kernel modules.

Thank you :)

lwfinger, 2015-11-07:
What Realtek BT device do you have?

dp4qb, 2015-11-07:
Totally forgot about that.
0bda:b720 Realtek Semiconductor Corp.

lwfinger, 2015-11-07:
The kernel branch of http://github.com/lwfinger/rtl8723au_bt.git will now work for the BT part of RTL8723BU. I just pushed a fix.

dp4qb, 2015-11-08:
Thank you.
Unfortunately, got these errors:
    
    make -C /lib/modules/4.1.7-v7+/build M=/home/pi/rtl8723au_bt modules
    make[1]: Entering directory '/usr/src/linux-headers-4.1.7-v7+'
      CC [M]  /home/pi/rtl8723au_bt/btusb.o
      CC [M]  /home/pi/rtl8723au_bt/btrtl.o
    /home/pi/rtl8723au_bt/btrtl.c:343:5: error: redefinition of ‘btrtl_setup_realtek’
    In file included from /home/pi/rtl8723au_bt/btrtl.c:26:0:
    /home/pi/rtl8723au_bt/btrtl.h:47:19: note: previous definition of ‘btrtl_setup_realtek’ was here
    scripts/Makefile.build:264: recipe for target '/home/pi/rtl8723au_bt/btrtl.o' failed
    make[2]: *** [/home/pi/rtl8723au_bt/btrtl.o] Error 1
    Makefile:1384: recipe for target '_module_/home/pi/rtl8723au_bt' failed
    make[1]: *** [_module_/home/pi/rtl8723au_bt] Error 2
    make[1]: Leaving directory '/usr/src/linux-headers-4.1.7-v7+'
    Makefile:24: recipe for target 'all' failed
    make: *** [all] Error 2

I've installed the latest Raspbian (Jessie), installed gcc-4.7, downloaded headers from this folder: http://www.niksula.hut.fi/~mhiienka/Rpi/linux-headers-rpi/

lwfinger, 2015-11-08:
I think this is now fixed, but I cannot test with my current kernel configuration.

Please pull again and retry.

NickNamerr, 2015-11-09:
I am using branch rtl8723bu and kernel version 4.2.0-17,
lsusb:
Bus 002 Device 008: ID 0bda:b720 Realtek Semiconductor Corp.

WiFi is working well,
BT interface does not appearing in system:
~/rtl8723bu$ hcitool dev
Devices:

lwfinger, 2015-11-09:
There is no BT driver here. You need to use the kernel branch of http://github.com/lwfinger/rtl8723au_bt.git.

dp4qb, 2015-11-11:
Thank you, now it's installing correctly.
Although I can't make it scan or appear visible. hcitool, hciconfig are showing the adapter as working, but it alwasy returns no scan results. That's odd, but I couldn't make it scan in Windows also. Maybe my usb dongle is broken.

lwfinger, 2015-11-11:
My dongle scans and connects just like any other device. It also works with Windows 10 using the driver built into that OS. I did not need the mini-CD that came with the device.

LorenzoBianconi, 2015-11-26:
Hi Larry,

I have a RTL8723BU usb device. WiFi device works fine but BT one is not able to identify other nearby BT devices (using hcitool) despite that it is recognized by the system. I used master branch. Any suggestions?
Thanks

Lorenzo

bomellberg, 2015-11-26:
Lorenzo,

We have had this issue as well. Try turning the WiFi interface on first, before scanning for nearby Bluetooth devices. It solved our problem.

/Bo

LorenzoBianconi, 2015-11-26:
Hi Bo,

I tried to turn on/off WiFi interface but still same issue. which device are you using?
Thx
Lorenzo

bomellberg, 2015-11-27:
We have the LM811 from LM-Technologies. It's based on RTL8723BU.

http://lm-technologies.com/pcat/modules/bluetooth-and-wifi-modules/

I need to run the wpa_supplicant which starts the wifi driver and gets my wlan0-interface up, before I can trust the bluetooth scanning results.

LorenzoBianconi, 2015-11-27:
I am using the same chip and wpa_supplicant is running as well since I am connected to a wpa2 AP with that chip. Are you using hcitool to perform BT scan? Are you running the master branch at latest commit?
Thx

Lorenzo

bomellberg, 2015-11-27:
Yes, latest commit. This is a session:

hcid[123]: Bluetooth HCI daemon
hdev=0
hcid[123]: HCI dev 0 up
hcid[123]: Starting security manager 0
Starting sdpd
OBEX File Transfer service registered
~ # hcitool scan
Scanning ...
        D8:FC:93:12:9A:BB       MELLBB1-L1
        08:ED:B9:DC:FD:D8       n/a

Make sure you have an antenna connected.

rvsrvs, 2016-03-07:
I have the same problem Lorenzo has above.  Wifi runs great but Bluetooth seems to fail silently

I'm running the Dragon WiFi/Bluetooth dongle shown here: http://www.amazon.com/gp/product/B00X9SSUXO which is based on the RealTek RTL8723BU chipset

uname -a reports:
Linux raspberrypi 4.1.18-v7+ #846 SMP Thu Feb 25 14:22:53 GMT 2016 armv7l GNU/Linux

I built and installed the master branch of: https://github.com/lwfinger/rtl8723bu with no errors or warnings
I built and installed the kernel branch of:  http://github.com/lwfinger/rtl8723au_bt.git with no errors or warnings

On boot I see the following in /var/log/messages:

Mar  7 20:17:10 raspberrypi kernel: [    6.338175] RTL871X: module init start
Mar  7 20:17:10 raspberrypi kernel: [    6.338204] RTL871X: rtl8723bu v4.3.6.11_12942.20141204_BTCOEX20140507-4E40
Mar  7 20:17:10 raspberrypi kernel: [    6.338212] RTL871X: rtl8723bu BT-Coex version = BTCOEX20140507-4E40
Mar  7 20:17:10 raspberrypi kernel: [    6.496968] RTL871X: rtw_ndev_init(wlan0)
Mar  7 20:17:10 raspberrypi kernel: [    6.498601] usbcore: registered new interface driver rtl8723bu
Mar  7 20:17:10 raspberrypi kernel: [    6.498617] RTL871X: module init ret=0
Mar  7 20:17:10 raspberrypi kernel: [   12.303766] RTL871X: RTW_ADAPTIVITY_EN_AUTO, chplan:0x20, Regulation:0,0
Mar  7 20:17:10 raspberrypi kernel: [   12.303797] RTL871X: RTW_ADAPTIVITY_MODE_NORMAL
Mar  7 20:17:10 raspberrypi kernel: [   12.949213] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Mar  7 20:17:10 raspberrypi kernel: [   14.956691] RTL871X: nolinked power save enter
Mar  7 20:17:11 raspberrypi kernel: [   15.984491] RTL871X: RTW_ADAPTIVITY_EN_AUTO, chplan:0x20, Regulation:0,0
Mar  7 20:17:11 raspberrypi kernel: [   15.984539] RTL871X: RTW_ADAPTIVITY_MODE_NORMAL
Mar  7 20:17:11 raspberrypi kernel: [   16.368986] RTL871X: nolinked power save leave
Mar  7 20:17:13 raspberrypi kernel: [   18.432298] RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
Mar  7 20:17:13 raspberrypi kernel: [   18.462749] RTL871X: start auth
Mar  7 20:17:13 raspberrypi kernel: [   18.467759] RTL871X: auth success, start assoc
Mar  7 20:17:13 raspberrypi kernel: [   18.472249] RTL871X: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
Mar  7 20:17:13 raspberrypi kernel: [   18.472283] RTL871X: assoc success
Mar  7 20:17:13 raspberrypi kernel: [   18.472406] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
Mar  7 20:17:13 raspberrypi kernel: [   18.505914] RTL871X: send eapol packet
Mar  7 20:17:13 raspberrypi kernel: [   18.514633] RTL871X: send eapol packet
Mar  7 20:17:13 raspberrypi kernel: [   18.514942] RTL871X: set pairwise key camid:4, addr:04:a1:51:aa:32:5a, kid:0, type:AES
Mar  7 20:17:13 raspberrypi kernel: [   18.521511] RTL871X: set group key camid:5, addr:04:a1:51:aa:32:5a, kid:2, type:AES

Which is odd bc I would think I should not see RTL871X but RTL872X

ifconfig reports: 

wlan0     Link encap:Ethernet  HWaddr 7c:c7:09:67:71:d6  
          inet addr:192.168.5.9  Bcast:192.168.5.255  Mask:255.255.255.0
          inet6 addr: fe80::8c7a:d93a:374b:87b4/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:35 errors:0 dropped:199 overruns:0 frame:0
          TX packets:29 errors:0 dropped:2 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:6291 (6.1 KiB)  TX bytes:5628 (5.4 KiB)

and I am able to ssh to the device on that IP address, so WiFi is functional

lsusb reports:
Bus 001 Device 008: ID 0bda:b720 Realtek Semiconductor Corp. 

hciconfig -a reports:
hci0:	Type: BR/EDR  Bus: USB
	BD Address: 7C:C7:08:67:71:D6  ACL MTU: 820:8  SCO MTU: 255:16
	UP RUNNING 
	RX bytes:750 acl:0 sco:0 events:54 errors:0
	TX bytes:641 acl:0 sco:0 commands:54 errors:0
	Features: 0xff 0xff 0xff 0xfe 0xdb 0xff 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF PARK 
	Link mode: SLAVE ACCEPT 
	Name: 'RTK_BT_4.0'
	Class: 0x000000
	Service Classes: Unspecified
	Device Class: Miscellaneous, 
	HCI Version: 4.0 (0x6)  Revision: 0xb
	LMP Version: 4.0 (0x6)  Subversion: 0x8723
	Manufacturer: Realtek Semiconductor Corporation (93)

'sudo hcitool lescan' reports:
Set scan parameters failed: Operation not permitted

so I do:
sudo hciconfig hci0 down
sudo hciconfig hcio up
sudo service bluetooth restart
sudo service dbus restart

and now 'sudo hcitool lescan' reports:

LE Scan ...

But never finds anything... There is definitely a nearby iBeacon because if I now unplug the realtek dongle and plug in another USB bluetooth dongle based on the broadcom 20702A1 chipset and say:

sudo hcitool lescan

I get:
LE Scan ...
C0:C4:AB:1C:AF:12 (unknown)
5B:0E:27:ED:68:CF (unknown)
5B:0E:27:ED:68:CF (unknown)
6F:C6:2C:55:72:CA (unknown)
6F:C6:2C:55:72:CA (unknown)
F7:37:18:B7:BB:85 (unknown)
F7:37:18:B7:BB:85 BKON
D3:E1:91:92:C9:3F (unknown)
D2:05:3A:D6:F1:EC (unknown)

Any thoughts on what I'm doing wrong?

ben423423n32j14e, 2016-07-27:
I have the same problem as everyone else it would seem with the RTL8723BU, the bluetooth device is found by hcitool but cannot see any devices if I run hcitool scan, anyone found a fix for this?

nagpai, 2016-09-23:
i successfully installed driver for RTL8723BU. however if i upgrade kernel wifi goes off.. How do i fix this?

lwfinger, 2016-09-23:
Every out-of-kernel driver needs to be reinstalled when the kernel changes. You cannot escate that truth!

nagpai, 2016-09-24:
sorry, my bad. new to this.

ferbar, 2016-10-27:
For everyone using the _kernel_ branch of rtl8723au_bt with Linux kernel 4.1 without any scan results: It seems that the firmware upload is disabled by default for kernel 4.1, as well as there is the usb ID missing for our device. See https://github.com/lwfinger/rtl8723au_bt/issues/54 for a patch.