lwfinger/rtl8723bu #3: Wifi bandwith drops when Bluetooth is used to stream sound in A2DP

arthur-eberhardt, 2015-03-05:
Dear lwfinger,

I have a question about the Wifi/BT coexistence performance.

My use case: Stream a song from a music streaming service by Wifi and play it on a Bluetooth speaker

I used iperf to check the Wifi bandwith:

When there is no sound streamed to the Bluetooth speaker: 1,5 Mbit/s
When there is sound streamed to the Bluetooth speaker: 300kbit/s

Is there anything possible to optimize that?

Thanks for your help

lwfinger, 2015-03-05:
Maybe.

arthur-eberhardt, 2015-03-06:
How close are the RTL8723BU and RTL8723BE chipset?

Can I use the rtlwifi_new driver with RTL8723BU?

Thanks

lwfinger, 2015-03-06:
One is USB (BU), and the other is PCI (BE). They are very different beasts. For the RTL8723BU, you need the driver at http://github.com/lwfinger/rtl8723bu.git. Note that I do not have that hardware, and that the driver is provided as a convenience.