lwfinger/rtl8723bu #5: Drivers restart in loop

arthur-eberhardt, 2015-05-04:
Hi,

The driver constantly restarts in loop when I'm trying to connect to an AP with wpa_supplicant.conf.

The command I typed is:

wpa_supplicant -iwlan0 -c /etc/wpa_supplicant.conf

I put the dmesg output below, do you have any idea where this problem should come from?

thanks for your help

```
[ 3052.613991] RTL871X: module init start
[ 3052.623965] RTL871X: rtl8723bu v4.3.6.11_12942.20141204_BTCOEX20140507-4E40
[ 3052.636404] RTL871X: rtl8723bu BT-Coex version = BTCOEX20140507-4E40
[ 3052.646964] RTL871X:
[ 3052.646981] usb_endpoint_descriptor(0):
[ 3052.655378] RTL871X: bLength=7
[ 3052.661211] RTL871X: bDescriptorType=5
[ 3052.667595] RTL871X: bEndpointAddress=84
[ 3052.674511] RTL871X: wMaxPacketSize=512
[ 3052.680585] RTL871X: bInterval=0
[ 3052.687408] RTL871X: RT_usb_endpoint_is_bulk_in = 4
[ 3052.696234] RTL871X:
[ 3052.696249] usb_endpoint_descriptor(1):
[ 3052.704538] RTL871X: bLength=7
[ 3052.710398] RTL871X: bDescriptorType=5
[ 3052.716719] RTL871X: bEndpointAddress=5
[ 3052.723536] RTL871X: wMaxPacketSize=512
[ 3052.729625] RTL871X: bInterval=0
[ 3052.736555] RTL871X: RT_usb_endpoint_is_bulk_out = 5
[ 3052.745509] RTL871X:
[ 3052.745522] usb_endpoint_descriptor(2):
[ 3052.753816] RTL871X: bLength=7
[ 3052.759614] RTL871X: bDescriptorType=5
[ 3052.765910] RTL871X: bEndpointAddress=6
[ 3052.772722] RTL871X: wMaxPacketSize=512
[ 3052.778794] RTL871X: bInterval=0
[ 3052.785699] RTL871X: RT_usb_endpoint_is_bulk_out = 6
[ 3052.794587] RTL871X:
[ 3052.794599] usb_endpoint_descriptor(3):
[ 3052.802897] RTL871X: bLength=7
[ 3052.808708] RTL871X: bDescriptorType=5
[ 3052.815091] RTL871X: bEndpointAddress=87
[ 3052.821890] RTL871X: wMaxPacketSize=64
[ 3052.827578] RTL871X: bInterval=3
[ 3052.835955] RTL871X: RT_usb_endpoint_is_int_in = 7, Interval = 3
[ 3052.865846] RTL871X:
[ 3052.865858] usb_endpoint_descriptor(4):
[ 3052.873812] RTL871X: bLength=7
[ 3052.879370] RTL871X: bDescriptorType=5
[ 3052.885660] RTL871X: bEndpointAddress=8
[ 3052.892796] RTL871X: wMaxPacketSize=512
[ 3052.898904] RTL871X: bInterval=0
[ 3052.905867] RTL871X: RT_usb_endpoint_is_bulk_out = 8
[ 3052.914761] RTL871X:
[ 3052.914774] usb_endpoint_descriptor(5):
[ 3052.923080] RTL871X: bLength=7
[ 3052.928932] RTL871X: bDescriptorType=5
[ 3052.935227] RTL871X: bEndpointAddress=9
[ 3052.942077] RTL871X: wMaxPacketSize=512
[ 3052.948140] RTL871X: bInterval=0
[ 3052.955044] RTL871X: RT_usb_endpoint_is_bulk_out = 9
[ 3052.964555] RTL871X: nr_endpoint=6, in_num=2, out_num=4
[ 3052.964574]
[ 3052.973885] RTL871X: USB_SPEED_HIGH
[ 3052.980498] RTL871X: CHIP TYPE: RTL8723BU
[ 3052.988796] RTL871X: register rtw_netdev_ops to netdev_ops
[ 3052.998626] RTL871X: rtw_wdev_alloc(padapter=d0ddf000)
[... 992 lines cut ...]
[ 3115.657794] RTL871X: -rtl8723bu_c2h_packet_handler()
[ 3115.666950] RTL871X: SetHwRegWithBuf8723B len=9
[ 3115.674897] RTL871X: SetHwRegWithBuf8723B len=9
[ 3115.683125] RTL871X: SetHwRegWithBuf8723B len=9
[ 3115.693514] RTL871X: SetHwRegWithBuf8723B len=9
[ 3117.009623] RTL871X: IsBtDisabled=0, IsBtControlLps=0
[ 3119.009620] RTL871X: IsBtDisabled=0, IsBtControlLps=0
[ 3119.027033] RTL871X: rx C2H_PACKET, ID=9, seq=10, len=7
[ 3119.036852] RTL871X: +rtl8723bu_c2h_packet_handler() length=9
[ 3119.046466] RTL871X: -rtl8723bu_c2h_packet_handler()
[ 3119.056905] RTL871X: ==>rtw_ps_processor .fw_state(8)
[ 3119.066622] RTL871X: ==>ips_enter cnts:7
[ 3119.080502] RTL871X: nolinked power save enter
[ 3119.089377] RTL871X: ===> rtw_ips_pwr_down...................
[ 3119.098614] RTL871X: ====> rtw_ips_dev_unload...
[ 3119.109638] RTL871X: usb_read_port_cancel
[ 3119.122045] RTL871X: usb_read_port_complete()-1119: RX Warning! bDriverStopped(0) OR bSurpriseRemoved(0)
[ 3119.140297] RTL871X: usb_read_port_complete()-1119: RX Warning! bDriverStopped(0) OR bSurpriseRemoved(0)
[ 3119.158425] RTL871X: usb_read_port_complete()-1119: RX Warning! bDriverStopped(0) OR bSurpriseRemoved(0)
[ 3119.176556] RTL871X: usb_read_port_complete()-1119: RX Warning! bDriverStopped(0) OR bSurpriseRemoved(0)
[ 3119.194797] RTL871X: usb_read_port_complete()-1119: RX Warning! bDriverStopped(0) OR bSurpriseRemoved(0)
[ 3119.212925] RTL871X: usb_read_port_complete()-1119: RX Warning! bDriverStopped(0) OR bSurpriseRemoved(0)
[ 3119.231053] RTL871X: usb_read_port_complete()-1119: RX Warning! bDriverStopped(0) OR bSurpriseRemoved(0)
[ 3119.249322] RTL871X: usb_read_port_complete()-1119: RX Warning! bDriverStopped(0) OR bSurpriseRemoved(0)
```

lwfinger, 2015-05-04:
I do not have any device with the RTL8723BU chip, thus all I can provide are general suggestions. The problem seems to start when the device initiates power-saving mode, thus you could try loading the module with "rtw_ips_mode=0" as an option. After the initial problem, your user-space code disables the wireless interface. I see no real reason for that, but then I do not even know what distro you are using.

arthur-eberhardt, 2015-05-05:
Thanks, it solved my issue !

Is it possible to change the loglevel of this driver?

closed 2015-05-05 by arthur-eberhardt

lwfinger, 2015-05-05:
You have the source and the device. Patches are welcome.

tossy0917, 2015-05-18:
You can commented out "#define CONFIG_DEBUG" in include/autoconf.h

arthur-eberhardt, 2015-05-24:
Thanks tossy0917 ! I just compiled the driver and it's now quiet !