lwfinger/rtl8723bu #6: how to build on beagle bone black

mogar, 2015-07-06:
Hi,

I'm trying to build this on a BBB. I have cloned the repo, but trying to make it gives me an error:

    root@beaglebone:~#git clone https://github.com/lwfinger/rtl8723bu.git
    root@beaglebone:~#cd rtl8723bu
    root@beaglebone:~/rtl8723bu# make
    make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/3.14.46-ti-r70/build M=/root/rtl8723bu modules
    make[1]: Entering directory `/lib/modules/3.14.46-ti-r70/build'
    make[1]: *** No rule to make target `modules'.  Stop.
    make[1]: Leaving directory `/lib/modules/3.14.46-ti-r70/build'
    make: *** [modules] Error 2
    root@beaglebone:~/rtl8723bu#

What is the proper way to build this?

lwfinger, 2015-07-09:
You are missing the kernel headers.

mogar, 2015-07-09:
Thanks for your advice! I installed the headers.

I then had to change the SUBARCH option in the Makefile to arm in order for it to find the correct directory in the headers.

I then tried to make it, but got the following error:

```
  CC [M]  /root/rtl8723bu/os_dep/ioctl_cfg80211.o
/root/rtl8723bu/os_dep/ioctl_cfg80211.c: In function 'rtw_cfg80211_ibss_indicate_connect':
/root/rtl8723bu/os_dep/ioctl_cfg80211.c:725:9: warning: passing argument 3 of 'cfg80211_ibss_joined' makes integer from pointer without a cast [enabled by default]
include/net/cfg80211.h:3900:6: note: expected 'gfp_t' but argument is of type 'struct ieee80211_channel *'
/root/rtl8723bu/os_dep/ioctl_cfg80211.c:725:9: error: too many arguments to function 'cfg80211_ibss_joined'
include/net/cfg80211.h:3900:6: note: declared here
make[2]: *** [/root/rtl8723bu/os_dep/ioctl_cfg80211.o] Error 1
make[1]: *** [_module_/root/rtl8723bu] Error 2
make[1]: Leaving directory `/usr/src/linux-headers-3.14.47-ti-r71'
make: *** [modules] Error 2
```

lwfinger, 2015-07-10:
Sorry it took so long to get to this problem. I think there was an error in the logic that should now be fixed. Please repull.

mogar, 2015-07-11:
Looks like that worked. Thanks for your help!

closed 2015-07-11 by mogar