lwfinger/rtl8723bu #7: backtrace dump when joining WPA with rtl8723bu

purplehorace, 2015-07-15:
Hello Larry
I think the work you have done to supply the ported driver is fantastic but I am having some issues when joining a WPA network.
My system is using a 3.14.47 kernel configured to give all the appropriate drivers to support the rtl8723bu.  I have built the driver and it brings up the card beautifully and I can scan, detect without problems.  Connecting to an unprotected network is working well but when connecting to a protected network there is a stack dump

[   52.664277] ------------[ cut here ]------------
[   52.669379] WARNING: CPU: 0 PID: 668 at net/wireless/sme.c:655 __cfg80211_connect_result+0x348/0x3a4 [cfg80211]()
[   52.680149] Modules linked in: btusb(O) 8723bu(O) bluetooth cfg80211 inv_mpu6050 c_can_platform c_can industrialio_triggered_buffer kfifo_buf can_dev industrialio option 6lowpan_iphc usb_wwan
[   52.698080] CPU: 0 PID: 668 Comm: kworker/u2:2 Tainted: G           O 3.14.47-dirty #4
[   52.706391] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[   52.712513] Backtrace:
[   52.715087] [<c001731c>] (dump_backtrace) from [<c00174b8>] (show_stack+0x18/0x1c)
[   52.723069]  r6:0000028f r5:00000009 r4:00000000 r3:c06f854c
[   52.729032] [<c00174a0>] (show_stack) from [<c04b10b0>] (dump_stack+0x20/0x28)
[   52.736607] [<c04b1090>] (dump_stack) from [<c0042790>] (warn_slowpath_common+0x6c/0x8c)
[   52.745092] [<c0042724>] (warn_slowpath_common) from [<c00427d4>] (warn_slowpath_null+0x24/0x2c)
[   52.754295]  r8:00000001 r7:ddb80000 r6:00000000 r5:00000000 r4:ddc8be00
[   52.761442] [<c00427b0>] (warn_slowpath_null) from [<bf0dd6c8>] (__cfg80211_connect_result+0x348/0x3a4 [cfg80211])
[   52.772498] [<bf0dd380>] (__cfg80211_connect_result [cfg80211]) from [<bf0c1bbc>] (cfg80211_process_wdev_events+0x158/0x15c [cfg80211])
[   52.785303]  r10:00100100 r9:ddc8ec0c r8:00200200 r7:ddc8be60 r6:ddc8be20 r5:ddc8be00
[   52.793530]  r4:ddc8ec00
[   52.796221] [<bf0c1a64>] (cfg80211_process_wdev_events [cfg80211]) from [<bf0c1c00>] (cfg80211_process_rdev_events+0x40/0x8c [cfg80211])
[   52.809067]  r10:00000000 r9:ddae2000 r8:00000000 r7:dd870900 r6:dd86ca00 r5:ddc61c38
[   52.817289]  r4:ddc8be00
[   52.819974] [<bf0c1bc0>] (cfg80211_process_rdev_events [cfg80211]) from [<bf0bec4c>] (cfg80211_event_work+0x1c/0x24 [cfg80211])
[   52.831998]  r5:ddc61cb8 r4:ddc61cb8
[   52.835772] [<bf0bec30>] (cfg80211_event_work [cfg80211]) from [<c0055544>] (process_one_work+0xfc/0x338)
[   52.845796]  r4:ddac8080 r3:bf0bec30
[   52.849556] [<c0055448>] (process_one_work) from [<c00558d8>] (worker_thread+0x11c/0x364)
[   52.858122]  r10:c0721cdf r9:dd86ca00 r8:00000001 r7:ddae2000 r6:ddac8098 r5:dd86ca10
[   52.866378]  r4:ddac8080
[   52.869086] [<c00557bc>] (worker_thread) from [<c005b4dc>] (kthread+0xc4/0xe0)
[   52.876645]  r10:00000000 r9:00000000 r8:00000000 r7:c00557bc r6:ddac8080 r5:ddacd600
[   52.884908]  r4:00000000
[   52.887583] [<c005b418>] (kthread) from [<c00142a0>] (ret_from_fork+0x14/0x34)
[   52.895141]  r7:00000000 r6:00000000 r5:c005b418 r4:ddacd600
[   52.901089] ---[ end trace b49df6b01bd39533 ]---

The issue seems to be around the area of bss in sme.c in the kernel
	if (WARN_ON(!bss))
		return;

And I think it may be linked to a warning that is generated when compiling the rtl8723bu driver
/board-support/rtl8723bu-master/os_dep/ioctl_linux.c: In function translate_scan:
/board-support/rtl8723bu-master/os_dep/ioctl_linux.c:798:1: warning: the frame size of 1160 bytes is larger than 1024 bytes [-Wframe-larger-than=]

That function is relying on the stack to return the bss structure instead of allocating memory and returning the pointer.

Has this issue been seen on any of the other drivers, maybe the pci version ?

Any help to track this issue down is appreciated.

Once the backtrace has happened I am connected and can obtain an ip address, and I can see the mac is registered on the access point so there is network just the "knowledge" of the SSID and connection is lost due to the error.

Kind Regards
Marc

lwfinger, 2015-07-16:
That warning is not the problem, but I will just pushed a patch to fix it.

Too bad your oops message does not yield any symbolic traceback info. That makes debugging a lot easier.

If I ever knew, I have forgotten. What is your hardware with the RTL8723BU in it?

purplehorace, 2015-07-16:
Thanks for the update but the compile warnings are still there and there is still an oops when running.

I have managed to track it down to a potential issue in the is_bss function in net/wireless/scan.c 
When it is called the bssid MAC is populated and matched but the ssid is blank and the ssid_len is 0.  The check for ssid is for NULL and not blank so it passes but there should also be a check for ssid_len > 0.

I have applied the check 
if ( (!ssid) || (!ssid_len) )

and everything is happy, no warnings and life is happy.

I will suggest a patch to kernel.org and see what they think.

The hardware we are using the RTL8723BU with is a BeagleBone Black.

lwfinger, 2015-07-17:
That sounds like a good fix. Send the patch to linux-wireless@vger.kernel.org. That is the proper place for wireless and mac80211 fixes.