lwfinger/rtl8723bu #10: Compiling on Arch fails

slin31, 2015-11-01:
I tried to build this on Arch Linux.
>uname -r
>4.2.5-1-ARCH

But when I do `make` I get this output:  
>make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.2.5-1-ARCH/build M=/mnt/data/Documents/Source/Kernel module test/rtl8723bu-master  modules  
make[1]: Entering directory '/usr/lib/modules/4.2.5-1-ARCH/build'  
Makefile:659: Cannot use CONFIG_CC_STACKPROTECTOR_STRONG: -fstack-protector-strong not supported by compiler  
make[1]: *** No rule to make target 'module'.  Stop.  
make[1]: Leaving directory '/usr/lib/modules/4.2.5-1-ARCH/build'  
Makefile:355: recipe for target 'modules' failed  
make: *** [modules] Error 2  
  
I admit I am not an expert on compiling kernel modules, this is actually the first time I have to do this.  
So I apologize if I am missing something obvious here but after googling and trying different things for a few hours I am still clueless to what is going wrong here.  
  
I do have base-devel as well as the linux-headers installed.
The strange thing is that I tried compiling it on my main computer while setting up my laptop (which needs this module) and somehow I got it to work.  
During the laptop install I noticed that it installed a newer kernel than the one I compiled with on my main PC so I updated that PC. Later all attempts to compile it failed and the update seems to be the only thing that changed.  
Thanks for any potential clues on how to solve this.

slin31, 2015-11-01:
Well I certainly am not an expert on this.  
I just got this to work by changing the directory from which I was doing it. Apparently spaces in the folder names are a no no for this?  
It compiled. Sorry :)

closed 2015-11-01 by slin31

lwfinger, 2015-11-01:
Folders are a Windows term. It is best not to use special characters in Linux directories as they need the directory name to be surrounded by quotation marks or the special characters need to be escaped.

If you create this source directory by the recommended 'git clone' command, you will get a usable directory structure. BTW, I hope you are using git, and not downloading a zip file. There will be a number of important fixes submitted within the next couple of days.