lwfinger/rtl8723bu #11: error: ‘critical_count’ undeclared

NickNamerr, 2015-11-06:
realtek/rtl8723bu/include/osdep_service_linux.h:191:6: error: ‘critical_count’ undeclared (first use in this function)
  if (critical_count) {
      ^
realtek/rtl8723bu/include/osdep_service_linux.h:191:6: note: each undeclared identifier is reported only once for each function it appears in
realtek/rtl8723bu/core/rtw_cmd.c: In function ‘_rtw_dequeue_cmd’:
/realtek/rtl8723bu/core/rtw_cmd.c:242:7: error: ‘critical_count’ undeclared (first use in this function)
  if (!critical_count) {
       ^

make[1]: Leaving directory '/usr/src/linux-headers-4.2.0-17-generic'

Tried to add 
int critical_count;
int critical_count_bh;
into 'include/osdep_service_linux.h'
received a lot of :
rtl8723bu/core/rtw_bt_mp.o:(.bss+0x4): multiple definition of `critical_count'
or 
rtl8723bu/core/rtw_bt_mp.o:(.bss+0x0): multiple definition of `critical_count_bh'

lwfinger, 2015-11-06:
Sorry, that repo is in a bad state. I will make a fix and push it in a few hours.

lwfinger, 2015-11-06:
It is fixed now.

NickNamerr, 2015-11-08:
Thank you!! Will check it tomorrow)