lwfinger/rtl8723bu #17: rtl8723bu in odroidxu3

suat1479, 2016-01-11:
Hi,
I try to install driver in odroid xu3 and i got these error message

"odroid@odroid:~$ cd rtl8723bu
odroid@odroid:~/rtl8723bu$ make
make: Warning: File `Makefile' has modification time 9.6e+07 s in the future
make ARCH=arm CROSS_COMPILE= -C /lib/modules/3.10.69+/build M=/home/odroid/rtl87                   23bu  modules
make: *** /lib/modules/3.10.69+/build: No such file or directory.  Stop.
make: *** [modules] Error 2
odroid@odroid:~/rtl8723bu$ sudo make install
[sudo] password for odroid:
make: Warning: File `Makefile' has modification time 9.6e+07 s in the future
install -p -m 644 8723bu.ko  /lib/modules/3.10.69+/kernel/drivers/net/wireless/
install: cannot stat ‘8723bu.ko’: No such file or directory
make: *** [install] Error 1"

I use ubuntu 14.04 and 3.10.69+, 
Do you have any idea to fix that problem ?

lwfinger, 2016-01-11:
You need to install the kernel-headers package to be able to compile an external module.

suat1479, 2016-01-12:
Thank you so much for your fast reply. I am a little beginner. I have installed below linux headers. 

'wget http://oph.mdrjr.net/meveric/kernel/XU3/linux-image-3.10.69+_3.10.69-20150224-XU3_armhf.deb
dpkg -i linux-image-3.10.69+_3.10.69-20150224-XU3_armhf.deb
wget http://oph.mdrjr.net/meveric/kernel/XU3/linux-headers-3.10.69+_3.10.69-20150224-XU3_armhf.deb
dpkg -i linux-headers-3.10.69+_3.10.69-20150224-XU3_armhf.deb'

Can you explain me with simple steps to compile as an external module? 
Best Regards;
Suat

lwfinger, 2016-01-12:
Each distro has different ways to install the necessary components, thus there is no simple prescription. The necessary pieces are the kernel headers, the compiler, and make. Once those are present, all that is needed is as follows:

make
sudo make install

Usually the headers are installed with "apt-get", "yum", "zypper", or whatever package manager is used by your distro, but wget and dpkg are likely OK. The critical thing is that the command "ls /lib/modules/$(uname -r)/build" returns a listing for a directory named "include". That contains the kernel headers.

suat1479, 2016-01-16:
Thank you so much for your information. Finally I have installed bluetooth and Wi-Fi, in bluetooth the driver looks like succes but cant connect any device. When i use hcitool scan cant find any bluetooth device. For Wi-Fi driver, When I use sudo make I see that warning "WARNING: "cfg80211_inform_bss_frame" [/home/odroid/rtl8723bu/8723bu.ko] undefined!
WARNING: "cfg80211_del_sta" [/home/odroid/rtl8723bu/8723bu.ko] undefined!
WARNING: "wiphy_new" [/home/odroid/rtl8723bu/8723bu.ko] undefined!"  After that I use sudo make install got "target '/lib/modules/3.10.92+/kernel/drivers/net/wireless/' is not a directory: No such file or directory" this error. Then I create a directory under net directory as a "wireless". The driver looks like installed but the device hasnt known yet? In iwconfig section I see nothing. You can check my command line window steps and errors in below. Do you have any idea about these warning and errors?

odroid@odroid:~$ cd rtl8723bu
odroid@odroid:~/rtl8723bu$ sudo make clean
cd hal ; rm -fr *.mod.c *.mod *.o .*.cmd */*.ko
cd core ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd os_dep ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
cd platform ; rm -fr *.mod.c *.mod *.o .*.cmd *.ko
rm -fr Module.symvers ; rm -fr Module.markers ; rm -fr modules.order
rm -fr *.mod.c *.mod *.o .*.cmd *.ko *~
rm -fr .tmp_versions
odroid@odroid:~/rtl8723bu$ make
make ARCH=arm CROSS_COMPILE= -C /lib/modules/3.10.92+/build M=/home/odroid/rtl8723bu  modules
make[1]: Entering directory '/home/odroid/linux'
  CC [M]  /home/odroid/rtl8723bu/core/rtw_cmd.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_security.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_debug.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_io.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_ioctl_query.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_ioctl_set.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_ieee80211.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_mlme.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_mlme_ext.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_wlan_util.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_vht.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_pwrctrl.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_rf.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_recv.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_sta_mgt.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_ap.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_xmit.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_p2p.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_tdls.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_br_ext.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_iol.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_sreset.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_btcoex.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_beamforming.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_odm.o
  CC [M]  /home/odroid/rtl8723bu/core/rtw_efuse.o
  CC [M]  /home/odroid/rtl8723bu/os_dep/osdep_service.o
  CC [M]  /home/odroid/rtl8723bu/os_dep/os_intfs.o
  CC [M]  /home/odroid/rtl8723bu/os_dep/usb_intf.o
  CC [M]  /home/odroid/rtl8723bu/os_dep/usb_ops_linux.o
  CC [M]  /home/odroid/rtl8723bu/os_dep/ioctl_linux.o
  CC [M]  /home/odroid/rtl8723bu/os_dep/xmit_linux.o
  CC [M]  /home/odroid/rtl8723bu/os_dep/mlme_linux.o
  CC [M]  /home/odroid/rtl8723bu/os_dep/recv_linux.o
  CC [M]  /home/odroid/rtl8723bu/os_dep/ioctl_cfg80211.o
  CC [M]  /home/odroid/rtl8723bu/os_dep/wifi_regd.o
  CC [M]  /home/odroid/rtl8723bu/os_dep/rtw_android.o
  CC [M]  /home/odroid/rtl8723bu/os_dep/rtw_proc.o
  CC [M]  /home/odroid/rtl8723bu/hal/hal_intf.o
  CC [M]  /home/odroid/rtl8723bu/hal/hal_com.o
  CC [M]  /home/odroid/rtl8723bu/hal/hal_com_phycfg.o
  CC [M]  /home/odroid/rtl8723bu/hal/hal_phy.o
  CC [M]  /home/odroid/rtl8723bu/hal/hal_btcoex.o
  CC [M]  /home/odroid/rtl8723bu/hal/hal_usb.o
  CC [M]  /home/odroid/rtl8723bu/hal/hal_usb_led.o
  CC [M]  /home/odroid/rtl8723bu/hal/HalPwrSeqCmd.o
  CC [M]  /home/odroid/rtl8723bu/hal/Hal8723BPwrSeq.o
  CC [M]  /home/odroid/rtl8723bu/hal/rtl8723b_sreset.o
  CC [M]  /home/odroid/rtl8723bu/hal/rtl8723b_hal_init.o
  CC [M]  /home/odroid/rtl8723bu/hal/rtl8723b_phycfg.o
  CC [M]  /home/odroid/rtl8723bu/hal/rtl8723b_rf6052.o
  CC [M]  /home/odroid/rtl8723bu/hal/rtl8723b_dm.o
  CC [M]  /home/odroid/rtl8723bu/hal/rtl8723b_rxdesc.o
  CC [M]  /home/odroid/rtl8723bu/hal/rtl8723b_cmd.o
[... 192 lines cut ...]
        lewlsz  Read size of LE White List
        lewlclr Clear LE White list
        lecc    Create a LE Connection
        ledc    Disconnect a LE Connection
        lecup   LE Connection Update

For more information on the usage of each command use:
        hcitool <command> --help
odroid@odroid:~$ hciconfig -a
hci0:   Type: BR/EDR  Bus: USB
        BD Address: 54:E4:BD:DA:97:59  ACL MTU: 820:8  SCO MTU: 255:16
        UP RUNNING PSCAN
        RX bytes:1124 acl:0 sco:0 events:59 errors:0
        TX bytes:1038 acl:0 sco:0 commands:59 errors:0
        Features: 0xff 0xff 0xff 0xfe 0xdb 0xff 0x7b 0x87
        Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3
        Link policy: RSWITCH HOLD SNIFF PARK
        Link mode: SLAVE ACCEPT
        Name: 'odroid-0'
        Class: 0x600100
        Service Classes: Audio, Telephony
        Device Class: Computer, Uncategorized
        HCI Version: 4.0 (0x6)  Revision: 0xb
        LMP Version: 4.0 (0x6)  Subversion: 0x8723
        Manufacturer: Realtek Semiconductor Corporation (93)

lwfinger, 2016-01-18:
I do not know about the BT problems, but the problem of undefined globals in linking the wifi driver is due to missing items in your kernel configuration. You should have selected CFG80211. My parameters are

CONFIG_WIRELESS_EXT=y
CONFIG_WEXT_CORE=y
CONFIG_WEXT_PROC=y
CONFIG_WEXT_PRIV=y
CONFIG_CFG80211=m
# CONFIG_NL80211_TESTMODE is not set
# CONFIG_CFG80211_DEVELOPER_WARNINGS is not set
# CONFIG_CFG80211_REG_DEBUG is not set
CONFIG_CFG80211_DEFAULT_PS=y
# CONFIG_CFG80211_DEBUGFS is not set
# CONFIG_CFG80211_INTERNAL_REGDB is not set
CONFIG_CFG80211_CRDA_SUPPORT=y
CONFIG_CFG80211_WEXT=y
CONFIG_LIB80211=m