lwfinger/rtl8723bu #18: multiple rtl8723bu units at once

qmnonic, 2016-01-12:
If I run multiple rtl8723bu (odba:b720) realtek units at once, only one of the units will work.  From what I can tell, it's consistently the the last one loaded (i.e. wlan3 vs wlan2), so order matters.

I have 3 RTL8723BU USB units, and tried plugging random pairs of them in at once to ensure it wasn't the actual device, or the usb port itself.

Seems only to be related to the order they are picked up by the OS.

#### OS details (happens with kernel 3 and 4)
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.1.13+-v7 #1 SMP PREEMPT Mon Jan 11 07:08:40 UTC 2016 armv7l GNU/Linux
```

#### USB device details (0bda:b720 are the rtl8723bu's)
```
pi@raspberrypi:~ $ lsusb
Bus 001 Device 008: ID 0bda:8176 Realtek Semiconductor Corp. RTL8188CUS 802.11n WLAN Adapter
Bus 001 Device 006: ID 0bda:8176 Realtek Semiconductor Corp. RTL8188CUS 802.11n WLAN Adapter
Bus 001 Device 005: ID 0bda:8176 Realtek Semiconductor Corp. RTL8188CUS 802.11n WLAN Adapter
Bus 001 Device 009: ID 0bda:b720 Realtek Semiconductor Corp. 
Bus 001 Device 007: ID 0bda:b720 Realtek Semiconductor Corp. 
Bus 001 Device 004: ID 0424:2514 Standard Microsystems Corp. USB 2.0 Hub
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9514 Standard Microsystems Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

I'm using these units specifically:
http://www.ccandc.com.tw/products_2.php?prod=269

#### Further context

I tried configuring each as a normal wifi card and only the second one would work.

Similarly, configuring it with hostapd (creating an access point), only the second one would create an access point.  That is, if i configure hostapd.conf to use wlan3, it works.  If i configure hostapd.conf to use wlan2, it doesn't work.

However, the same scenarios work fine with the RTL8188CUS usb wifi adapters that i have plugged into my device.  Thus, I believe it's a driver problem, rather than a kernel/OS problem.

Matt.

lwfinger, 2016-01-13:
I think I know the cause of the problem. The USB drivers from Realtek use a lot of global variables. This is fine with one device using the driver, but with more than one, the values get mixed up whenever the variables are read/write. The fix is to move the global quantities into the private variables, which are unique for each device.

That said, I have no idea how long it will take to find and move all the variables. Keep pulling the repo and retrying the experiment. As I only have one device, I will not be able to perform a definitive test.

qmnonic, 2016-01-13:
I can test any changes you make and let you know.  Thanks!