lwfinger/rtl8723bu #19: Chance of getting put in mainline?

steelcowboy, 2016-01-24:
Is there any chance that this will eventually be incorporated into the main Linux kernel, or is there something preventing it?

lwfinger, 2016-01-25:
The only thing preventing it is the 4-6 month's work making the source fit kernel requirements, even the lesser criteria of the staging tree.

dsx724, 2016-01-31:
Hi Larry, do you have a rough estimate as to the cost to do so?

lwfinger, 2016-02-01:
No not really, but it would be in the many thousands of dollars. To get an idea of the number of things to fix, I ran the script that checks for violations of the kernel coding standards on a catenated copy of the source. It reported 48,000 errors and 25,000 warnings.

One option is to use the driver rtl8xxxu that was inserted into kernel 4.4. It is very much a work in progress, but it might work for you. For example, the author of that driver and I have identical dongles. The driver works with his APs, but not with mine. Thus far, I have not gotten time to capture and decode the over-the-air packets to see what is wrong.

dsx724, 2016-02-02:
Thanks for the information. In my tests, the RTL8188EU performs much worse than the RTL8188CU with the driver in staging. The RTL8188EU throughput is very inconsistent and normally 30% lower than the RTL8188CU.  Do you have a rough figure for bringing the performance to the same level or greater? It doesn't have to be mainline.