lwfinger/rtl8723bu #20: Low Bluetooth signal strength with wifi down

studiofuga, 2016-01-28:
Hello, and thank you very much for your work.
We are using the rtl8723bu driver and rtl8724au_bt driver on a rtl8723bu usb dongle. It works perfectly with some minor issue that we were able to workaround.
The big problem we're experiencing now (that was mentioned in the issue #2) is that if the wifi adapter isn't upped at least once, the bt transmit level is very low.
Our device (based on freescale ix6 arm) is booted with wifi driver loaded but adapter down. Our application is advertising through btle. If we test with LightBlue, an iOS app that displays the signal strength, we can see a signal level of about -90dbm at 1.5m from the dongle. When we up the adapter (ifup wlan0), the signal immediately raises to -60dbm.
if we down the adapter, the signals is still to pretty good levels (-60dbm).
Notice that we had an issue with the IPS, we removed the CONFIG_IPS flags because when the driver entered the power save mode, we were unable to communicate reliably with the driver again.
The USB ids are: 0bda:b720. kernel is 4.2.0

Thank you and best regards

Federico Fuga

lwfinger, 2016-01-28:
It is well known that most Realtek drivers have problems with power save. I would expect the commenting out the line "#define CONFIG_IPS   1" would turn off IPS.

I am not surprised that BT has low signal until the radio has been properly initialized, but I would have no idea how to fix that.

studiofuga, 2016-01-29:
Ok Thanks, I'll let you know if we're able to fix it.

Pivek, 2020-06-24:
Hello, were you able to fix it?
I am experiencing the similar issue. When wlan0 is down or 8723bu driver is removed, signal of bluetooth is very low. I am curious if you were able to resolve it.