lwfinger/rtl8723bu #22: Compile errorn on kernel 4.3.4

waulee, 2016-02-08:
Hi there,

recently I have updated my fedora 22 to kernel 4.3.4.200 now the driver could not be compiled with the following errors. Could You make a patch please.

Thanks in adwance:
Bence

  CC [M]  $/rtl8723bu/core/rtw_cmd.o
In file included from $/rtl8723bu/include/drv_types.h:65:0,
                 from $/rtl8723bu/core/rtw_cmd.c:22:
$/rtl8723bu/core/rtw_cmd.c: In function ‘btinfo_evt_dump’:
$/rtl8723bu/include/rtw_debug.h:187:19: error: void value not ignored as it ought to be
  #define _seqdump seq_printf
                   ^
$/rtl8723bu/include/rtw_debug.h:240:7: note: in expansion of macro ‘_seqdump’
    if(_seqdump(sel, fmt, ##arg)) /*rtw_warn_on(1)*/; \
       ^
$/rtl8723bu/core/rtw_cmd.c:3166:2: note: in expansion of macro ‘DBG_871X_SEL_NL’
  DBG_871X_SEL_NL(sel, "cid:0x%02x, len:%u\n", info->cid, info->len);
  ^
$/rtl8723bu/include/rtw_debug.h:187:19: error: void value not ignored as it ought to be
  #define _seqdump seq_printf
                   ^
$/rtl8723bu/include/rtw_debug.h:240:7: note: in expansion of macro ‘_seqdump’
    if(_seqdump(sel, fmt, ##arg)) /*rtw_warn_on(1)*/; \
       ^
$/rtl8723bu/core/rtw_cmd.c:3169:2: note: in expansion of macro ‘DBG_871X_SEL_NL’
  DBG_871X_SEL_NL(sel, "byte2:%s%s%s%s%s%s%s%s\n"
  ^
$/rtl8723bu/include/rtw_debug.h:187:19: error: void value not ignored as it ought to be
  #define _seqdump seq_printf
                   ^
$/rtl8723bu/include/rtw_debug.h:240:7: note: in expansion of macro ‘_seqdump’
    if(_seqdump(sel, fmt, ##arg)) /*rtw_warn_on(1)*/; \
       ^
$/rtl8723bu/core/rtw_cmd.c:3181:2: note: in expansion of macro ‘DBG_871X_SEL_NL’
  DBG_871X_SEL_NL(sel, "retry_cnt:%u\n", info->retry_cnt);
  ^
$/rtl8723bu/include/rtw_debug.h:187:19: error: void value not ignored as it ought to be
  #define _seqdump seq_printf
                   ^
$/rtl8723bu/include/rtw_debug.h:240:7: note: in expansion of macro ‘_seqdump’
    if(_seqdump(sel, fmt, ##arg)) /*rtw_warn_on(1)*/; \
       ^
$/rtl8723bu/core/rtw_cmd.c:3184:2: note: in expansion of macro ‘DBG_871X_SEL_NL’
  DBG_871X_SEL_NL(sel, "rssi:%u\n", info->rssi);
  ^
$/rtl8723bu/include/rtw_debug.h:187:19: error: void value not ignored as it ought to be
  #define _seqdump seq_printf
                   ^
$/rtl8723bu/include/rtw_debug.h:240:7: note: in expansion of macro ‘_seqdump’
    if(_seqdump(sel, fmt, ##arg)) /*rtw_warn_on(1)*/; \
       ^
$/rtl8723bu/core/rtw_cmd.c:3187:2: note: in expansion of macro ‘DBG_871X_SEL_NL’
  DBG_871X_SEL_NL(sel, "byte5:%s%s\n"
  ^
scripts/Makefile.build:258: recipe for target '$/rtl8723bu/core/rtw_cmd.o' failed

lwfinger, 2016-02-08:
That problem was fixed in the source on December 2, 2015. Did you pull a fresh copy? Note, if you are using the zip form of the repo, then you need to download a complete fresh copy of the source everytime I change anything. After getting the repo with a "git clone" command, all that is needed is a "git pull".

waulee, 2016-02-08:
argh, 
Thank You! I must have overseen the result from the git pull part. Its working now.

closed 2016-02-08 by waulee