lwfinger/rtl8723bu #23: implicit declaration of function ‘autosuspend_enter’ when compiling on BeagleBoneBlack

mogar, 2016-02-11:
Hi, 

When I try to compile this on a BBB, I get the following error:

	root@beaglebone:~/rtl8723bu# make
	make ARCH=arm CROSS_COMPILE= -C /lib/modules/3.8.13-bone79/build M=/root/rtl8723bu  modules
	make[1]: Entering directory `/usr/src/linux-headers-3.8.13-bone79'
	  CC [M]  /root/rtl8723bu/core/rtw_pwrctrl.o
	/root/rtl8723bu/core/rtw_pwrctrl.c: In function ‘rtw_ps_processor’:
	/root/rtl8723bu/core/rtw_pwrctrl.c:387:5: error: implicit declaration of function ‘autosuspend_enter’ [-Werror=implicit-function-declaration]
	cc1: some warnings being treated as errors
	make[2]: *** [/root/rtl8723bu/core/rtw_pwrctrl.o] Error 1
	make[1]: *** [_module_/root/rtl8723bu] Error 2
	make[1]: Leaving directory `/usr/src/linux-headers-3.8.13-bone79'
	make: *** [modules] Error 2

It looks like the function autosuspend_enter is defined in os_dep/usb_intf.c, but I can't find it's declaration. I pulled the most recent code today.

Thanks for any help you can give.

lwfinger, 2016-02-12:
Your kernel configuration obviously defines CONFIG_USB_SUSPEND, which does not happen for x86. I suspect it is an ARM thing. I added the necessary header include to fix the build error.

Note: Having CONFIG_USB_SUSPEND defined compiles quite a bit of code that I have never tested.

mogar, 2016-02-17:
Thanks for adding that header. Things now compile and install fine, but I'm unable to connect to a wpa network. I'm not sure what to test next; can you give me some guidance on this?

I'm trying to use wpa_supplicant like so:
root@beaglebone:~# wpa_supplicant -B -iwlan0 -c/etc/wpa_supplicant/wpa_supplicant.conf
ioctl[SIOCSIWAP]: Operation not permitted
ioctl[SIOCSIWENCODEEXT]: Invalid argument
ioctl[SIOCSIWENCODEEXT]: Invalid argument
root@beaglebone:~# dhclient wlan0

When I look at dmesg, I see the following:

root@beaglebone:~# dmesg | grep -i wlan0
[   15.516670] RTL871X: rtw_ndev_init(wlan0)
[  404.326386] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  409.013589] RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
... This message repeats~10 times
[  992.527719] RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
[ 1071.031944] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready

root@beaglebone:~# dmesg | grep -i rtl8723bu
[   15.224605] RTL871X: rtl8723bu v4.3.6.11_12942.20141204_BTCOEX20140507-4E40
[   15.224612] RTL871X: rtl8723bu BT-Coex version = BTCOEX20140507-4E40
[   15.224755] rtl8723bu 1-1:1.2: usb_probe_interface
[   15.224818] rtl8723bu 1-1:1.2: usb_probe_interface - got id
[   15.549576] usbcore: registered new interface driver rtl8723bu
[  216.113357] RTL871X: RTW_ADAPTIVITY_MODE_NORMAL
[  216.664882] RTL871X: nolinked power save leave
[  216.873714] RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
[  216.960385] RTL871X: start auth
[  216.980211] RTL871X: auth success, start assoc
[  222.694951] RTL871X: nolinked power save enter
[  228.573751] RTL871X: RTW_ADAPTIVITY_EN_AUTO, chplan:0x20, Regulation:0,0
[  228.573810] RTL871X: RTW_ADAPTIVITY_MODE_NORMAL
[  229.231296] RTL871X: nolinked power save leave
[  231.333187] RTL871X: nolinked power save enter
[  232.035707] RTL871X: RTW_ADAPTIVITY_EN_AUTO, chplan:0x20, Regulation:0,0
[  232.035768] RTL871X: RTW_ADAPTIVITY_MODE_NORMAL