lwfinger/rtl8723bu #26: Setting up a WIFI AP via hostapd

MH6, 2016-04-21:
Has anyone been successful with using this RTL8723bu driver to setup a WIFI access point via hostapd?

I can see the network from my client device, but every time it tries to connect, it fails with the following line in dmesg:

> RTL871X: ap recv disassoc reason code(8)

This is the same no matter what client I use (Android phone, Windows PC, etc.)

This reason code maps to "Disassociated because sending station is leaving (or has left) BSS". Which doesn't make sense in this context as the phone is stationary with a strong signal. The module works fine as a WIFI client, just not as an AP.

At this point I'm not sure if it's an issue with the driver, hostapd, or some configuration setting somewhere.

mkborg, 2016-04-21:
We use hostapd with **driver=nl80211** to host access point on rtl8723bu chip. It works.

MH6, 2016-04-21:
Ok thanks for letting me know @mkborg 

Unfortunately using the stock hostapd binary in Raspbian with **driver=nl80211** doesn't even bring up the network at all.

> nl80211: interface wlan0 in phy phy0
rfkill: initial event: idx=0 type=2 op=0 soft=0 hard=0
rfkill: initial event: idx=1 type=1 op=0 soft=0 hard=0
nl80211: Using driver-based off-channel TX
nl80211: Add own interface ifindex 4
nl80211: Set mode ifindex 4 iftype 3 (AP)
nl80211: Create interface iftype 6 (MONITOR)
Failed to create interface mon.wlan0: -19 (No such device)
nl80211: Failed to set interface wlan0 into AP mode
netlink: Operstate: linkmode=0, operstate=6
nl80211: Set mode ifindex 4 iftype 2 (STATION)
nl80211 driver initialization failed.

My hostapd.conf:
>
interface=wlan0
driver=nl80211
ssid=Pi
hw_mode=g
channel=11
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=password
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

I will keep working on the problem.

lwfinger, 2016-04-21:
Make certain that configuration symbol CONFIG_IOCTL_CFG80211 is defined in your build.

MH6, 2016-04-21:
@lwfinger It seems to be defined in autoconf.h

Do I need to use the CONFIG_AUTOCFG_CP flag when I build? It is set to no by default.

I as of yet have made no changes to the driver from the current repo version. Just did a git clone and make/make install.

lwfinger, 2016-04-21:
Do not use CONFIG_AUTOCFG_CP. If you do, autoconfig.h will be replaced every time you run make, and every source file will need to be recompiled. That is a lot of work for nothing.

closed 2016-04-21 by MH6