lwfinger/rtl8723bu #27: porting to Mace

ahmednoori94, 2016-04-23:


lwfinger, 2016-04-23:
I know nothing of writing drivers for OS X.

kaiyes, 2016-08-12:
@lwfinger could you please point us the direction. @ahmednoori94 and I came from the same forum. Trying to make the usb wifi and bluetooth of a tab work. 

This one to be exact.
https://www.youtube.com/watch?v=7JBCLv0486E

We were able to install mac os xon it and its running great. But wifi doesn't work. Some one was able to use your driver and make it work with linux. But We need to port this to mac. 

Anyway thanks for all the work you have done.

lwfinger, 2016-08-12:
I know nothing about writing drivers for OS X. No doubt the Linux driver could be rewritten, but it would be a lot of work. I certainly have no interest in having any part of such an undertaking.

closed 2016-08-12 by lwfinger

Snailll, 2016-10-20:
@kaiyes maybe you can try this [http://www.hackintosh.zone/file/933-realtek-usb-wlanu-mac-osx-1011/](url)  `1830.2.0904.2015
	
	Reason:
	1. [8188EU] Revise IOT Issue with Atheros AP
	2. [8192EU] Fix wake up hang when RF is on.
	3. Modify the communication mechanism between user/kernel levels.
	4. [8814A] FIX VHT 1SS MCS 8, 9
	5. [MacOS] Support 802.1x WEP.
	6. Indicate DataRate behaviour.
	7. [8723BU] revise .plist about interface index.
	8. Add debug message.
	9. [8814AU] Customization for specific ID. 
	10. [8723BU] Fix 10.11 OS restart then Tx stuck.
	11. [8814AU] Customization for specific ID (Revision).
	12. Fix system crash issue when hot-plug.

`