lwfinger/rtl8723bu #29: Mp_Precomp.h: No such file or directory

howardqiao, 2016-07-17:
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.4.15-v7+ #1 SMP Sun Jul 17 18:21:18 CST 2016 armv7l GNU/Linux

pi@raspberrypi:~/test2/rtl8723bu $ make -j5
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- -C /home/pi/rpikernel/linux M=/home/pi/test2/rtl8723bu  modules
make[1]: Entering directory '/home/pi/rpikernel/linux'
  CC [M]  /home/pi/test2/rtl8723bu/hal/hal_btcoex.o
  CC [M]  /home/pi/test2/rtl8723bu/hal/rtl8723b_rf6052.o
  CC [M]  /home/pi/test2/rtl8723bu/hal/rtl8723b_dm.o
  CC [M]  /home/pi/test2/rtl8723bu/hal/rtl8723b_rxdesc.o
  CC [M]  /home/pi/test2/rtl8723bu/hal/rtl8723b_cmd.o
/home/pi/test2/rtl8723bu/hal/hal_btcoex.c:26:24: fatal error: Mp_Precomp.h: No such file or directory
 #include <Mp_Precomp.h>
                        ^
compilation terminated.

howardqiao, 2016-07-17:
1. copy Mp_Precomp.h to hal/ 
2. nano hal_btcoex.c changed <Mp_Precomp.h> to "Mp_Preocmp.h"
3.compile driver
4.install driver
pi@raspberrypi:~ $ sudo modprobe 8723bu
modprobe: ERROR: could not insert '8723bu': Unknown symbol in module, or unknown parameter (see dmesg)

[ 1362.813929] 8723bu: Unknown symbol EXhalbtc8723b2ant_ConnectNotify (err 0)
[ 1362.814030] 8723bu: Unknown symbol EXhalbtc8723b1ant_InitHwConfig (err 0)
[ 1362.814090] 8723bu: Unknown symbol EXhalbtc8723b2ant_HaltNotify (err 0)
[ 1362.814137] 8723bu: Unknown symbol EXhalbtc8723b1ant_Periodical (err 0)
[ 1362.814339] 8723bu: Unknown symbol EXhalbtc8723b2ant_BtInfoNotify (err 0)
[ 1362.814406] 8723bu: Unknown symbol EXhalbtc8723b1ant_BtInfoNotify (err 0)
[ 1362.814496] 8723bu: Unknown symbol EXhalbtc8723b1ant_PowerOnSetting (err 0)
[ 1362.814538] 8723bu: Unknown symbol EXhalbtc8723b2ant_MediaStatusNotify (err 0                                                                             )
[ 1362.814692] 8723bu: Unknown symbol EXhalbtc8723b1ant_ConnectNotify (err 0)
[ 1362.814808] 8723bu: Unknown symbol EXhalbtc8723b2ant_ScanNotify (err 0)
[ 1362.815019] 8723bu: Unknown symbol EXhalbtc8723b2ant_PowerOnSetting (err 0)
[ 1362.815084] 8723bu: Unknown symbol EXhalbtc8723b1ant_SpecialPacketNotify (err                                                                              0)
[ 1362.815116] 8723bu: Unknown symbol EXhalbtc8723b1ant_CoexDmReset (err 0)
[ 1362.815219] 8723bu: Unknown symbol EXhalbtc8723b1ant_PnpNotify (err 0)
[ 1362.815301] 8723bu: Unknown symbol EXhalbtc8723b1ant_IpsNotify (err 0)
[ 1362.815400] 8723bu: Unknown symbol EXhalbtc8723b1ant_InitCoexDm (err 0)
[ 1362.815648] 8723bu: Unknown symbol EXhalbtc8723b1ant_RfStatusNotify (err 0)
[ 1362.815774] 8723bu: Unknown symbol EXhalbtc8723b1ant_MediaStatusNotify (err 0                                                                             )
[ 1362.815834] 8723bu: Unknown symbol EXhalbtc8723b2ant_IpsNotify (err 0)
[ 1362.815871] 8723bu: Unknown symbol EXhalbtc8723b2ant_InitHwConfig (err 0)
[ 1362.816044] 8723bu: Unknown symbol EXhalbtc8723b1ant_LpsNotify (err 0)
[ 1362.816086] 8723bu: Unknown symbol EXhalbtc8723b2ant_LpsNotify (err 0)
[ 1362.816123] 8723bu: Unknown symbol EXhalbtc8723b2ant_PnpNotify (err 0)
[ 1362.816156] 8723bu: Unknown symbol EXhalbtc8723b1ant_ScanNotify (err 0)
[ 1362.816197] 8723bu: Unknown symbol EXhalbtc8723b2ant_DisplayCoexInfo (err 0)
[ 1362.816235] 8723bu: Unknown symbol EXhalbtc8723b1ant_DisplayCoexInfo (err 0)
[ 1362.816309] 8723bu: Unknown symbol EXhalbtc8723b2ant_SpecialPacketNotify (err                                                                              0)
[ 1362.816347] 8723bu: Unknown symbol EXhalbtc8723b2ant_Periodical (err 0)
[ 1362.816429] 8723bu: Unknown symbol EXhalbtc8723b2ant_InitCoexDm (err 0)
[ 1362.816466] 8723bu: Unknown symbol EXhalbtc8723b1ant_HaltNotify (err 0)

lwfinger, 2016-07-17:
Now you see the problems with making random code changes. The file Mp_Precomp.h is present; however, the search path on the RPi compiler apparently differs from that used for x86. The problem is now fixed and it should build on both architectures. Use 'git checkout -f" to remove all your changes and then "git pull" to get the updated code.

howardqiao, 2016-07-18:
Thank you very much!Finally succeeded！

closed 2016-07-18 by howardqiao