lwfinger/rtl8723bu #30: make

dosgo, 2016-07-19:
[image]

lwfinger, 2016-07-19:
You have not installed the kernel headers. Depending on your distro, it may be called the kernel-devel package.

dosgo, 2016-07-19:
Raspberry Pi

dosgo, 2016-07-19:
Unable to locate package kernel-devel

howardqiao, 2016-08-08:
use rpi-source to get the right kernel-source
https://github.com/notro/rpi-source/wiki

stumbino, 2017-09-30:
get this error when I use make
make
make ARCH=arm CROSS_COMPILE=arm-linux-gnueabihf- -C /lib/modules/4.4.24-v7+/build M=/home/pi/rtl8723bu modules
make[1]: *** /lib/modules/4.4.24-v7+/build: No such file or directory. Stop.
Makefile:332: recipe for target 'modules' failed
make: *** [modules] Error 2

lwfinger, 2017-09-30:
Did you even think to Google this error? It means that you have not installed the prerequisites needed to build out-of-kernel modules.