lwfinger/rtl8723bu #31: Sigmentation fault when unloading driver

advisoft, 2016-07-29:
Hello lwfinger,

I have run into an issue, when unloading the driver I get the following segfault:
```
/ # rmmod 8723bu
[  967.657737] RTL871X: module exit start
[  967.661925] usbcore: deregistering interface driver rtl8723bu
[  967.672778] ------------[ cut here ]------------
[  967.677678] kernel BUG at net/wireless/core.c:653!
[  967.682721] Internal error: Oops - BUG: 0 [#1] SMP ARM
[  967.688125] Modules linked in: can_raw can c_can_platform c_can can_dev rtk_btusb(O) bluetooth 6lowpan_iphc rt2800usb rt2800lib crc_ccitt rt2x00usb rt2x00lib mac80211 pwm_bl leds_pwm usb_f_acm u_serial usb_f_ecm g_multi usb_f_mass_storage usb_f_rndis u_ether libcomposite 8723bu(O-) cfg80211
[  967.715464] CPU: 0 PID: 314 Comm: rmmod Tainted: G           O 3.14.1+ #4
[  967.722608] task: de696400 ti: da008000 task.ti: da008000
[  967.728806] PC is at wiphy_unregister+0xb0/0x258 [cfg80211]
[  967.734959] LR is at wiphy_unregister+0x98/0x258 [cfg80211]
[  967.740830] pc : [<bf000530>]    lr : [<bf000518>]    psr: 200d0013
[  967.740830] sp : da009e70  ip : 00000000  fp : 00000000
[  967.752891] r10: 00000000  r9 : 00000000  r8 : de44f000
[  967.758388] r7 : de44e800  r6 : de502000  r5 : 00000000  r4 : de502140
[  967.765253] r3 : de502038  r2 : de502c08  r1 : 600d0013  r0 : c08caee8
[  967.772119] Flags: nzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  967.779622] Control: 10c5387d  Table: 9a0ac019  DAC: 00000015
[  967.785666] Process rmmod (pid: 314, stack limit = 0xda008248)
[  967.791800] Stack: (0xda009e70 to 0xda00a000)
[  967.796393] 9e60:                                     da33cab0 de41f0d8 de0280c4 da009ec0
[  967.805001] 9e80: e090f000 e0912f04 de502c00 e0912000 de44e9b8 de44e800 de44f000 bf0df4ac
[  967.813609] 9ea0: e090f000 00000000 de44e9b8 bf0cc5d4 e090f000 de41b600 de41b600 00000000
[  967.822218] 9ec0: bf122f3c bf0cda9c de41b620 de3eac00 de41b600 c03afb7c de41b620 bf122f3c
[  967.830826] 9ee0: de41b654 bec2eb80 c000dbc4 da008000 00000000 c033e434 de41b620 bf122f3c
[  967.839434] 9f00: de41b654 c033ea98 bf122f3c bf122f08 c08604c8 c033e188 bf122f08 c03af120
[  967.848041] 9f20: 00000000 bf13e748 c08604c8 bec2eb80 c000dbc4 bf115ab0 00000000 c00975d0
[  967.856648] 9f40: 00000000 bf13e748 00000880 da009f40 33323738 b6007562 00000001 00000001
[  967.865254] 9f60: 00000001 c010bc54 00100871 00000000 00000000 b6f1f000 00000003 de6a3aa8
[  967.873862] 9f80: 00000000 b6f1f568 00027d38 bd94362f 0000007d 0001fc1b 33323738 b6007562
[  967.882469] 9fa0: 00000081 c000da40 0001fc1b 33323738 bec2eb80 00000880 00000000 bec2ee38
[  967.891077] 9fc0: 0001fc1b 33323738 b6007562 00000081 bec2ee34 000000b6 b6f1f000 00000000
[  967.899684] 9fe0: bec2eb78 bec2eb68 0001fb5d b6e25442 800d0030 bec2eb80 00000000 00000000
[  967.909332] [<bf000530>] (wiphy_unregister [cfg80211]) from [<bf0df4ac>] (rtw_wdev_unregister+0x48/0x4c [8723bu])
[  967.921027] [<bf0df4ac>] (rtw_wdev_unregister [8723bu]) from [<bf0cc5d4>] (rtw_unregister_netdevs+0x38/0x64 [8723bu])
[  967.932983] [<bf0cc5d4>] (rtw_unregister_netdevs [8723bu]) from [<bf0cda9c>] (rtw_dev_remove+0x20/0x78 [8723bu])
[  967.944112] [<bf0cda9c>] (rtw_dev_remove [8723bu]) from [<c03afb7c>] (usb_unbind_interface+0x60/0x160)
[  967.953936] [<c03afb7c>] (usb_unbind_interface) from [<c033e434>] (__device_release_driver+0x7c/0xc4)
[  967.963647] [<c033e434>] (__device_release_driver) from [<c033ea98>] (driver_detach+0xa4/0xcc)
[  967.972714] [<c033ea98>] (driver_detach) from [<c033e188>] (bus_remove_driver+0x64/0x8c)
[  967.981235] [<c033e188>] (bus_remove_driver) from [<c03af120>] (usb_deregister+0x5c/0xd0)
[  967.990231] [<c03af120>] (usb_deregister) from [<bf115ab0>] (rtw_drv_halt+0x38/0x6c [8723bu])
[  967.999598] [<bf115ab0>] (rtw_drv_halt [8723bu]) from [<c00975d0>] (SyS_delete_module+0x13c/0x1c0)
[  968.009044] [<c00975d0>] (SyS_delete_module) from [<c000da40>] (ret_fast_syscall+0x0/0x30)
[  968.017746] Code: e5c43220 e2443f42 e1520003 0a000000 (e7f001f2) 
[  968.024187] ---[ end trace 831f666d51db0056 ]---
Segmentation fault
```
Here is some information about the system.
```
/ # uname -a
Linux EnatelLinux 3.14.1+ #4 SMP Fri Jul 29 13:55:49 NZST 2016 armv7l GNU/Linux
```
It is on a TI AM3352 using a LM811 USB IC.
I have looked at the line of the code raising the error, and understanding the potential cause is beyond me at this point in time.

Are you able to suggest a potential starting point of investigation?

I guess a named stack trace would be a good start, I will have a go at getting one of those and updating the post.

Thanks,
Phil

lwfinger, 2016-07-29:
Having a named stack trace would be a good step. Any possibility of running a newer kernel? The current version of net/wireless/core.c contains no BUG() statements, only WARN() lines. In the old days, some authors went overboard and improperly used BUG() when WARN() was appropriate. An out-of=order shutdown sequence for a device is not usually a good reason to crash the system.

advisoft, 2016-07-30:
It would be good to move to a newer kernel, but too much time pressure at this point in time. I guess the best approach would be to look at the latest version of that function, and patch the current one if everything makes sense, and be done with it. If further investigation is of benefit to you, I am happy to look into it further.

Thanks again for your quick response.

advisoft, 2016-08-04:
I have upgraded the kernel to 4.6.5, and the crash is swapped for a warning. However it still hangs on shutdown if the interface associated with the driver is up. None of the other wifi devices seem to do this. Is this something fixable? Cheers.

mpvader, 2016-09-26:
@advisoft hi, you should probably try again. See also #34, #36 and the recent commits.