lwfinger/rtl8723bu #32: Authentication timed out

a8924040, 2016-08-02:
Dear lwfinger,
This driver can't connect to AP , debug MSG as below ,  my kernel version is 3.0.8
please help to assist this issue

# wpa_cli -i wlan0 wps_pbc
OK
# Trying to associate with cc:e1:dRTL871X: set bssid:cc:e1:d5:2e:06:76
5:2e:06:76 (SSIDRTL871X: set ssid [BuffaloBenny] fw_state=0x00000188
='BuffaloBenny' freq=2462 MHz)
RTL871X: start auth
Authentication with cc:e1:d5:2e:06:76 timed out.
Trying to associate with cc:e1:dRTL871X: set bssid:cc:e1:d5:2e:06:76
5:2e:06:76 (SSIDRTL871X: set ssid [BuffaloBenny] fw_state=0x00000188
='BuffaloBenny' freq=2462 MHz)
RTL871X: start auth
Authentication with cc:e1:d5:2e:06:76 timed out.
Trying to associate with cc:e1:dRTL871X: set bssid:cc:e1:d5:2e:06:76
5:2e:06:76 (SSIDRTL871X: set ssid [BuffaloBenny] fw_state=0x00000188
='BuffaloBenny' freq=2462 MHz)
RTL871X: start auth
RTL871X: sta recv deauth reason code(1) sta:cc:e1:d5:2e:06:76, ignore = 0
Authentication with cc:e1:d5:2e:06:76 timed out.
Trying to associate with cc:e1:dRTL871X: set bssid:cc:e1:d5:2e:06:76
5:2e:06:76 (SSIDRTL871X: set ssid [BuffaloBenny] fw_state=0x00000188
='BuffaloBenny' freq=2462 MHz)
RTL871X: start auth
Authentication with cc:e1:d5:2e:06:76 timed out.

lwfinger, 2016-08-04:
Are you sure that your configuration file for wpa_supplicant is correct? The command-line version of that utility is difficult to use.

neophob, 2017-01-20:
I have a similar issue:

```
Jan 20 18:21:19 hostname wpa_supplicant[226]: wlan0: CTRL-EVENT-DISCONNECTED bssid=00:23:df:f8:b7:51 reason=3 locally_generated=1
Jan 20 18:21:20 hostname wpa_supplicant[226]: wlan0: Trying to associate with 00:23:df:f8:b7:51 (SSID='myssid' freq=2452 MHz)
Jan 20 18:21:20 hostname kernel: RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
Jan 20 18:21:21 hostname kernel: RTL871X: start auth
Jan 20 18:21:21 hostname kernel: RTL871X: auth success, start assoc
Jan 20 18:21:30 hostname wpa_supplicant[226]: wlan0: Authentication with 00:23:df:f8:b7:51 timed out.
Jan 20 18:21:30 NEEO-0b036d11 wpa_supplicant[226]: wlan0: CTRL-EVENT-DISCONNECTED bssid=00:23:df:f8:b7:51 reason=3 locally_generated=1
Jan 20 18:21:30 NEEO-0b036d11 wpa_supplicant[226]: wlan0: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="myssid" auth_failures=6 duration=143 reason=CONN_FAILED
```

the same message appears after about 2 minutes. interesting fact:
- the same driver works with some netgear AP, but not with an Apple Express (v7.6)
- older versions of the driver worked with the Apple Express (https://github.com/lwfinger/rtl8723bu/commit/b8554dd19e306597ed51287838c7b460abbeede3)

any hints how to debug?

lwfinger, 2017-01-20:
"git bisect" is likely the easiest way. Harder would be capturing the over-the-air packets. My APs are Netgear, Linksys, and Trendnet. All work.

neophob, 2017-01-21:
more details, i tested with another hardware to be sure its not hardware related..

when i run `wpa_supplicant -dd` the output shows this:

```
wlan0: set_disable_max_amsdu: -1
wlan0: set_ampdu_factor: -1
wlan0: set_ampdu_density: -1
wlan0: set_disable_ht40: 0
wlan0: set_disable_sgi: 0
wlan0: set_disable_ldpc: 0
wlan0: Determining shared radio frequencies (max len 1)
wlan0: Shared frequencies (len=0): completed iteration
nl80211: Set mode ifindex 8 iftype 2 (STATION)
nl80211: Unsubscribe mgmt frames handle 0x889193c1 (mode change)
nl80211: Subscribe to mgmt frames with non-AP handle 0x191b48
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=040a
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=040b
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=040c
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=040d
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=090a
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=090b
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=090c
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=090d
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=0409506f9a09
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=7f506f9a09
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=0801
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=06
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=0a07
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=0a11
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=1101
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=1102
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=0505
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=0500
nl80211: Connect (ifindex=8)
  * bssid=00:23:df:f8:b7:51
  * bssid_hint=00:23:df:f8:b7:51
  * freq=2452
  * freq_hint=2452
  * SSID - hexdump_ascii(len=13):
     79 6f 75 4d 75 73 69 63 53 75 63 6b 73            youMusicSucks
  * IEs - hexdump(len=22): 30 14 01 00 00 0f ac 04 01 00 00 0f ac 04 01 00 00 0f ac 02 00 00
  * WPA Versions 0x2
  * pairwise=0xfac04
  * group=0xfac04
  * akm=0xfac02
  * htcaps - hexdump(len=26): 63 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  * htcaps_mask - hexdump(len=26): 63 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  * vhtcaps - hexdump(len=12): 00 00 00 00 00 00 00 00 00 00 00 00
  * vhtcaps_mask - hexdump(len=12): 00 00 00 00 00 00 00 00 00 00 00 00
  * Auth Type 0
nl80211: Connect request send successfully
wlan0: Setting authentication timeout: 10 sec 0 usec
EAPOL: External notification - EAP success=0
EAPOL: External notification - EAP fail=0
EAPOL: External notification - portControl=Auto
```

neophob, 2017-01-21:
and this is the wpa_supplicant log when i try to connect an open network:

```
Limit connection to BSSID 00:23:df:f8:b7:51 freq=2452 MHz based on scan results (bssid_set=0)
wlan0: set_disable_max_amsdu: -1
wlan0: set_ampdu_factor: -1
wlan0: set_ampdu_density: -1
wlan0: set_disable_ht40: 0
wlan0: set_disable_sgi: 0
wlan0: set_disable_ldpc: 0
wlan0: Determining shared radio frequencies (max len 1)
wlan0: Shared frequencies (len=0): completed iteration
nl80211: Set mode ifindex 8 iftype 2 (STATION)
nl80211: Unsubscribe mgmt frames handle 0x889193c1 (mode change)
nl80211: Subscribe to mgmt frames with non-AP handle 0x191b48
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=040a
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=040b
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=040c
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=040d
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=090a
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=090b
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=090c
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=090d
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=0409506f9a09
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=7f506f9a09
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=0801
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=06
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=0a07
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=0a11
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=1101
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=1102
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=0505
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x191b48 match=0500
nl80211: Connect (ifindex=8)
  * bssid=00:23:df:f8:b7:51
  * bssid_hint=00:23:df:f8:b7:51
  * freq=2452
  * freq_hint=2452
  * SSID - hexdump_ascii(len=13):
     79 6f 75 4d 75 73 69 63 53 75 63 6b 73            youMusicSucks
  * IEs - hexdump(len=0):
  * htcaps - hexdump(len=26): 63 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  * htcaps_mask - hexdump(len=26): 63 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  * vhtcaps - hexdump(len=12): 00 00 00 00 00 00 00 00 00 00 00 00
  * vhtcaps_mask - hexdump(len=12): 00 00 00 00 00 00 00 00 00 00 00 00
  * Auth Type 0
nl80211: Connect request send successfully
wlan0: Setting authentication timeout: 10 sec 0 usec
EAPOL: External notification - EAP success=0
EAPOL: External notification - EAP fail=0
EAPOL: External notification - portControl=ForceAuthorized
wlan0: Control interface command 'STATUS'
EAPOL: EAP Session-Id not available
CTRL-DEBUG: ctrl_sock-sendto: sock=12 sndbuf=163840 outq=0 send_len=127
wlan0: Control interface command 'STATUS'
EAPOL: EAP Session-Id not available
CTRL-DEBUG: ctrl_sock-sendto: sock=12 sndbuf=163840 outq=0 send_len=127
wlan0: Control interface command 'STATUS'
EAPOL: EAP Session-Id not available
CTRL-DEBUG: ctrl_sock-sendto: sock=12 sndbuf=163840 outq=0 send_len=127
wlan0: Control interface command 'STATUS'
EAPOL: EAP Session-Id not available
CTRL-DEBUG: ctrl_sock-sendto: sock=12 sndbuf=163840 outq=0 send_len=127
wlan0: Control interface command 'STATUS'
EAPOL: EAP Session-Id not available
CTRL-DEBUG: ctrl_sock-sendto: sock=12 sndbuf=163840 outq=0 send_len=127
wlan0: Control interface command 'STATUS'
EAPOL: EAP Session-Id not available
CTRL-DEBUG: ctrl_sock-sendto: sock=12 sndbuf=163840 outq=0 send_len=127
wlan0: Authentication with 00:23:df:f8:b7:51 timed out.
Added BSSID 00:23:df:f8:b7:51 into blacklist
TDLS: Remove peers on disassociation
wlan0: WPA: Clear old PMK and PTK
wlan0: Request to deauthenticate - bssid=00:00:00:00:00:00 pending_bssid=00:23:df:f8:b7:51 reason=3 state=ASSOCIATING
TDLS: Tear down peers
wpa_driver_nl80211_disconnect(reason_code=3)
wlan0: Event DEAUTH (12) received
wlan0: Deauthentication notification
wlan0:  * reason 3 (locally generated)
Deauthentication frame IE(s) - hexdump(len=0): [NULL]
wlan0: CTRL-EVENT-DISCONNECTED bssid=00:23:df:f8:b7:51 reason=3 locally_generated=1
```