lwfinger/rtl8723bu #33: RTL871X: ERROR rtw_cmd_thread: DriverStopped(0) SurpriseRemoved(1) break at line 452

howardqiao, 2016-08-14:
Could you give me some advices?
this module that with a usb hub cannot work on raspberry pi ,but can work on X86 Linux PC.
https://wholesaler.alibaba.com/product-detail/Sale-Hot-Realtek-chip-8723bu-Combo_60498416094.html?spm=a2700.7724857.0.0.KfZu8a
`
pi@raspberrypi:~ $ dmesg | grep rtl
[    7.867635] Bluetooth: hci0: rtl: examining hci_ver=06 hci_rev=000b lmp_ver=06 lmp_subver=8723
[    7.867683] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
[    9.414529] RTL871X: rtl8723bu v4.3.6.11_12942.20141204_BTCOEX20140507-4E40
[    9.414571] RTL871X: rtl8723bu BT-Coex version = BTCOEX20140507-4E40
[    9.649064] usbcore: registered new interface driver rtl8723bu
pi@raspberrypi:~ $ dmesg | grep RTL
[    9.414529] RTL871X: rtl8723bu v4.3.6.11_12942.20141204_BTCOEX20140507-4E40
[    9.414571] RTL871X: rtl8723bu BT-Coex version = BTCOEX20140507-4E40
[   13.182269] RTL871X: ERROR rtw_cmd_thread: DriverStopped(0) SurpriseRemoved(1) break at line 452

`

howardqiao, 2016-08-14:
after disable line 449-454 and line 477-482,then set CONFIG_POWER_SAVING = n in Makefile
'/ if ((padapter->bDriverStopped == _TRUE)||(padapter->bSurpriseRemoved== _TRUE))
		// {
			// DBG_871X_LEVEL(_drv_always_, "%s: DriverStopped(%d) SurpriseRemoved(%d) break at line %d\n",
				// __FUNCTION__, padapter->bDriverStopped, padapter->bSurpriseRemoved, __LINE__);
			// break;
		// }

pi@raspberrypi:~ $ dmesg | grep RTL
[    8.514794] RTL871X: module init start
[    8.514841] RTL871X: rtl8723bu v4.3.6.11_12942.20141204_BTCOEX20140507-4E40
[    8.514853] RTL871X: rtl8723bu BT-Coex version = BTCOEX20140507-4E40
[    8.728323] RTL871X: rtw_ndev_init(wlan0)
[    8.731399] RTL871X: module init ret=0
[    9.961239] RTL871X: RTW_ADAPTIVITY_EN_ENABLE
[    9.961264] RTL871X: RTW_ADAPTIVITY_MODE_NORMAL

pi@raspberrypi:~ $ iwlist wlan0 scanning
wlan0     No scan results

'
still cannot find any singnal,and cannot start asoc

howardqiao, 2016-08-14:
after disable line 449-454 and line 477-482,but set CONFIG_POWER_SAVING = y in Makefile
`
pi@raspberrypi:~ $ dmesg | grep RTL
[    9.405998] RTL871X: module init start
[    9.406033] RTL871X: rtl8723bu v4.3.6.11_12942.20141204_BTCOEX20140507-4E40
[    9.406045] RTL871X: rtl8723bu BT-Coex version = BTCOEX20140507-4E40
[    9.576217] RTL871X: rtw_ndev_init(wlan0)
[    9.579646] RTL871X: module init ret=0
[   10.666944] RTL871X: RTW_ADAPTIVITY_EN_ENABLE
[   10.666976] RTL871X: RTW_ADAPTIVITY_MODE_NORMAL
[   13.575320] RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
[   21.581996] RTL871X: nolinked power save enter
[   23.732526] RTL871X: nolinked power save leave
[   23.790906] RTL871X: nolinked power save leave
[   23.944647] RTL871X: nolinked power save leave
[   24.000615] RTL871X: nolinked power save leave
[   25.054916] RTL871X: nolinked power save leave
[   25.110661] RTL871X: nolinked power save leave
[   26.165005] RTL871X: nolinked power save leave
[   26.220731] RTL871X: nolinked power save leave
[   31.280766] RTL871X: nolinked power save leave
[   31.339229] RTL871X: nolinked power save leave
[   36.398584] RTL871X: nolinked power save leave
[   36.457139] RTL871X: nolinked power save leave
[   46.516564] RTL871X: nolinked power save leave
[   46.575205] RTL871X: nolinked power save leave
[   56.633321] RTL871X: nolinked power save leave
[   56.689134] RTL871X: nolinked power save leave

pi@raspberrypi:~ $ iwlist wlan0 scanning | grep ESSID
                    ESSID:"1234567890"
                    ESSID:"TP-LINK_0B52"

`

By iwlist command can be scanned several signals , but still can not be connected to any signal which prompts the connection fails

lwfinger, 2016-08-15:
Power?

howardqiao, 2016-08-15:
I use with a separate power supply USB HUB, still have the same problem.

Osndok, 2017-12-04:
I started to get this message after a recent fedora update, but in my case, the wireless chip is integrated into my x86 tablet.

I've always had an issue with this hardware/driver stopping after long-term use, but I don't recall ever seeing this message before (maybe I've never looked at dmesg during such an outage?).

Usually, just restarting NetworkManager is enough, but in this case, I found that unloading/reloading the module (along with restarting NetworkManager) seems to have worked around the issue, so I have tweaked the script and we'll see if it holds up long-term!