lwfinger/rtl8723bu #34: Can't unplug the wireless adaptor

WeiqiChenCN, 2016-08-28:
Hello,

Thank you for developing and maintaining the driver of the USB wireless adaptor.

I tried my wireless adaptor with raspberry pi 2b (Ubuntu 16.10 armhf installed)
And of course, I can use it as AP and connect other wireless network.

But I found it's strange after I un-pluged it. I can still see the interface from `ifconfig`.

And I tried my wireless adaptor with my PC (Ubuntu 16.04 amd64 installed) And I can
reproduce this problem too.

I use the newest code to compile the driver module, 

And here is my other information:

Raspberry pi:
```
$ uname -a
Linux RaspberryPi-2 4.6.1-v7+ #6 SMP Sat Aug 27 16:57:43 CST 2016 armv7l armv7l armv7l GNU/Linux
$lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu Yakkety Yak (development branch)
Release:	16.10
Codename:	yakkety
```

PC:
```
$ uname -a
Linux B85i-Ubuntu 4.4.0-34-generic #53-Ubuntu SMP Wed Jul 27 16:06:39 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
$ lsb_release -a
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu 16.04.1 LTS
Release:	16.04
Codename:	xenial
```

On my Raspberry pi
----

Here is part of `dmesg` output:
```
[  848.043406] RTL871X: rtw_ndev_init(wlan0)
[  848.045656] usbcore: registered new interface driver rtl8723bu
[  848.045677] RTL871X: module init ret=0
[  848.079239] rtl8723bu 1-1.4:1.2 wlx40a5ef0ea956: renamed from wlan0
[  848.951553] RTL871X: RTW_ADAPTIVITY_EN_AUTO, chplan:0x20, Regulation:0,0
[  848.951589] RTL871X: RTW_ADAPTIVITY_MODE_NORMAL
[  849.538021] IPv6: ADDRCONF(NETDEV_UP): wlx40a5ef0ea956: link is not ready
[  851.532426] RTL871X: nolinked power save enter
[  854.776184] usb 1-1.4: USB disconnect, device number 5
[ 1080.610323] INFO: task kworker/0:1:36 blocked for more than 120 seconds.
[ 1080.617266]       Tainted: G           O    4.6.1-v7+ #6
[ 1080.622785] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[ 1080.630916] kworker/0:1     D 806d04cc     0    36      2 0x00000000
[ 1080.630966] Workqueue: usb_hub_wq hub_event
[ 1080.631007] [<806d04cc>] (__schedule) from [<806d0a4c>] (schedule+0x50/0xa8)
[ 1080.631214] [<806d0a4c>] (schedule) from [<7f1bcb74>] (wiphy_unregister+0x1e4/0x31c [cfg80211])
[ 1080.631641] [<7f1bcb74>] (wiphy_unregister [cfg80211]) from [<7f2d751c>] (rtw_wdev_unregister+0x54/0x58 [8723bu])
[ 1080.631975] [<7f2d751c>] (rtw_wdev_unregister [8723bu]) from [<7f2c3f70>] (rtw_unregister_netdevs+0x48/0x74 [8723bu])
[ 1080.632294] [<7f2c3f70>] (rtw_unregister_netdevs [8723bu]) from [<7f2c5524>] (rtw_dev_remove+0x2c/0x84 [8723bu])
[ 1080.632468] [<7f2c5524>] (rtw_dev_remove [8723bu]) from [<8052ddb4>] (usb_unbind_interface+0x8c/0x298)
[ 1080.632498] [<8052ddb4>] (usb_unbind_interface) from [<804bb828>] (__device_release_driver+0x98/0x12c)
[ 1080.632525] [<804bb828>] (__device_release_driver) from [<804bb8e8>] (device_release_driver+0x2c/0x38)
[ 1080.632551] [<804bb8e8>] (device_release_driver) from [<804bafd8>] (bus_remove_device+0xe8/0x114)
[ 1080.632575] [<804bafd8>] (bus_remove_device) from [<804b7c34>] (device_del+0x11c/0x220)
[ 1080.632598] [<804b7c34>] (device_del) from [<8052b3d8>] (usb_disable_device+0xb0/0x1f4)
[... 416 lines cut ...]
          RX packets:1514 errors:0 dropped:0 overruns:0 frame:0
          TX packets:1514 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:120702 (120.7 KB)  TX bytes:120702 (120.7 KB)

wlp3s0    Link encap:Ethernet  HWaddr 80:19:34:55:a8:48  
          inet addr:192.168.0.100  Bcast:192.168.0.255  Mask:255.255.255.0
          inet6 addr: fe80::9d58:aa6:315b:c1/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:49923 errors:0 dropped:0 overruns:0 frame:0
          TX packets:39878 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:24758761 (24.7 MB)  TX bytes:6822221 (6.8 MB)

wlx40a5ef0ea956 Link encap:Ethernet  HWaddr 40:a5:ef:0e:a9:56  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:5 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```

And It sill here.

And I can't run `lsusb` because it stucked and never return.

WeiqiChenCN, 2016-09-22:
I found another information:

If I unplug the usb wireless adaptor , I can still see the `wlan0` device .
And It will not "disappear" until I `ifconfig wlan0 down` it.

 ( Actually， it's name is `wlx40a5ef0ea956`,  But I used `wlan0` here. )

I add more debug printk info kernel code:
```c
    struct cfg80211_registered_device *rdev = wiphy_to_rdev(wiphy);
    printk(KERN_DEBUG"%s: Before wait_event.\n", __func__);
    wait_event(rdev->dev_wait, ({  
        int __count;
        rtnl_lock();
        __count = rdev->opencount;
        rtnl_unlock();
        __count == 0; }));
    printk(KERN_DEBUG"%s: After wait_event.\n", __func__);

    if (rdev->rfkill)
        rfkill_unregister(rdev->rfkill);

    rtnl_lock();
    nl80211_notify_wiphy(rdev, NL80211_CMD_DEL_WIPHY);
    rdev->wiphy.registered = false;

    WARN_ON(!list_empty(&rdev->wdev_list));

```

And Here is my kernel output:

You can find another issue here, 
 `WARN_ON(!list_empty(&rdev->wdev_list))`  was triggered too.
```
[   45.810164] usb 1-1.2: new high-speed USB device number 4 using dwc_otg
[   45.931544] usb 1-1.2: New USB device found, idVendor=0bda, idProduct=b720
[   45.938661] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   45.946281] usb 1-1.2: Product: 802.11n WLAN Adapter
[   45.951493] usb 1-1.2: Manufacturer: Realtek
[   45.955934] usb 1-1.2: SerialNumber: 00e04c000001
[   46.041035] Bluetooth: Core ver 2.21
[   46.046129] NET: Registered protocol family 31
[   46.050885] Bluetooth: HCI device and connection manager initialized
[   46.060329] Bluetooth: HCI socket layer initialized
[   46.065460] Bluetooth: L2CAP socket layer initialized
[   46.072831] Bluetooth: SCO socket layer initialized
[   46.111516] usbcore: registered new interface driver btusb
[   46.124081] Bluetooth: hci0: rtl: examining hci_ver=06 hci_rev=000b lmp_ver=06 lmp_subver=8723
[   46.133178] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
[   46.180666] Bluetooth: hci0: rom_version status=0 version=1
[   47.134102] RTL871X: module init start
[   47.138045] RTL871X: rtl8723bu v4.3.6.11_12942.20141204_BTCOEX20140507-4E40
[   47.145334] RTL871X: rtl8723bu BT-Coex version = BTCOEX20140507-4E40
[   47.327179] RTL871X: rtw_ndev_init(wlan0)
[   47.335274] usbcore: registered new interface driver rtl8723bu
[   47.341433] RTL871X: module init ret=0
[   47.381483] rtl8723bu 1-1.2:1.2 wlx40a5ef0ea956: renamed from wlan0
[   48.987896] IPv6: ADDRCONF(NETDEV_UP): wlx40a5ef0ea956: link is not ready
[   50.982217] RTL871X: nolinked power save enter
[   67.811953] usb 1-1.2: USB disconnect, device number 4
[   67.826015] wiphy_unregister: Before wait_event.          ##If I don't run 'ifconfig down', the log output will stuck at here.
[   76.662864] RTL871X: nolinked power save leave
[   76.671136] wiphy_unregister: After wait_event.
[   76.681550] ------------[ cut here ]------------
[   76.690348] WARNING: CPU: 0 PID: 36 at net/wireless/core.c:796 wiphy_unregister+0x340/0x344 [cfg80211]
[   76.704279] CPU: 0 PID: 36 Comm: kworker/0:1 Tainted: G           O    4.6.1-v7+ #7
[   76.715545] Hardware name: BCM2709
[   76.722781] [<80110b04>] (unwind_backtrace) from [<8010c43c>] (show_stack+0x20/0x24)
[   76.734494] [<8010c43c>] (show_stack) from [<80432eec>] (dump_stack+0xc8/0x10c)
[... 75 lines cut ...]
[  240.871224] [<801363b4>] (worker_thread) from [<8013c3f8>] (kthread+0xec/0x104)
[  240.883375] [<8013c3f8>] (kthread) from [<80107f68>] (ret_from_fork+0x14/0x2c)
[  360.890366] INFO: task kworker/0:3:737 blocked for more than 120 seconds.
[  360.902080]       Tainted: G           O    4.6.1-v7+ #7
[  360.912250] "echo 0 > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
[  360.925012] kworker/0:3     D 806d04cc     0   737      2 0x00000000
[  360.936265] Workqueue: usb_hub_wq hub_event
[  360.945331] [<806d04cc>] (__schedule) from [<806d0a4c>] (schedule+0x50/0xa8)
[  360.957413] [<806d0a4c>] (schedule) from [<7f1aeb9c>] (wiphy_unregister+0x20c/0x344 [cfg80211])
[  360.971493] [<7f1aeb9c>] (wiphy_unregister [cfg80211]) from [<7f2c951c>] (rtw_wdev_unregister+0x54/0x)
[  360.991670] [<7f2c951c>] (rtw_wdev_unregister [8723bu]) from [<7f2b5f70>] (rtw_unregister_netdevs+0x4)
[  361.012268] [<7f2b5f70>] (rtw_unregister_netdevs [8723bu]) from [<7f2b7524>] (rtw_dev_remove+0x2c/0x8)
[  361.032296] [<7f2b7524>] (rtw_dev_remove [8723bu]) from [<8052ddb4>] (usb_unbind_interface+0x8c/0x298)
[  361.046662] [<8052ddb4>] (usb_unbind_interface) from [<804bb828>] (__device_release_driver+0x98/0x12c)
[  361.060964] [<804bb828>] (__device_release_driver) from [<804bb8e8>] (device_release_driver+0x2c/0x38)
[  361.075209] [<804bb8e8>] (device_release_driver) from [<804bafd8>] (bus_remove_device+0xe8/0x114)
[  361.088993] [<804bafd8>] (bus_remove_device) from [<804b7c34>] (device_del+0x11c/0x220)
[  361.102000] [<804b7c34>] (device_del) from [<8052b3d8>] (usb_disable_device+0xb0/0x1f4)
[  361.114984] [<8052b3d8>] (usb_disable_device) from [<80522634>] (usb_disconnect+0x84/0x238)
[  361.128344] [<80522634>] (usb_disconnect) from [<80524164>] (hub_event+0x60c/0x10d0)
[  361.140990] [<80524164>] (hub_event) from [<80135fbc>] (process_one_work+0x170/0x508)
[  361.153790] [<80135fbc>] (process_one_work) from [<801363b4>] (worker_thread+0x60/0x5c8)
[  361.166859] [<801363b4>] (worker_thread) from [<8013c3f8>] (kthread+0xec/0x104)
[  361.179157] [<8013c3f8>] (kthread) from [<80107f68>] (ret_from_fork+0x14/0x2c)
```

WeiqiChenCN, 2016-09-24:
I found a solution !!

```patch
diff --git a/os_dep/os_intfs.c b/os_dep/os_intfs.c
index ddcc66f..a77d4bb 100644
--- a/os_dep/os_intfs.c
+++ b/os_dep/os_intfs.c
@@ -829,14 +829,15 @@ void rtw_unregister_netdevs(struct dvobj_priv *dvobj)
 
                pnetdev = padapter->pnetdev;
 
-#ifdef CONFIG_IOCTL_CFG80211
-               rtw_wdev_unregister(padapter->rtw_wdev);
-#endif
-
                if((padapter->DriverState != DRIVER_DISAPPEAR) && pnetdev) {
 
                        unregister_netdev(pnetdev); //will call netdev_close()
                }
+
+#ifdef CONFIG_IOCTL_CFG80211
+               rtw_wdev_unregister(padapter->rtw_wdev);
+#endif
+
        }
 }
```

Should call `unregister_netdev()` first remove  the network device, then call `wiphy_unregister()` remove the wireless stuff.

I don't know whether  I did it right.

Now, I can unplug or `rmmod` it smoothly, It works well.

lwfinger, 2016-09-24:
That solution looks good to me, and it works here. Thanks. I just pushed the fix to the repo.

closed 2016-09-26 by WeiqiChenCN