lwfinger/rtl8723bu #35: error compiling with CONFIG_CONCURRENT_MODE enabled

mpvader, 2016-09-16:
Hi,

When compiling with the CONFIG_CONCURRENT_MODE define enabled in autoconf.h, I get an error during compilation. See below.

My goal is to work in concurrent mode (AP+STA). And I am following this nice post:
http://randomstuffidosometimes.blogspot.nl/2016/03/rtl8192cu-and-rtl8188cus-in-station-and.html

any ideas? 

best, Matthijs

```
| make[1]: Entering directory '/media/venusbuilds/wipbbbjethro/build/tmp-glibc/work-shared/beaglebone/kernel-source'
| make[2]: Entering directory '/media/venusbuilds/wipbbbjethro/build/tmp-glibc/work-shared/beaglebone/kernel-build-artifacts'
|   CC [M]  /media/venusbuilds/wipbbbjethro/build/tmp-glibc/work/beaglebone-ve-linux-gnueabi/rtl8723bu/0.0-r0/git/core/rtw_cmd.o
|   CC [M]  /media/venusbuilds/wipbbbjethro/build/tmp-glibc/work/beaglebone-ve-linux-gnueabi/rtl8723bu/0.0-r0/git/core/rtw_security.o
|   CC [M]  /media/venusbuilds/wipbbbjethro/build/tmp-glibc/work/beaglebone-ve-linux-gnueabi/rtl8723bu/0.0-r0/git/core/rtw_debug.o
|   CC [M]  /media/venusbuilds/wipbbbjethro/build/tmp-glibc/work/beaglebone-ve-linux-gnueabi/rtl8723bu/0.0-r0/git/core/rtw_io.o
|   CC [M]  /media/venusbuilds/wipbbbjethro/build/tmp-glibc/work/beaglebone-ve-linux-gnueabi/rtl8723bu/0.0-r0/git/core/rtw_ioctl_query.o
|   CC [M]  /media/venusbuilds/wipbbbjethro/build/tmp-glibc/work/beaglebone-ve-linux-gnueabi/rtl8723bu/0.0-r0/git/core/rtw_ioctl_set.o
|   CC [M]  /media/venusbuilds/wipbbbjethro/build/tmp-glibc/work/beaglebone-ve-linux-gnueabi/rtl8723bu/0.0-r0/git/core/rtw_ieee80211.o
|   CC [M]  /media/venusbuilds/wipbbbjethro/build/tmp-glibc/work/beaglebone-ve-linux-gnueabi/rtl8723bu/0.0-r0/git/core/rtw_mlme.o
|   CC [M]  /media/venusbuilds/wipbbbjethro/build/tmp-glibc/work/beaglebone-ve-linux-gnueabi/rtl8723bu/0.0-r0/git/core/rtw_mlme_ext.o
| /media/venusbuilds/wipbbbjethro/build/tmp-glibc/work/beaglebone-ve-linux-gnueabi/rtl8723bu/0.0-r0/git/core/rtw_mlme_ext.c: In function 'site_survey':
| /media/venusbuilds/wipbbbjethro/build/tmp-glibc/work/beaglebone-ve-linux-gnueabi/rtl8723bu/0.0-r0/git/core/rtw_mlme_ext.c:9381:3: error: 'else' without a previous 'if'
|    else if(padapter->pbuddy_adapter
|    ^
| /media/venusbuilds/wipbbbjethro/build/tmp-glibc/work-shared/beaglebone/kernel-source/scripts/Makefile.build:258: recipe for target '/media/venusbuilds/wipbbbjethro/build/tmp-glibc/work/beaglebone-ve-linux-gnueabi/rtl8723bu/0.0-r0/git/core/rtw_mlme_ext.o' failed
```

lwfinger, 2016-09-16:
It seems obvious that the "else" in the statement that errors should not be there. Fix pushed.

mpvader, 2016-09-21:
@lwfinger yes, that fixed the one issue. After hitting the compile button again I found some more. And fixed them this time:

1. [0003-concurrent_mode-fix-ioctl_cfg80211.c-compile-errors.patch](https://raw.githubusercontent.com/victronenergy/meta-victronenergy/WIP-bbe-rtl8723bu-concurrent/meta-bsp/recipes-kernel/rtl8723bu/files/0003-concurrent_mode-fix-ioctl_cfg80211.c-compile-errors.patch)
2. [0004-concurrent_mode-fix-rtw_p2p.c-compile-errors.patch](https://raw.githubusercontent.com/victronenergy/meta-victronenergy/WIP-bbe-rtl8723bu-concurrent/meta-bsp/recipes-kernel/rtl8723bu/files/0004-concurrent_mode-fix-rtw_p2p.c-compile-errors.patch)

Let me know if you need/want those patches in a pull request or something like that.

After doing those patches, the whole thing compiled and worked: I used hostapd to make an accesspoint, and then I could connect to it. Note that I only did some very basic testing, I'll do more later.

More in general: is there any information somewhere about the status of this concurrent mode feature? And/or the used rtl driver source code in general?

Its obviously a not very often (never?) used feature. And my plan was to use it as a feature in our product. Not just something nice in my home. Which means it needs to work real well, or not all.

lwfinger, 2016-09-21:
I have incorporated those patches into the repo and will push them soon. Thanks.

As you note, this feature has never been used, at least recently. How well it works will require testing. Only you will be able to see if it works well enough for your product.

What kernel version were you using when you got the compile warnings for cfg80211_rtw_add_virtual_intf? I do not see that for kernel 4.7, which is what I am running now.

mpvader, 2016-09-22:
Thanks. Kernel 4.1.15, same as in the reboot issue thread.

lwfinger, 2016-09-22:
Strange. That API has not changed since kernel 3.7.0.

In include/net/cfg80211.h, that prototype is defined as:

        struct wireless_dev * (*add_virtual_intf)(struct wiphy *wiphy,
                                                  const char *name,
                                                  unsigned char name_assign_type,
                                                  enum nl80211_iftype type,
                                                  u32 *flags,
                                                  struct vif_params *params);

What do you see in that spot?

lwfinger, 2016-09-22:
I just committed changes that turn on CONFIG_CONCURRENT_MODE for all builds, and one that fixes your unload problem.

mpvader, 2016-09-22:
> I just committed changes that turn on CONFIG_CONCURRENT_MODE for all builds

super, thanks.

> and one that fixes your unload problem.

Yes, tested it, and can confirm it unloads, and therefore reboots now!

closed 2016-09-26 by mpvader