lwfinger/rtl8723bu #36: stuck while rebooting when driver is loaded?

mpvader, 2016-09-16:
Hi, 

with the rtl8723bu driver loaded, my device won't reboot anymore. Its 100% reproducible. And solved by booting without loading the driver.

system log output is:
```
root@beaglebone:~# reboot

Broadcast message from root@beaglebone (ttyO0) (Fri Sep 16 00:00:49 2016):
The system is going down for reboot NOW!
INIT: Sending processes the TERM signal
root@beaglebone:~#
INIT: Sending processes the KILL signal
Stopping simple-upnpd
Stopping system message bus
hwclock: can't open '/dev/misc/rtc': No such file or directory
Stopping syslogd/klogd: stopped syslogd (pid 1842)
stopped klogd (pid 1843)
done
Stopping Connection Manager
[  325.401405] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
Stopping crond: OK
Gracefully shutting down php-fpm . done
Stopping Hiawatha Web Server: stopped /usr/sbin/hiawatha (pid 1967)
hiawatha.
Deconfiguring network interfaces... done.
Sending all processes the TERM signal...
[  328.963540] RTL871X: nolinked power save enter
Sending all processes the KILL signal...
Unmounting remote filesystems...
Deactivating swap...
Unmounting local filesystems...
[  333.493695] EXT4-fs (mmcblk1p3): re-mounted. Opts: data=ordered
Rebooting... [  335.629452] musb-hdrc musb-hdrc.1.auto: remove, state 1
[  335.635012] usb usb1: USB disconnect, device number 1
[  335.640312] usb 1-1: USB disconnect, device number 2
[  335.645536] usb 1-1.4: USB disconnect, device number 3
```

the used device is a beaglebone enhanced from Sancloud, running our own image called Venus. So not the IoT beaglebone image. I am still awaiting feedback to know if this same issue happens on their beaglebone IoT image, as I know they use this same driver.

Any ideas? Is this a known issue?

Matthijs

ps. I have this issue (also) when not working with the concurrent_mode. About which I made that other issue for last night, #35. So thought it best to keep these two issues seperate. Also I have some more compile errors with the concurrent_mode. And fixes for them this time ;-). I'll post that in #35, after doing some more tests.

sancloud, 2016-09-17:
The board hangs when rebooting with the IoT 8.4 image, don't know exactly what is causing it yet:
root@beaglebone:~# reboot
         Starting Synchronise Hardware Clock to System Clock...
[  OK  ] Stopped target System Time Synchronized.
[  OK  ] Stopped Update UTMP about System Runlevel Changes.
[  OK  ] Stopped target Graphical Interface.
         Stopping Entropy daemon using the HAVEGE algorithm...
[  OK  ] Stopped target Multi-User System.
         Stopping System Logging Service...
         Stopping Bonescript autorun...
         Stopping Regular background program processing daemon...
         Stopping OpenBSD Secure Shell server...
         Stopping Avahi mDNS/DNS-SD Stack...
         Stopping Login Service...
[  OK  ] Stopped target Login Prompts.
         Stopping Serial Getty on ttyGS0...
         Stopping Getty on tty1...
         Stopping Serial Getty on ttyS0..[  OK  ] Stopped LSB: Advanced IEEE 802.11 management daemon.
         Stopping LSB: set CPUFreq kernel parameters...
[  OK  ] Stopped LSB: Apache2 web server.
[  OK  ] Stopped target Network is Online.
         Stopping Wait for network to be configured by ConnMan...
         Stopping LSB: Start busybox udhcpd at boot time...
[  OK  ] Stopped target Bluetooth.
         Stopping Bluetooth service...
         Starting Store Sound Card State...
[  OK  ] Stopped Entropy daemon using the HAVEGE algorithm.
[  OK  ] Stopped Bonescript autorun.
[  OK  ] Stopped Regular background program processing daemon.
[  OK  ] Stopped Login Service.
[  OK  ] Stopped Avahi mDNS/DNS-SD Stack.
[  OK  ] Stopped System Logging Service.
[  OK  ] Stopped Wait for network to be configured by ConnMan.
[  OK  ] Stopped Getty on tty1.
[  OK  ] Stopped Serial Getty on ttyS0.
[  OK  ] Stopped Bluetooth service.
[  OK  ] Stopped OpenBSD Secure Shell server.
[  OK  ] Stopped Serial Getty on ttyGS0.
[  OK  ] Stopped LSB: set CPUFreq kernel parameters.
[  OK  ] Stopped LSB: Start busybox udhcpd at boot time.
[FAILED] Failed to start Store Sound Card State.
See 'systemctl status alsa-store.service' for details.
         Stopping LSB: Load kernel modules needed to enable cpufreq scaling...
[  OK  ] Removed slice system-serial\x2dgetty.slice.
[  OK  ] Removed slice system-getty.slice.
         Stopping /etc/rc.local Compatibility...
[  OK  ] Stopped /etc/rc.local Compatibility.
[  OK  ] Stopped target Network.
         Stopping WPA supplicant...
         Stopping Permit User Sessions...
         Stopping Connection service...
[  OK  ] Started Synchronise Hardware Clock to System Clock.
[  OK  ] Stopped WPA supplicant.
[  OK  ] Stopped LSB: Load kernel modules needed to enable cpufreq scaling.
[  OK  ] Stopped Permit User Sessions.
[  OK  ] Stopped Connection service.
         Stopping D-Bus System Message Bus...
[  OK  ] Stopped target Remote File Systems.
[  OK  ] Stopped target Remote File Systems (Pre).
[  OK  ] Stopped D-Bus System Message Bus.
[  OK  ] Stopped target Basic System.
[  OK  ] Stopped target Slices.
[  OK  ] Removed slice User and Session Slice.
[  OK  ] Stopped target Paths.
[  OK  ] Stopped target Timers.
[  OK  ] Stopped target Sockets.
[  OK  ] Closed node-red.socket.
[  OK  ] Closed cloud9.socket.
[  OK  ] Closed bonescript.socket.
[  OK  ] Closed Syslog Socket.
[  OK  ] Closed Avahi mDNS/DNS-SD Stack Activation Socket.
[  OK  ] Closed D-Bus System Message Bus Socket.
[  OK  ] Stopped target System Initialization.
         Stopping Load/Save RF Kill Switch Status of rfkill1...
         Stopping Load/Save RF Kill Switch Status of rfkill0...
         Stopping Network Time Synchronization...
         Stopping Apply Kernel Variables...
[  OK  ] Stopped Apply Kernel Variables.
         Stopping Load Kernel Modules...
[  OK  ] Stopped Load Kernel Modules.
         Stopping Update UTMP about System Boot/Shutdown...
[  OK  ] Stopped target Encrypted Volumes.
         Stopping LSB: Raise network interfaces....
[  OK  ] Stopped target Swap.
[  OK  ] Stopped Network Time Synchronization.
[  OK  ] Stopped Load/Save RF Kill Switch Status of rfkill1.
[  OK  ] Stopped Load/Save RF Kill Switch Status of rfkill0.
[  OK  ] Stopped Update UTMP about System Boot/Shutdown.
[  OK  ] Stopped LSB: Raise network interfaces..
         Stopping Load/Save Random Seed...
[  OK  ] Removed slice system-systemd\x2drfkill.slice.
         Stopping Create Volatile Files and Directories...
[  OK  ] Stopped Create Volatile Files and Directories.
[  OK  ] Stopped target Local File Systems.
         Unmounting /sys/kernel/debug...
[  OK  ] Stopped Load/Save Random Seed.
[  OK  ] Unmounted /sys/kernel/debug.
[  OK  ] Reached target Unmount All Filesystems.
[  OK  ] Stopped target Local File Systems (Pre).
         Stopping Create Static Device Nodes in /dev...
[  OK  ] Stopped Create Static Device Nodes in /dev.
         Stopping Remount Root and Kernel File Systems...
[  OK  ] Stopped Remount Root and Kernel File Systems.
[  OK  ] Reached target Shutdown.
[  OK  ] Reached target Final Step.
         Starting Reboot...
[  101.974320] watchdog watchdog0: watchdog did not stop!

lwfinger, 2016-09-17:
Unfortunately, your logs do not show anything about what is happening. I'm not that familiar with any of the special boot operations for Beaglebone and similar devices. Is it possible to get any boot details?

RobertCNelson, 2016-09-18:
@lwfinger nothing really special, it's ti's v4.4.x branch, with a debian-ish kitchen sink config:

https://github.com/RobertCNelson/ti-linux-kernel-dev/blob/ti-linux-4.4.y/patches/defconfig

We build rtl8723bu with the default settings and package up 8723bu.ko into a deb package.

https://github.com/rcn-ee/rtl8723bu/blob/master/build.sh#L9

My SanCloud variant is at work, so i can't test this on other kernel's till monday.

Regards,

mpvader, 2016-09-19:
@lwfinger I went ahead and made a clean build of the whole system to cut & paste the bootlog for you, but now the behavior has changed (...): It does reboot now, but only after seeing a kernel deadlock message. See full log below.

If I move that .ko file out of its normal location to make sure it isn't loaded, all works fine again: no kernel deadlocks.

Note that this is a different distro than the IoT one from the Beaglebone organisation. What I am running on the Beaglebone Enhanced is our [Venus](https://github.com/victronenergy/venus/wiki) distro. Similar to @RobertCNelson, I just compile this rtl8723bu driver with its default settings and add it to the rootfs. See the ['add rtl8723bu driver for wifi on bbe' commit](https://github.com/victronenergy/meta-victronenergy/commit/e987d6a567e0a72801738bc4b6d9de7805f1e216) for details.

The used kernel sources & config is here:
https://github.com/victronenergy/linux/tree/v4.1.15-bbb5

For anyone interested, binary installer image can be downloaded here:
[venus-install-sdcard-beaglebone-20160918102415-bbe-rt8723bu.img.zip](https://www.dropbox.com/s/v191zswz0ky8tup/venus-install-sdcard-beaglebone-20160918102415-bbe-rt8723bu.img.zip?dl=1)

````
U-Boot SPL 2015.07-venus (Sep 02 2016 - 09:01:50)
reading u-boot.img
reading u-boot.img

U-Boot 2015.07-venus (Sep 02 2016 - 09:01:50 +0200)

       Watchdog enabled
I2C:   ready
DRAM:  1 GiB
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Net:   <ethaddr> not set. Validating first E-fuse MAC
cpsw, usb_ether
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
reading boot.scr
396 bytes read in 3 ms (128.9 KiB/s)
Running bootscript from mmc0 ...
## Executing script at 82000000
gpio: pin 72 (gpio 72) value is 1
switch to partitions #0, OK
mmc1(part 0) is current device
SD/MMC found on device 1
reading boot.scr
** Unable to read file boot.scr **
reading uEnv.txt
831 bytes read in 4 ms (202.1 KiB/s)
Loaded environment from uEnv.txt
Importing environment from mmc ...
Running uenvcmd ...
30731 bytes read in 27 ms (1.1 MiB/s)
Loaded bbb-venus.dtb
3659280 bytes read in 229 ms (15.2 MiB/s)
Kernel image @ 0x82000000 [ 0x000000 - 0x37d610 ]
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Loading Device Tree to 8fff5000, end 8ffff80a ... OK

Starting kernel ...

[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.1.15-venus (matthijs@matthijs-VirtualBox) (gcc version 5.2.0 (GCC) ) #1 Sun Sep 18 12:28:03 CEST 2016
[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: TI AM335x BeagleBone Black
[    0.000000] cma: Reserved 16 MiB at 0xbf000000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM335X ES2.1 (sgx neon )
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 260624
[    0.000000] Kernel command line: console=ttyO0,115200n8 consoleblank=0 root=/dev/mmcblk1p3 ro rootfstype=ext4 rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 1007408K/1048576K available (4993K kernel code, 322K rwdata, 1752K rodata, 276K init, 8172K bss, 24784K reserved, 16384K cma-reserved, 253952K highmem)
[    0.000000] Virtual kernel memory layout:
[... 371 lines cut ...]

U-Boot SPL 2015.07-venus (Sep 02 2016 - 09:01:50)
reading u-boot.img
reading u-boot.img

U-Boot 2015.07-venus (Sep 02 2016 - 09:01:50 +0200)

       Watchdog enabled
I2C:   ready
DRAM:  1 GiB
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Net:   <ethaddr> not set. Validating first E-fuse MAC
cpsw, usb_ether
Hit any key to stop autoboot:  0
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
reading boot.scr
396 bytes read in 4 ms (96.7 KiB/s)
Running bootscript from mmc0 ...
## Executing script at 82000000
gpio: pin 72 (gpio 72) value is 1
switch to partitions #0, OK
mmc1(part 0) is current device
```

RobertCNelson, 2016-09-20:
Here's some more details:

```
debian@arm:~$ dmesg | grep RTL
[    3.051249] musb-hdrc: MHDRC RTL version 2.0 
[    3.061751] musb-hdrc: MHDRC RTL version 2.0 
[   24.665566] RTL871X: module init start
[   24.665572] RTL871X: rtl8723bu v4.3.6.11_12942.20141204_BTCOEX20140507-4E40
[   24.665574] RTL871X: rtl8723bu BT-Coex version = BTCOEX20140507-4E40
[   25.543205] RTL871X: rtw_ndev_init(wlan0)
[   25.660080] RTL871X: module init ret=0
[   26.566004] RTL871X: RTW_ADAPTIVITY_EN_AUTO, chplan:0x20, Regulation:0,0
[   26.572848] RTL871X: RTW_ADAPTIVITY_MODE_NORMAL
[   30.170148] RTL871X: nolinked power save enter
[   31.047917] RTL871X: RTW_ADAPTIVITY_EN_AUTO, chplan:0x20, Regulation:0,0
[   31.054733] RTL871X: RTW_ADAPTIVITY_MODE_NORMAL
[   31.690974] RTL871X: nolinked power save leave
[   31.904252] RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
[   31.933099] RTL871X: start auth
[   31.942345] RTL871X: auth success, start assoc
[   31.978136] RTL871X: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
[   31.985125] RTL871X: assoc success
[   32.017594] RTL871X: send eapol packet
[   32.033803] RTL871X: send eapol packet
[   32.131883] RTL871X: set pairwise key camid:4, addr:2c:f7:f1:06:1d:68, kid:0, type:AES
[   32.144356] RTL871X: set group key camid:5, addr:2c:f7:f1:06:1d:68, kid:1, type:AES
```
call reboot:
```
[  OK  ] Reached target Shutdown.
[   67.463821] watchdog watchdog0: watchdog did not stop!
[   67.505056] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[   67.532646] systemd-journald[554]: Received SIGTERM from PID 1 (systemd-shutdow).
[   67.549066] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[   67.575314] systemd-shutdown[1]: Hardware watchdog 'OMAP Watchdog', version 0
[   67.584202] systemd-shutdown[1]: Unmounting file systems.
[   67.591132] systemd-shutdown[1]: Remounting '/' read-only with options 'errors=remount-ro,data=ordered'.
[   67.687322] EXT4-fs (mmcblk0p1): re-mounted. Opts: errors=remount-ro,data=ordered
[   67.700664] systemd-shutdown[1]: Remounting '/' read-only with options 'errors=remount-ro,data=ordered'.
[   67.711454] EXT4-fs (mmcblk0p1): re-mounted. Opts: errors=remount-ro,data=ordered
[   67.719198] systemd-shutdown[1]: All filesystems unmounted.
[   67.725128] systemd-shutdown[1]: Deactivating swaps.
[   67.730579] systemd-shutdown[1]: All swaps deactivated.
[   67.735951] systemd-shutdown[1]: Detaching loop devices.
[   67.744949] systemd-shutdown[1]: All loop devices detached.
[   67.750673] systemd-shutdown[1]: Detaching DM devices.
[   67.756499] systemd-shutdown[1]: All DM devices detached.
[   67.778153] systemd-shutdown[1]: Failed to read reboot parameter file: No such file or directory
[   67.787505] systemd-shutdown[1]: Rebooting.
[   67.798378] musb-hdrc musb-hdrc.1.auto: remove, state 1
[   67.803756] usb usb1: USB disconnect, device number 1
[   67.808861] usb 1-1: USB disconnect, device number 2
[   67.813870] usb 1-1.3: USB disconnect, device number 3
[   67.968936] RTL871X: rtw_cmd_thread: DriverStopped(0) SurpriseRemoved(1) break at line 452
```

(without the usb module installed)
```
[  OK  ] Reached target Shutdown.
[   57.007062] watchdog watchdog0: watchdog did not stop!
[   57.048119] systemd-shutdown[1]: Sending SIGTERM to remaining processes...
[   57.074182] systemd-journald[536]: Received SIGTERM from PID 1 (systemd-shutdow).
[   57.090878] systemd-shutdown[1]: Sending SIGKILL to remaining processes...
[   57.115555] systemd-shutdown[1]: Hardware watchdog 'OMAP Watchdog', version 0
[   57.124509] systemd-shutdown[1]: Unmounting file systems.
[   57.131352] systemd-shutdown[1]: Remounting '/' read-only with options 'errors=remount-ro,data=ordered'.
[   57.462285] EXT4-fs (mmcblk0p1): re-mounted. Opts: errors=remount-ro,data=ordered
[   57.476658] systemd-shutdown[1]: Remounting '/' read-only with options 'errors=remount-ro,data=ordered'.
[   57.487498] EXT4-fs (mmcblk0p1): re-mounted. Opts: errors=remount-ro,data=ordered
[   57.495187] systemd-shutdown[1]: All filesystems unmounted.
[   57.500934] systemd-shutdown[1]: Deactivating swaps.
[   57.506364] systemd-shutdown[1]: All swaps deactivated.
[   57.511765] systemd-shutdown[1]: Detaching loop devices.
[   57.522152] systemd-shutdown[1]: All loop devices detached.
[   57.527973] systemd-shutdown[1]: Detaching DM devices.
[   57.534136] systemd-shutdown[1]: All DM devices detached.
[   57.554420] systemd-shutdown[1]: Failed to read reboot parameter file: No such file or directory
[   57.563496] systemd-shutdown[1]: Rebooting.
[   57.573927] musb-hdrc musb-hdrc.1.auto: remove, state 4
[   57.579337] usb usb1: USB disconnect, device number 1
[   57.584453] usb 1-1: USB disconnect, device number 2
[   57.591645] musb-hdrc musb-hdrc.1.auto: USB bus 1 deregistered
[   57.614004] reboot: Restarting system

U-Boot SPL 2016.09-00001-gc6ebf08 (Sep 12 2016 - 15:30:47)
Trying to boot from MMC1
Expected Linux image is not found. Trying to start U-boot

U-Boot 2016.09-00001-gc6ebf08 (Sep 12 2016 - 15:30:47 -0500), Build: jenkins-github_Bootloader-Builder-455
```

sancloud, 2016-09-21:
I have found the system is more reliable at rebooting when it isn't associated with an AP.
If it is in client mode, it can be up and scan to identify the available AP's and it will reboot.
If its associated then we have problems.

lwfinger, 2016-09-21:
I am a little confused. On reboot, the NIC is never associated with any AP. I understand that you might have problems shutting down while associated. Is that what is happening?

lwfinger, 2016-09-21:
I just found a lock that is not needed and pushed a fix. Does this help?

mpvader, 2016-09-21:
@lwfinger wrote:
> As an aside, blacklisting a driver is a lot easier than moving it. In addition, you do not have crda installed in your system. It isn't necessary, but having it would remove all those failure messages from your log.

Yes, thanks. I took crda out a while ago, but it can go back in now.

> Did the network work is the case above?

Yes.

---

@sancloud wrote:
> I have found the system is more reliable at rebooting when it isn't associated with an AP.
If it is in client mode, it can be up and scan to identify the available AP's and it will reboot.
If its associated then we have problems.

there is a bit more nuance to it: just enabling wifi (using connman) is enough to make it get stuck.

---

@lwfinger wrote:
> I just found a lock that is not needed and pushed a fix. Does this help?

No. (After enabling wifi) it is still stuck at rebooting. Also there is a deadlock error when enabling wifi:
```
(...)
[    1.258171] usbcore: registered new interface driver asix
[    1.264022] usbcore: registered new interface driver ax88179_178a
[    1.270516] usbcore: registered new interface driver cdc_ether
[    1.276802] usbcore: registered new interface driver smsc95xx
[    1.282951] usbcore: registered new interface driver net1080
[    1.288984] usbcore: registered new interface driver cdc_subset
[    1.295313] usbcore: registered new interface driver zaurus
[    1.301352] usbcore: registered new interface driver cdc_ncm
[    1.317538] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
[    1.324479] ehci-omap: OMAP-EHCI Host Controller driver
[    1.330312] ohci_hcd: USB 1.1 'Open' Host Controller (OHCI) Driver
[    1.337034] usbcore: registered new interface driver cdc_wdm
[    1.343175] usbcore: registered new interface driver usbtest
[    1.351740] 47401300.usb-phy supply vcc not found, using dummy regulator
[    1.368241] 47401b00.usb-phy supply vcc not found, using dummy regulator
[    1.380523] musb-hdrc musb-hdrc.1.auto: MUSB HDRC host driver
[    1.389835] musb-hdrc musb-hdrc.1.auto: new USB bus registered, assigned bus number 1
[    1.399911] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002
[    1.407082] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    1.414673] usb usb1: Product: MUSB HDRC host driver
[    1.419878] usb usb1: Manufacturer: Linux 4.1.15-venus musb-hcd
[    1.426097] usb usb1: SerialNumber: musb-hdrc.1.auto
[    1.435486] hub 1-0:1.0: USB hub found
[    1.440266] hub 1-0:1.0: 1 port detected
[    1.451144] mousedev: PS/2 mouse device common for all mice
[    1.460263] i2c /dev entries driver
[    1.464095] Driver for 1-wire Dallas network protocol.
[    1.474630] omap_wdt: OMAP Watchdog Timer Rev 0x01: initial timeout 60 sec
[    1.484347] omap_hsmmc 48060000.mmc: Got CD GPIO
[    1.569828] mmc0: host does not support reading read-only switch, assuming write-enable
[    1.581739] ledtrig-cpu: registered to indicate activity on CPUs
[    1.588629] usbcore: registered new interface driver usbhid
[    1.594507] usbhid: USB HID core driver
[    1.599549] oprofile: using arm/armv7
[    1.604070] Initializing XFRM netlink socket
[    1.608877] NET: Registered protocol family 10
[    1.616091] mmc0: new high speed SDHC card at address 59b4
[    1.628839] mmcblk0: mmc0:59b4 USD   7.51 GiB
[    1.638649] sit: IPv6 over IPv4 tunneling driver
[    1.645724] NET: Registered protocol family 17
[    1.650462] NET: Registered protocol family 15
[    1.655194] can: controller area network core (rev 20120528 abi 9)
[    1.661800] NET: Registered protocol family 29
[    1.666579] can: raw protocol (rev 20120528)
[... 267 lines cut ...]
[  199.343096]
[  199.343096] stack backtrace:
[  199.347665] CPU: 0 PID: 2622 Comm: reboot Tainted: G        W  O    4.1.15-venus #1
[  199.355690] Hardware name: Generic AM33XX (Flattened Device Tree)
[  199.362100] [<c0014844>] (unwind_backtrace) from [<c0012994>] (show_stack+0x10/0x14)
[  199.370226] [<c0012994>] (show_stack) from [<c0067fe0>] (__lock_acquire+0x1454/0x1da0)
[  199.378531] [<c0067fe0>] (__lock_acquire) from [<c006922c>] (lock_acquire+0xac/0x12c)
[  199.386753] [<c006922c>] (lock_acquire) from [<c04c1f14>] (_raw_spin_lock_bh+0x44/0x54)
[  199.395238] [<c04c1f14>] (_raw_spin_lock_bh) from [<bf0faebc>] (rtw_free_xmitframe_queue+0x18/0x58 [8723bu])
[  199.405699] [<bf0faebc>] (rtw_free_xmitframe_queue [8723bu]) from [<bf0f6ab8>] (rtw_free_stainfo+0x50/0x360 [8723bu])
[  199.416976] [<bf0f6ab8>] (rtw_free_stainfo [8723bu]) from [<bf0f9994>] (free_mlme_ap_info+0x3c/0x48 [8723bu])
[  199.427533] [<bf0f9994>] (free_mlme_ap_info [8723bu]) from [<bf108dac>] (rtw_usb_if1_deinit+0x34/0xa4 [8723bu])
[  199.438275] [<bf108dac>] (rtw_usb_if1_deinit [8723bu]) from [<bf108f14>] (rtw_dev_remove+0x68/0x74 [8723bu])
[  199.448669] [<bf108f14>] (rtw_dev_remove [8723bu]) from [<c0308bd4>] (usb_unbind_interface+0x7c/0x278)
[  199.458435] [<c0308bd4>] (usb_unbind_interface) from [<c02a942c>] (__device_release_driver+0x70/0xe4)
[  199.468106] [<c02a942c>] (__device_release_driver) from [<c02a94c0>] (device_release_driver+0x20/0x2c)
[  199.477867] [<c02a94c0>] (device_release_driver) from [<c02a8f18>] (bus_remove_device+0xdc/0x108)
[  199.487172] [<c02a8f18>] (bus_remove_device) from [<c02a629c>] (device_del+0x110/0x210)
[  199.495567] [<c02a629c>] (device_del) from [<c03067c4>] (usb_disable_device+0xa0/0x1e8)
[  199.503963] [<c03067c4>] (usb_disable_device) from [<c02fe134>] (usb_disconnect+0x74/0x280)
[  199.512722] [<c02fe134>] (usb_disconnect) from [<c02fe2d0>] (usb_disconnect+0x210/0x280)
[  199.524515] musb-hdrc musb-hdrc.1.auto: USB bus 1 deregistered
[  199.531148] udc musb-hdrc.0.auto: releasing 'musb-hdrc.0.auto'
[  199.538230] reboot: Restarting system
```

mpvader, 2016-09-22:
Hi. I've just rebuilt the whole thing, using @lwfinger's latest commit `Fix unload problem With CONFIG_IOCTL_CFG80211 defined` (4a134f76430531289f11daea5ec24a86f4ff7d7f), and it now unloads and the device reboots again. thanks!

For me this issue can be closed now. And am curious if it is solved for you as well @sancloud.

sancloud, 2016-09-24:
Using the latest it seems to be stable now at rebooting.
I have tried it in VE image and iOT image from bb org and they both reboot.

closed 2016-09-26 by mpvader