lwfinger/rtl8723bu #37: Driver will not load automatically and ifconfig shows no wlan0

lkw16, 2016-09-28:
I just got a new WiFi/BT adapter.  Trying to use it on my RP2.  The adapter is using rtl8723bu chip; therefore, I am hoping this driver will work.

I have no problem to compile and install the driver.  Once restarted, the 8723bu.ko will not load automatically and I have to run `insmod 8723bu.ko` command in order for it to load.  After that, command `ifconfig` will show no wlan interface.

`lsusb` shows the following ID.
ID 7392:a611 Edimax Technology Co., Ltd

Is it the ID that is causing the problem?  My RP2 is running Raspbian kernel v4.4.21-v7+, which is the latest.  Any help would be appreciated!

Regards,
lkw

lwfinger, 2016-09-29:
No one has ever reported that device as using the RTL8723BU chip. Note that driver names are lower case, and device names are upper. When Linux walks the various buses, it will autoload a driver only when that ID is mentioned in the driver. Similarly, a device will only be picked up by a manually-loaded driver is that device's ID is enabled by the driver.

How do you know the device uses that chip?

lkw16, 2016-09-29:
I installed the Windows driver on my Windows 10 system.  The adapter Details Property shows `Inf section `value as RTL8723bu.ndi.NT.  My assumption is it is using the RTL8723BU chip; otherwise, it should be loading from other sections, right?

lwfinger, 2016-09-29:
Yes, that seems to be proof that the device should use this driver. I just pushed a change that should cause the driver to recognize it. I will have to dig out my Windows 10 laptop and examine the .inf for other devices.

lkw16, 2016-09-29:
Thanks!  I will re-compile it and report back after that.

lkw16, 2016-10-04:
Sorry for taking so long to report back.  Kind of busy lately!

It still won't work with the change.  Everything seemed to be OK.  No errors whatsoever.  After rebooted, the adapter LED was not lighted up.  `lsmod` showed 8723bu was not loaded.  So, `insmod 8723bu.ko` to load it manually.  Then, I did see the adapter LED lighted up and `lsmod` was showing the loaded module.  But `ifconfig` still didn't have wlan0 interface, just eth0 and lo only.

Did I miss anything here?  Thanks!

jamescochran, 2016-10-04:
I'm having the same issues. Someone else forked this and added stuff specific for compiling on the pi. His instructions here: https://matthewdippel.blogspot.com/2016/09/howto-getting-hubpiwi-blue-drivers.html

It didn't have the checkin done a few days ago for the Edimax so I forked it and merged Matthew's changes along with the changes here:
https://github.com/jamescochran/rtl8723bu

I still don't get wlan0 when running ifconfig though although everything appeared to make and install ok. lsusb: Bus 001 Device 004: ID 7392:a611 Edimax Technology Co., Ltd

I downloaded the official driver from edimax and extracted it and it's all under the folder [code]RTL8723BU_WiFi_linux_v4.3.16_14189.20150519_BTCOEX2015119-5844[/code] so I assume rtl8723bu is correct for this device.

mgdransfield, 2016-10-08:
I have the same device and the same issue.
It seems like the device isn't being recognised and loading the driver for some reason.
 lsusb
Bus 001 Device 004: ID 7392:a611 Edimax Technology Co., Ltd
Bus 001 Device 005: ID 0bda:8176 Realtek Semiconductor Corp. RTL8188CUS 802.11n WLAN Adapter

lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/5p, 480M
        |__ Port 1: Dev 3, If 0, Class=Vendor Specific Class, Driver=smsc95xx, 480M
        |__ Port 4: Dev 5, If 0, Class=Vendor Specific Class, Driver=rtl8192cu, 480M
        |__ Port 5: Dev 4, If 0, Class=Wireless, Driver=btusb, 480M
        |__ Port 5: Dev 4, If 1, Class=Wireless, Driver=btusb, 480M
        |__ Port 5: Dev 4, If 2, Class=Vendor Specific Class, Driver=, 480M

Note there is is no Driver for port 5 Dev 4 Vendor specific class
Port 4 picks up a driver fine (another usb wifi device)

mgdransfield, 2016-10-08:
Bearing in mind that I don't have much understanding of this. I found that the device id table seems to need updating to create a device alias for the edimax device:
static struct usb_device_id rtw_usb_id_tbl[] ={

        //*=== Realtek demoboard ===*/
        {USB_DEVICE_AND_INTERFACE_INFO(USB_VENDER_ID_REALTEK, 0xB720,0xff,0xff,0xff),.driver_info = RTL8723B}, /* 8723BU 1*1 */
        {USB_DEVICE_AND_INTERFACE_INFO(0x7392, 0xa611,0xff,0xff,0xff),.driver_info = RTL8723B}, /* 8723BU 1*1 */
        {}      /* Terminating entry */
};

Now the driver gets loaded
modinfo 8723bu | grep alias
alias:          usb:v7392pA611d*dc*dsc*dp*icFFiscFFipFFin*
alias:          usb:v0BDApB720d*dc*dsc*dp*icFFiscFFipFFin*

and I get 2 wifi interfaces which seems a little strange...
iwconfig
wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated
          Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

lo        no wireless extensions.

eth0      no wireless extensions.

wlan1     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated
          Sensitivity:0/0
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

lwfinger, 2016-10-08:
From the previous comment, it is obvious that I placed the new USB ID in the wrong table. That is now fixed.

The reason you are getting two interfaces is because CONFIG_CONCURRENT_MODE is defined. This parameter is needed if you plan to use the device as both a STA and an AP. If that is not the case, and you wish to use it only in STA mode, then find the line "EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE" in Makefile and comment it out by inserting a "#" at the beginning of the line.

jamescochran, 2016-10-10:
I was able to compile and get wifi working on a raspberry pi zero. Thanks! Now for bluetooth! I think it needs to target ARM or something to get the compile to work.

lkw16, 2016-10-10:
Thanks lwfinger for the fix!  I did get it to work with WiFi.  As jamescochran said, now is for the Bluetooth!

closed 2016-10-10 by lkw16