lwfinger/rtl8723bu #38: A single usb adaptor with two wifi ??

WeiqiChenCN, 2016-09-29:
Hello,  It's me again.

Thanks again for your programming the device driver.

I got another problem here.

I compiled the driver from the newest source code, and I  found two interface:
```
[  759.880608] usb 1-1.4: new high-speed USB device number 7 using dwc_otg
[  759.982108] usb 1-1.4: New USB device found, idVendor=0bda, idProduct=b720
[  759.982134] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  759.982151] usb 1-1.4: Product: 802.11n WLAN Adapter
[  759.982166] usb 1-1.4: Manufacturer: Realtek
[  759.982181] usb 1-1.4: SerialNumber: 00e04c000001
[  759.998857] Bluetooth: hci0: rtl: examining hci_ver=06 hci_rev=000b lmp_ver=06 lmp_subver=8723
[  759.998891] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
[  760.001510] Bluetooth: hci0: rom_version status=0 version=1
[  760.203715] RTL871X: rtw_ndev_init(wlan0)
[  760.206575] RTL871X: rtw_ndev_init(wlan1)
[  760.897178] rtl8723bu 1-1.4:1.2 wlx40a5ef0ea956: renamed from wlan0
[  761.014393] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[  761.805467] RTL871X: RTW_ADAPTIVITY_EN_AUTO, chplan:0x20, Regulation:3,3
[  761.805507] RTL871X: RTW_ADAPTIVITY_MODE_NORMAL
[  762.565995] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[  762.572329] IPv6: ADDRCONF(NETDEV_UP): wlx40a5ef0ea956: link is not ready
[  762.591467] IPv6: ADDRCONF(NETDEV_UP): wlx40a5ef0ea956: link is not ready
[  762.782852] IPv6: ADDRCONF(NETDEV_UP): wlan1: link is not ready
[  762.804952] IPv6: ADDRCONF(NETDEV_UP): wlx40a5ef0ea956: link is not ready
[  764.433991] RTL871X: nolinked power save enter
```

```bash
~$ ifconfig
enxb827ebe94cd6: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet6 fe80::ba27:ebff:fee9:4cd6  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:e9:4c:d6  txqueuelen 1000  (Ethernet)
        RX packets 3  bytes 567 (567.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 45  bytes 9366 (9.3 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 35  bytes 3026 (3.0 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 35  bytes 3026 (3.0 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan1: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 42:a5:ef:0e:a9:56  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 1  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlx40a5ef0ea956: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 40:a5:ef:0e:a9:56  txqueuelen 1000  (Ethernet)
        RX packets 1  bytes 0 (0.0 B)
        RX errors 0  dropped 97  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

```

I checkouted bc9e782 , recompiled the driver and had no this issue.  Did a bug introduced by recent commit(s)??

I don't know why, maybe my usb wifi adaptor's problems ?

It's late now, I will add more information after I'm free.

Am I the only person run into this situation?

lwfinger, 2016-09-29:
No. That is a result of enabling concurrent mode, which some people want to be able to operate as a station and an AP simultaneously. The return to the previous situation, find the line "EXTRA_CFLAGS += -DCONFIG_CONCURRENT_MODE" in Makefile and comment out that line.

WeiqiChenCN, 2016-09-30:
Oh, I comment out that line, And I can "disable" the feature. and It works well.

When I enable this feature, the wifi adaptor won't work more than few seconds.

I restart my raspberrypi , and ...
```bash
chen@RaspberryPi-2:~$ sudo ifconfig wlx40a5ef0ea956 
[sudo] password for chen: 
wlx40a5ef0ea956: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.105  netmask 255.255.255.0  broadcast 192.168.0.255
        ether 40:a5:ef:0e:a9:56  txqueuelen 1000  (Ethernet)
        RX packets 25  bytes 3779 (3.7 KB)
        RX errors 0  dropped 29  overruns 0  frame 0
        TX packets 50  bytes 7135 (7.1 KB)
        TX errors 0  dropped 1 overruns 0  carrier 0  collisions 0

chen@RaspberryPi-2:~$ sudo ifconfig wlx40a5ef0ea956 
wlx40a5ef0ea956: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.105  netmask 255.255.255.0  broadcast 192.168.0.255
        ether 40:a5:ef:0e:a9:56  txqueuelen 1000  (Ethernet)
        RX packets 25  bytes 3779 (3.7 KB)
        RX errors 0  dropped 29  overruns 0  frame 0
        TX packets 51  bytes 7209 (7.2 KB)
        TX errors 0  dropped 1 overruns 0  carrier 0  collisions 0

chen@RaspberryPi-2:~$ sudo ifconfig wlx40a5ef0ea956 
wlx40a5ef0ea956: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        ether 40:a5:ef:0e:a9:56  txqueuelen 1000  (Ethernet)
        RX packets 32  bytes 4651 (4.6 KB)
        RX errors 0  dropped 33  overruns 0  frame 0
        TX packets 55  bytes 8099 (8.0 KB)
        TX errors 0  dropped 1 overruns 0  carrier 0  collisions 0

```

just few second, I lost the connect.
My `wlan0` is managerd by `systemd-networkd`.

WeiqiChenCN, 2016-09-30:
```bash
$ uname -a
Linux RaspberryPi-2 4.6.7-v7+ #1 SMP Wed Sep 28 00:15:20 CST 2016 armv7l armv7l armv7l GNU/Linux
```

mpvader, 2016-09-30:
concurrent_mode did seem to work properly for me. I haven't extensively tested it though, I'll do so asap -hopefully in the coming days- and report back.

mpvader, 2016-10-09:
hi, an update: 

for me concurrent mode works stable. I downloaded a 1GB file, two times:

- first with only wlan0 up and connected, wlan1 down. download ok
- then wlan1 upped as well: download still works ok

All together this took 22 minutes. And I have had no interruption or connection loses during all of this.

Note that I do still see a kernel warning messages during driver load and also one during unload. I'll look further into that later; they seem harmless.

My tests are done on a [Beaglebone Enhanced](https://beagleboard.org/enhanced), running our [Venus OS](https://github.com/victronenergy/venus/wiki). Kernel 4.1.15.

willypuzzle, 2016-11-02:
With kernel 4.4.0-45-generic 64bit architeture on Ubuntu 16.04 I have the same issue, the connection is unstable with that flag enabled.

lwfinger, 2016-11-02:
If it does not work with CONCURRENT_MODE enabled, then disable it. Driver rtl8xxxu in newer kernels (4.9) is working with the RTL8723BU. It is highly unlikely that there will be any further development with this driver.

closed 2016-11-02 by lwfinger