lwfinger/rtl8723bu #40: kernel deadlock warning during driver load

mpvader, 2016-10-09:
Hi Larry,

see message at 24.146556 below. It doesn't seem to do any harm, at least, I did not find anything that doesn't work yet.

Any thoughts on below? Or should I perhaps go to connman with this?

best, Matthijs
```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.1.15-venus (matthijs@matthijs-VirtualBox) (gcc version 5.2.0 (GCC) ) #1 Sun Oct 9 21:47:33 CEST 2016
[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: TI AM335x BeagleBone Black
[    0.000000] cma: Reserved 16 MiB at 0xbf000000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM335X ES2.1 (sgx neon )
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 260624
[    0.000000] Kernel command line: console=ttyO0,115200n8 consoleblank=0 root=/dev/mmcblk1p3 rootfstype=ext4 rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 1007408K/1048576K available (4993K kernel code, 322K rwdata, 1752K rodata, 276K init, 8172K bss, 24784K reserved, 16384K cma-reserved, 253952K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xf0000000 - 0xff000000   ( 240 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xef800000   ( 760 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc069e928   (6747 kB)
[    0.000000]       .init : 0xc069f000 - 0xc06e4000   ( 276 kB)
[    0.000000]       .data : 0xc06e4000 - 0xc0734a38   ( 323 kB)
[    0.000000]        .bss : 0xc0734a38 - 0xc0f2fc18   (8173 kB)
[    0.000000] Running RCU self tests
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] IRQ: Found an INTC at 0xfa200000 (revision 5.0) with 128 interrupts
[    0.000000] OMAP clockevent source: timer2 at 24000000 Hz
[    0.000012] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
[    0.000034] clocksource timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000075] OMAP clocksource: timer1 at 24000000 Hz
[    0.000739] Console: colour dummy device 80x30
[    0.000790] Lock dependency validator: Copyright (c) 2006 Red Hat, Inc., Ingo Molnar
[    0.000798] ... MAX_LOCKDEP_SUBCLASSES:  8
[    0.000804] ... MAX_LOCK_DEPTH:          48
[    0.000810] ... MAX_LOCKDEP_KEYS:        8191
[    0.000816] ... CLASSHASH_SIZE:          4096
[    0.000822] ... MAX_LOCKDEP_ENTRIES:     32768
[    0.000828] ... MAX_LOCKDEP_CHAINS:      65536
[    0.000834] ... CHAINHASH_SIZE:          32768
[    0.000840]  memory used by lock dependency info: 5167 kB
[    0.000846]  per task-struct memory footprint: 1152 bytes
[    0.000871] Calibrating delay loop... 996.14 BogoMIPS (lpj=4980736)
[    0.078946] pid_max: default: 32768 minimum: 301
[    0.079148] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.079162] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.081389] CPU: Testing write buffer coherency: ok
[    0.082578] Setting up static identity map for 0x80008200 - 0x80008258
[    0.086560] devtmpfs: initialized
[    0.111873] VFP support v0.3: implementor 41 architecture 3 part 30 variant c rev 3
[    0.147591] omap_hwmod: tptc0 using broken dt data from edma
[    0.147993] omap_hwmod: tptc1 using broken dt data from edma
[    0.148375] omap_hwmod: tptc2 using broken dt data from edma
[    0.156163] omap_hwmod: debugss: _wait_target_disable failed
[    0.211265] clocksource jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.213701] pinctrl core: initialized pinctrl subsystem
[    0.218328] NET: Registered protocol family 16
[    0.224825] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.227542] cpuidle: using governor ladder
[    0.227579] cpuidle: using governor menu
[... 238 lines cut ...]
[   24.932718] EXT4-fs (mmcblk1p2): mounted filesystem with ordered data mode. Opts: (null)
Starting simple-upnpd

[   26.566484] RTL871X: RTW_ADAPTIVITY_EN_AUTO, chplan:0x20, Regulation:3,3
[   26.573663] RTL871X: RTW_ADAPTIVITY_MODE_NORMAL
[   27.912415] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   29.074533] RTL871X: nolinked power save enter
[   31.459176] RTL871X: RTW_ADAPTIVITY_EN_AUTO, chplan:0x20, Regulation:3,3
[   31.466352] RTL871X: RTW_ADAPTIVITY_MODE_NORMAL
[   35.041026] RTL871X: nolinked power save leave
[   35.264525] RTL871X: rtw_set_802_11_connect(wlan0)  fw_state=0x00000008
[   35.300008] c_can_platform 481cc000.can can0: setting BTR=1c05 BRPE=0000
[   35.307172] c_can_platform 481cc000.can: obtain a copy of previously claimed pinctrl
[   35.315461] IPv6: ADDRCONF(NETDEV_CHANGE): can0: link becomes ready
[   35.393186] RTL871X: start auth
[   35.398277] RTL871X: auth success, start assoc
[   35.405995] RTL871X: rtw_cfg80211_indicate_connect(wlan0) BSS not found !!
[   35.413307] RTL871X: assoc success
[   35.417140] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   36.405464] RTL871X: send eapol packet
[   36.414788] RTL871X: send eapol packet
[   36.419568] RTL871X: set pairwise key camid:4, addr:98:fc:11:ae:b5:04, kid:0, type:AES
bgscan simple: Failed to enable signal strength monitoring
[   36.456327] RTL871X: set group key camid:5, addr:98:fc:11:ae:b5:04, kid:1, type:AES
```

mpvader, 2016-10-09:
ps this is on a [Beaglebone Enhanced](https://beagleboard.org/enhanced), running [venus](https://github.com/victronenergy/venus/wiki). Kernel 4.1.15.

lwfinger, 2016-10-09:
That warning is harmless. It has been in the driver from the start, but I have been unable to fix it.

mpvader, 2016-10-09:
thank you for the quick reply. I'll stop worrying about it.