lwfinger/rtl8723bu #41: Two wifi adapters shown in network manager even if I have only one. Internet disconnects randomly

vivganes, 2016-10-12:
[image]

My computer has only one wifi card (inbuilt in the laptop).  But, I am seeing two entries as shown in the above screenshot.  Internet works only with one of the devices decided at random.  So, often I choose one device, check internet and choose the other when internet does not work.

```
vivek@vivek-laptop:~$ sudo lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 5986:065e Acer, Inc 
Bus 001 Device 002: ID 0bda:b720 Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

lwfinger, 2016-10-12:
This issue has been answered in other places in this repo. Disable CONFIG_CONCURRENT_MODE if you do not wish to use the device as both an AP and an STA.

As to the disconnects, fixing that would require more knowledge of the device's internals than I know.

You might try a 4.8 kernel and use rtl8xxxu. If it works for your device, it will be stable.

closed 2016-10-12 by lwfinger

vivganes, 2016-10-12:
Thanks @lwfinger 

I was able to disable CONFIG_CONCURRENT_MODE as suggested to [this comment](https://github.com/lwfinger/rtl8723bu/issues/38#issuecomment-250519108) 

Now I am seeing only one Wifi connection chooser.  

PS: I upgraded kernel and figured out that it broke the driver. So, downgraded to earlier kernel.  However, my internet seems stable now.

lwfinger, 2016-10-12:
What broke in the newer kernel. If there was a build error, please post it.

vivganes, 2016-10-14:
@lwfinger  There was no build error. I was able to `make` and `make install` in the 4.8 latest kernel without any error. 
But after installation and `modprobe` the wifi adapter was not detected.

Hence, I reverted to the earlier kernel and rebuilt with CONFIG_CURRENT_MODE disabled.

lwfinger, 2016-10-14:
It is not a problem with the 4.8 kernel. On my system

finger@larrylap:~/kernel> cd ../rtl8723bu/
finger@larrylap:~/rtl8723bu> make -j8
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.8.0-wl-13656-g3fb1111/build M=/home/finger/rtl8723bu  modules
make[1]: Entering directory '/home/finger/wireless-drivers-next'
 == snip ==
  LD [M]  /home/finger/rtl8723bu/8723bu.ko
make[1]: Leaving directory '/home/finger/wireless-drivers-next'
finger@larrylap:~/rtl8723bu> sudo insmod 8723bu.ko
finger@larrylap:~/rtl8723bu> iwconfig
wlan0     no wireless extensions.

lo        no wireless extensions.

eth0      no wireless extensions.

wlan5     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

wlan1     no wireless extensions.

wlan4     IEEE 802.11bgn  ESSID:"NETGEAR81"  Nickname:"<WIFI@REALTEK>"
          Mode:Managed  Frequency:2.437 GHz  Access Point: 20:E5:2A:01:F7:EA   
          Bit Rate:72.2 Mb/s   Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Power Management:off
          Link Quality=100/100  Signal level=100/100  Noise level=0/100
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

Wlan4 is the STA interface - wlan5 is the AP.

It works fine. In addition, with kernel 4.8, you can use the built-in driver rtl8xxxu to driver the RTL8723BU.

vivganes, 2016-10-14:
I just used `make` without any command line arguments.  I see that you have used ` make -j8
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.8.0-wl-13656-g3fb1111/build M=/home/finger/rtl8723bu modules`. Would that contribute to any difference?  I am not sure.

lwfinger, 2016-10-14:
The -j8 just takes advantage of the fact that I have 4 processors, thus it runs 8 jobs in parallel. The various directories are the ones for my system, yours will differ, but that will not make a difference.