lwfinger/rtl8723bu #45: Makefile Typo?

slin31, 2016-10-16:
Hi, I just tried to build and install the driver, but it failed at the "make install" step.
The error message claimed "no such file or directory" and when taking a closer look I noticed there was a dot in the filepath where there (I assume) shouldn't be a dot.
I changed the Makefile and now everything went fine.

The line in the Makefile reads:
`cp -f rtl8723b_fw.bin $(FW_DIR)./rtl8723b_fw.bin`
This was added in commit da63138 just recently.
I changed the line in my Makefile to:
`cp -f rtl8723b_fw.bin $(FW_DIR)/rtl8723b_fw.bin`

Am I right to assume this was a typo or is the error somewhere else on my end?

lwfinger, 2016-10-16:
You should always do a pull whenever you get an error. The changes in commit da63138 had a typo, but that has since been corrected.

closed 2016-10-16 by lwfinger

lwfinger, 2016-10-16:
I misread your previous post. In fact, the patch in commit 5c6e8a7 had two errors. The first was fixed in commit da63138 and you found the second. Your fix has been applied to the repo.