lwfinger/rtl8723bu #48: Build error when MP_MODE = 1

neophob, 2016-12-06:
Hi, thanks for this driver. I would like to enable the MP mode to test the wlan, however enabling the MP_MODE seems to be broken:

```
00:07:53.598 /source/drivers/staging/rtl8723bu/os_dep/mlme_linux.c: In function '_dynamic_check_timer_handlder':
00:07:53.598 /source/drivers/staging/rtl8723bu/os_dep/mlme_linux.c:96:51: error: '_adapter' has no member named 'mppriv'
00:07:53.598   if (adapter->registrypriv.mp_mode == 1 && adapter->mppriv.mp_dm ==0) //for MP ODM dynamic Tx power tracking
00:07:53.598                                                    ^
00:07:53.609 make[4]: *** [drivers/staging/rtl8723bu/os_dep/mlme_linux.o] Error 1
```

any advice?

neophob, 2016-12-06:
ok looks like the mp mode was removed completly in this commit 
https://github.com/lwfinger/rtl8723bu/commit/7b704bc0ff847f9c3c389b48e52d0ee4c9f7cba2

so i guess all mp_mode references could be removed as well

lwfinger, 2016-12-06:
Yes, MP mode is used in the factory, and that code was mostly pulled.

closed 2016-12-06 by lwfinger