lwfinger/rtl8723bu #50: Driver not loaded on Kernel 4.8.0.32

Lumumba, 2016-12-22:
Driver does not load on Kernel 4.8.0.32.
But it loads after resuming from suspend.

Tested on Linux Mint, Kubuntu and Ubuntu.

lwfinger, 2016-12-22:
That does not make any sense. If it loads on resume, it should load on boot.

Any reason why you failed to list the driver name? There are 15 of them in this repo!

First, check for any blacklist commands for mention of this mysterious driver that won't load. The command 'grep rtl /etc/modprobe.d/*' will show that info.

If you do not find any blacklist entries, then you need to prove that the driver is not loading. After boot, run the command 'lsmod | grep rtl' and post the output. After suspend and resume, repeat.

Lumumba, 2016-12-22:
Driver is rtl8723bu.
'grep rtl /etc/modprobe.d/*'  returns nothing.

http://imgur.com/a/dLWZA

thank you for helping.

lwfinger, 2016-12-22:
You are not using the driver from this repo. You need to ask your questions on the mailing list linux-wireless@vger.kernel.org.

closed 2016-12-22 by lwfinger