lwfinger/rtl8723bu #51: Connect RTL8723bu wifi + bluetooth dongle driver to Intel Galileo

udayatejwani, 2017-01-04:
Hi,
I am trying to setup wifi capability on Galileo.  I am using EDIMAX EW-7611ULB dongle.  I am using 3.8.7 Yocto.  Upon submitting the make command, I get the following error.

[image]

[image]

 
I appreciate your help.

 

Thanks,

lwfinger, 2017-01-04:
You have not installed the kernel headers. As the exact steps are distro dependent, I cannot tell you what you need to do. Check with your distro.

rumble-on, 2017-01-09:
Seems you need to check your Yocto recipe to correctly setup environment for calling make.
Additionally you _might_ modify the rtl8723bu MAKEFILE's variable initialization, e.g.
environment var KSRC is not used by yocto, afaik.

mpvader, 2017-01-09:
fyi https://github.com/victronenergy/meta-victronenergy/blob/master/meta-bsp/recipes-kernel/rtl8723bu/files/0001-fix-makefile-for-openembedded.patch