lwfinger/rtl8723bu #55: implicit declaration of function ‘rtw_resume_process’

gobo38, 2017-02-25:
Want to compile the driver at a odroid c1 and get the error : implicit declaration of function ‘rtw_resume_process’

`make ARCH=arm CROSS_COMPILE= -C /lib/modules/3.10.104/build M=/root/rtl8723bu  modules
make[1]: Entering directory '/lib/modules/3.10.104/build'
  CC [M]  /root/rtl8723bu/core/rtw_cmd.o
  CC [M]  /root/rtl8723bu/core/rtw_security.o
  CC [M]  /root/rtl8723bu/core/rtw_debug.o
  CC [M]  /root/rtl8723bu/core/rtw_io.o
  CC [M]  /root/rtl8723bu/core/rtw_ioctl_query.o
  CC [M]  /root/rtl8723bu/core/rtw_ioctl_set.o
  CC [M]  /root/rtl8723bu/core/rtw_ieee80211.o
  CC [M]  /root/rtl8723bu/core/rtw_mlme.o
  CC [M]  /root/rtl8723bu/core/rtw_mlme_ext.o
  CC [M]  /root/rtl8723bu/core/rtw_wlan_util.o
  CC [M]  /root/rtl8723bu/core/rtw_vht.o
  CC [M]  /root/rtl8723bu/core/rtw_pwrctrl.o
/rtl8723bu/core/rtw_pwrctrl.c: In function ‘rtw_late_resume’:
/rtl8723bu/core/rtw_pwrctrl.c:2127:3: error: implicit declaration of function ‘rtw_resume_process’ [-Werror=implicit-function-declaration]
`
please can you help me?

lwfinger, 2017-02-25:
Apparently no one has tried to build this under Android before. I just pushed some changes that should help. Let me know of other problems,

gobo38, 2017-02-25:
Oh sorry, but you missunderstood. I don't try to run it at Android!
I try to compile on an ODROID C1( http://www.hardkernel.com/main/products/prdt_info.php ) running Linux (DietPI) on it

lwfinger, 2017-02-25:
The code that failed is inside some #ifdef CONFIG_ANDROID* restrictions. That is what suggested Android to me. Does it work now?

gobo38, 2017-02-26:
good morning from here.. ;-)
It looks better but not perfect.
Now another error is showing

`/root/rtl8723bu/os_dep/usb_intf.c:686:12: error: static declaration of ‘rtw_resume_process’ follows non-static declaration
 static int rtw_resume_process(_adapter *padapter)
`

lwfinger, 2017-02-26:
Remote debugging is so much fun.

BTW, is your ODROID available for ssh login by me?

gobo38, 2017-02-26:
lol, I'm glad that you have fun..... ;-)

please try

ssh lwfinger@82.149.101.9
pw: remotedebug

Tell me if you need something else

lwfinger, 2017-02-26:
It now compiles, but there are missing globals. The main cause is that your kernel does not include cfg80211 in the configuration. I disabled CONFIG_IOCTL_CFG80211 and 
RTW_USE_CFG80211_STA_EVENT in Makefile, and CONFIG_IOCTL_CFG80211 in include/autoconf.h. Once I did that and fixed another code error, then the driver builds. As the device is not plugged in, I could go no further.

Without the cfg80211 functionality, the driver should work, but without the modern hostapd interface. If you wish to use it as an access point, then you will need to rebuild your kernel with CFG80211 enabled.

As the remote account details were posted in an open forum, I have changed the password on that account.

gobo38, 2017-02-26:
wow, fast as lightning.... :)
Tank you very much!
no need for the cfg80211 functionality.
i only use the odroid as a server for my 3d Printer an I don't want a 10meter network cable at my workdesk.
Will try later this evening to plug in the WiFi usb stick and configure it.
Again, Thank you very much

gobo38, 2017-02-26:
ok, module is up and running, wireless connection is enabled and working.
Perfect!

lwfinger, 2017-02-26:
Good to hear. Turnaround is much faster without needing to report results through GitHub.

I meant to say this earlier, but that odroid is pretty snappy. That was my first experience with a 64-bit ARM. With a quad-core CPU at 2 GHz, that could get a lot of work done. I might consider buying one of those for myself.

gobo38, 2017-02-26:
Yes, I love the Odroid. The only problem is that not much support is given for some special tasks.
There you are  on the better side with an Raspberry pi.
so i close down the ssh Port now.
Thank you again