lwfinger/rtl8723bu #56: /lib/modules/4.6.0-kali1-amd64/build: No such file or directory.  Stop.

davidedpg10, 2017-03-01:
I installed kali 2016.2 on the Cube i9, and I have updated it, upgraded it, and I have made sure that linux headers are installed. But every time I try to "make", the console spits out the error specified below, the folder that is missing is /build, the rest is there. If I manually create the folder it does not work either. Please let me know if you know how to correct that issue.

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.6.0-kali1-amd64/build M=/root/rtl8723bu  modules
make[1]: *** /lib/modules/4.6.0-kali1-amd64/build: No such file or directory.  Stop.
Makefile:321: recipe for target 'modules' failed
make: *** [modules] Error 2

lwfinger, 2017-03-01:
If 'ls /lib/modules/4.6.0-kali1-amd64/build' does not show a directory named include, then you have not installed the kernel headers, or they are not installed correctly.

freiSMS, 2017-11-09:
Getting the same error on a pi zero, running raspbian jessie with Kernel 4.4.21+. I did the header installation. That's what it says when ich do **sudo apt-get install raspberrypi-kernel-headers**
```
Reading package lists... Done
Building dependency tree       
Reading state information... Done
raspberrypi-kernel-headers is already the newest version.
```

freiSMS, 2017-11-09:
Well.. solved my Issue. Maybe my solution can help you too.
My problem in the beginning was that the Edimax How To Article (https://edimax.freshdesk.com/support/solutions/articles/14000047172-how-to-install-ew-7611ulb-adapter-on-raspberry-pi) told me to run **apt-get upgrade** and **dist-upgrade** before being able to build this driver. By doing this my Kernel got an upgrade too which made it incompatible to this driver.
By using **"sudo apt-mark raspberrypi-kernel"** I was able to prevent the system from booting a different kernel but got the files for the new kernel anyway.

When updating the kernel-headers with **"$ sudo apt-get install raspberrypi-kernel-headers"** the new (not used) kernel 4.9.* got the headers - not my 4.4.21+ kernel.
I fixed this by  downloading the header here (https://www.niksula.hut.fi/~mhiienka/Rpi/linux-headers-rpi/) as .deb package and install it via **dpkg -i $pathToDownloadedFile**.
Afterwards the installation ran successfully.

1bonp, 2017-11-10:
Jesus Christ this is some cringeworthy stuff
Why do you even use kali if you don't know what a Makefile is?
Who are you trying to ddos kid ? 
Stop stealing my grandmas WEP key!