lwfinger/rtl8723bu #59: Cannot install in Lubuntu

dagohil, 2017-03-24:
In Ubuntu I was able to install the driver. However, I chosed to install Lubuntu for resource reason. 

When giving the command make I receive the following message:

make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.8.0-41-generic/build M=/home/dago/rtl8723bu  modules
make[1]: Verzeichnis „/usr/src/linux-headers-4.8.0-41-generic“ wird betreten
arch/x86/Makefile:140: CONFIG_X86_X32 enabled but no binutils support
make[1]: gcc: Befehl nicht gefunden
make[1]: gcc: Befehl nicht gefunden
make[1]: gcc: Befehl nicht gefunden
make[1]: gcc: Befehl nicht gefunden
make[1]: gcc: Befehl nicht gefunden
  CC [M]  /home/dago/rtl8723bu/core/rtw_cmd.o
/bin/sh: 1: gcc: not found
scripts/Makefile.build:289: die Regel für Ziel „/home/dago/rtl8723bu/core/rtw_cmd.o“ scheiterte
make[2]: *** [/home/dago/rtl8723bu/core/rtw_cmd.o] Fehler 127
Makefile:1491: die Regel für Ziel „_module_/home/dago/rtl8723bu“ scheiterte
make[1]: *** [_module_/home/dago/rtl8723bu] Fehler 2
make[1]: Verzeichnis „/usr/src/linux-headers-4.8.0-41-generic“ wird verlassen
Makefile:321: die Regel für Ziel „modules“ scheiterte
make: *** [modules] Fehler 2

binutils is installed

I need help, thanks in advance.

lwfinger, 2017-03-24:
The errors are quite explicit.  You have not installed gcc.

dagohil, 2017-04-07:
Thanks. It worked.

closed 2017-04-07 by dagohil