lwfinger/rtl8723bu #61: Need a way to determine antenna port when using single antenna

sancloud, 2017-04-19:
We have used the RTL8723BU on our board and have had some complaints about the transmit power of the module.
The driver is built as is from the lwfinger/rtl8723bu repository and all seems OK when in close proximity to the board.
Our system uses only a single antenna for the WiFi and BT.

We would like to know how to select the desired antenna as it seems like the default one being used is not RF0.  We think the default mode is with antenna diversity.

Is there a way to actively select the antenna and to force it to single antenna mode ?

Thanks
Marc

lwfinger, 2017-04-19:
[antenna_debug.txt](https://github.com/lwfinger/rtl8723bu/files/938196/antenna_debug.txt)

The antenna selection uses register 0x860. In one place it says bits 12-14 are involved, and in another it uses bits 8-9. The file above is a patch that will watch writes to register 0x860 and log the value to be written when the mask involves any of those bits. Apply this patch and post the dmesg output.

mpvader, 2017-04-22:
@lwfinger, herewith the bootlog with that patch applied:

```
root@beaglebone:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.1.39-venus (matthijs@matthijs-VirtualBox) (gcc version 5.2.0 (GCC) ) #1 Wed Apr 19 23:02:48 CEST 2017
[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: SanCloud BeagleBone Enhanced with victronenergy cape
[    0.000000] cma: Reserved 16 MiB at 0xbf000000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 262144
[    0.000000] free_area_init_node: node 0, pgdat c07363d0, node_mem_map eeff9000
[    0.000000]   Normal zone: 1520 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 194560 pages, LIFO batch:31
[    0.000000]   HighMem zone: 67584 pages, LIFO batch:15
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM335X ES2.1 (sgx neon )
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 260624
[    0.000000] Kernel command line: console=ttyO0,115200n8 consoleblank=0 root=/dev/mmcblk1p2 rw rootfstype=ext4 rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 1007376K/1048576K available (5010K kernel code, 323K rwdata, 1760K rodata, 276K init, 8172K bss, 24816K reserved, 16384K cma-reserved, 253952K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xf0000000 - 0xff000000   ( 240 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xef800000   ( 760 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc06a4b48   (6771 kB)
[    0.000000]       .init : 0xc06a5000 - 0xc06ea000   ( 276 kB)
[    0.000000]       .data : 0xc06ea000 - 0xc073ae78   ( 324 kB)
[    0.000000]        .bss : 0xc073ae78 - 0xc0f36118   (8173 kB)
[    0.000000] Running RCU self tests
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] IRQ: Found an INTC at 0xfa200000 (revision 5.0) with 128 interrupts
[    0.000000] OMAP clockevent source: timer2 at 24000000 Hz
[    0.000013] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
[    0.000034] clocksource timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000079] OMAP clocksource: timer1 at 24000000 Hz
[    0.000766] Console: colour dummy device 80x30
[    0.000818] Lock dependency validator: Copyright (c) 2006 Red Hat, Inc., Ingo Molnar
[    0.000825] ... MAX_LOCKDEP_SUBCLASSES:  8
[    0.000832] ... MAX_LOCK_DEPTH:          48
[    0.000838] ... MAX_LOCKDEP_KEYS:        8191
[    0.000843] ... CLASSHASH_SIZE:          4096
[    0.000849] ... MAX_LOCKDEP_ENTRIES:     32768
[    0.000855] ... MAX_LOCKDEP_CHAINS:      65536
[    0.000861] ... CHAINHASH_SIZE:          32768
[    0.000867]  memory used by lock dependency info: 5167 kB
[    0.000874]  per task-struct memory footprint: 1152 bytes
[    0.000899] Calibrating delay loop... 996.14 BogoMIPS (lpj=4980736)
[    0.078965] pid_max: default: 32768 minimum: 301
[    0.079180] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.079193] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.081479] CPU: Testing write buffer coherency: ok
[    0.082677] Setting up static identity map for 0x80008200 - 0x80008258
[    0.086727] devtmpfs: initialized
[    0.113868] VFP support v0.3: implementor 41 architecture 3 part 30 variant c rev 3
[    0.149748] omap_hwmod: tptc0 using broken dt data from edma
[    0.150160] omap_hwmod: tptc1 using broken dt data from edma
[    0.150550] omap_hwmod: tptc2 using broken dt data from edma
[    0.158327] omap_hwmod: debugss: _wait_target_disable failed
[    0.213502] clocksource jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.213582] futex hash table entries: 256 (order: 1, 11264 bytes)
[... 578 lines cut ...]
[   21.264850] RTL871X: nolinked power save enter
[   21.338623] cpsw 4a100000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   21.560595] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   21.654906] c_can_platform 481d0000.can can1: setting BTR=1c05 BRPE=0000
[   21.662061] c_can_platform 481d0000.can: obtain a copy of previously claimed pinctrl
[   21.670320] IPv6: ADDRCONF(NETDEV_CHANGE): can1: link becomes ready
[   22.580332] c_can_platform 481d0000.can can1: setting BTR=1c05 BRPE=0000
[   22.587486] c_can_platform 481d0000.can: obtain a copy of previously claimed pinctrl
[   22.595740] IPv6: ADDRCONF(NETDEV_CHANGE): can1: link becomes ready
[   23.356966] cfg80211: Calling CRDA to update world regulatory domain
[   23.528404] c_can_platform 481d0000.can can1: setting BTR=1c05 BRPE=0000
[   23.535468] c_can_platform 481d0000.can: obtain a copy of previously claimed pinctrl
[... cut ...]
[   31.495699] BitMask 0xffffffff, Data 0x66f60110
[   31.729728] RTL871X: nolinked power save leave
[   31.962952] cfg80211: Calling CRDA to update world regulatory domain
[   32.002895] c_can_platform 481d0000.can can1: setting BTR=1c05 BRPE=0000
[   32.010063] c_can_platform 481d0000.can: obtain a copy of previously claimed pinctrl
[   32.018340] IPv6: ADDRCONF(NETDEV_CHANGE): can1: link becomes ready
[   33.917248] cfg80211: Calling CRDA to update world regulatory domain
[   33.979427] RTL871X: nolinked power save enter
```

mpvader, 2017-04-22:
And another one, now with the access point up, and being connected to:

```
root@beaglebone:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.1.39-venus (matthijs@matthijs-VirtualBox) (gcc version 5.2.0 (GCC) ) #1 Wed Apr 19 23:02:48 CEST 2017
[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] Machine model: SanCloud BeagleBone Enhanced with victronenergy cape
[    0.000000] cma: Reserved 16 MiB at 0xbf000000
[    0.000000] Memory policy: Data cache writeback
[    0.000000] On node 0 totalpages: 262144
[    0.000000] free_area_init_node: node 0, pgdat c07363d0, node_mem_map eeff9000
[    0.000000]   Normal zone: 1520 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 194560 pages, LIFO batch:31
[    0.000000]   HighMem zone: 67584 pages, LIFO batch:15
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM335X ES2.1 (sgx neon )
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 260624
[    0.000000] Kernel command line: console=ttyO0,115200n8 consoleblank=0 root=/dev/mmcblk1p2 rw rootfstype=ext4 rootwait
[    0.000000] PID hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 1007376K/1048576K available (5010K kernel code, 323K rwdata, 1760K rodata, 276K init, 8172K bss, 24816K reserved, 16384K cma-reserved, 253952K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xf0000000 - 0xff000000   ( 240 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xef800000   ( 760 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0008000 - 0xc06a4b48   (6771 kB)
[    0.000000]       .init : 0xc06a5000 - 0xc06ea000   ( 276 kB)
[    0.000000]       .data : 0xc06ea000 - 0xc073ae78   ( 324 kB)
[    0.000000]        .bss : 0xc073ae78 - 0xc0f36118   (8173 kB)
[    0.000000] Running RCU self tests
[    0.000000] NR_IRQS:16 nr_irqs:16 16
[    0.000000] IRQ: Found an INTC at 0xfa200000 (revision 5.0) with 128 interrupts
[    0.000000] OMAP clockevent source: timer2 at 24000000 Hz
[    0.000013] sched_clock: 32 bits at 24MHz, resolution 41ns, wraps every 89478484971ns
[    0.000034] clocksource timer1: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 79635851949 ns
[    0.000077] OMAP clocksource: timer1 at 24000000 Hz
[    0.000765] Console: colour dummy device 80x30
[    0.000820] Lock dependency validator: Copyright (c) 2006 Red Hat, Inc., Ingo Molnar
[    0.000828] ... MAX_LOCKDEP_SUBCLASSES:  8
[    0.000834] ... MAX_LOCK_DEPTH:          48
[    0.000840] ... MAX_LOCKDEP_KEYS:        8191
[    0.000846] ... CLASSHASH_SIZE:          4096
[    0.000851] ... MAX_LOCKDEP_ENTRIES:     32768
[    0.000857] ... MAX_LOCKDEP_CHAINS:      65536
[    0.000863] ... CHAINHASH_SIZE:          32768
[    0.000869]  memory used by lock dependency info: 5167 kB
[    0.000875]  per task-struct memory footprint: 1152 bytes
[    0.000899] Calibrating delay loop... 996.14 BogoMIPS (lpj=4980736)
[    0.078960] pid_max: default: 32768 minimum: 301
[    0.079175] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.079190] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes)
[    0.081469] CPU: Testing write buffer coherency: ok
[    0.082680] Setting up static identity map for 0x80008200 - 0x80008258
[    0.086722] devtmpfs: initialized
[    0.113853] VFP support v0.3: implementor 41 architecture 3 part 30 variant c rev 3
[    0.149701] omap_hwmod: tptc0 using broken dt data from edma
[    0.150117] omap_hwmod: tptc1 using broken dt data from edma
[    0.150509] omap_hwmod: tptc2 using broken dt data from edma
[    0.158283] omap_hwmod: debugss: _wait_target_disable failed
[    0.213460] clocksource jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.213538] futex hash table entries: 256 (order: 1, 11264 bytes)
[... 602 lines cut ...]
[   24.950968] c_can_platform 481d0000.can can1: setting BTR=1c05 BRPE=0000
[   24.958147] c_can_platform 481d0000.can: obtain a copy of previously claimed pinctrl
[   24.966409] IPv6: ADDRCONF(NETDEV_CHANGE): can1: link becomes ready
[   25.246981] cfg80211: Calling CRDA to update world regulatory domain
[   25.758631] cfg80211: Calling CRDA to update world regulatory domain
[   26.024396] c_can_platform 481d0000.can can1: setting BTR=1c05 BRPE=0000
[   26.031551] c_can_platform 481d0000.can: obtain a copy of previously claimed pinctrl
[   26.039851] IPv6: ADDRCONF(NETDEV_CHANGE): can1: link becomes ready
[   26.978831] c_can_platform 481d0000.can can1: setting BTR=1c05 BRPE=0000
[   26.985895] c_can_platform 481d0000.can: obtain a copy of previously claimed pinctrl
[   26.994236] IPv6: ADDRCONF(NETDEV_CHANGE): can1: link becomes ready
[   28.001109] c_can_platform 481d0000.can can1: setting BTR=1c05 BRPE=0000
[... cut ...]
[   34.729457] cfg80211: Calling CRDA to update world regulatory domain
[   37.891072] cfg80211: Calling CRDA to update world regulatory domain
[   41.052755] cfg80211: Exceeded CRDA call max attempts. Not calling CRDA
[   56.720536] cfg80211: Exceeded CRDA call max attempts. Not calling CRDA
[  119.329777] RTL871X: set group key camid:1, addr:00:00:00:00:00:00, kid:1, type:TKIP
[  119.341583] RTL871X: send eapol packet
[  119.349625] RTL871X: send eapol packet
[  119.358168] RTL871X: set pairwise key camid:4, addr:e8:b1:fc:29:77:b4, kid:0, type:TKIP
```

sancloud, 2017-04-24:
We also see fffffff as the bitmask:
[    4.052669] usb 1-1.4: new high-speed USB device number 3 using musb-hdrc
[    4.173360] usb 1-1.4: New USB device found, idVendor=0bda, idProduct=b720
[    4.180287] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[    4.187669] usb 1-1.4: Product: 802.11n WLAN Adapter
[    4.192673] usb 1-1.4: Manufacturer: Realtek
[    4.196962] usb 1-1.4: SerialNumber: 00e04c000001
[    5.747631] RTL871X: module init start
[    5.751431] RTL871X: rtl8723bu v4.3.6.11_12942.20141204_BTCOEX20140507-4E40
[    5.758493] RTL871X: rtl8723bu BT-Coex version = BTCOEX20140507-4E40
[    5.819883] random: nonblocking pool is initialized
[    5.857160] RTL871X: rtw_ndev_init(wlan0)
[    5.867928] RTL871X: rtw_ndev_init(wlan1)
[    5.884951] usbcore: registered new interface driver rtl8723bu
[    5.890830] RTL871X: module init ret=0

# ifconfig wlan0 up
[  754.364425] BitMask 0xffffffff, Data 0x66f60110
[  754.959206] RTL871X: RTW_ADAPTIVITY_EN_AUTO, chplan:0x20, Regulation:3,3
[  754.965975] RTL871X: RTW_ADAPTIVITY_MODE_NORMAL
[  755.277215] BitMask 0xffffffff, Data 0x66f60110
[  755.354039] BitMask 0xffffffff, Data 0x66f60110
[  755.568395] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
# [  757.563466] RTL871X: nolinked power save enter

lwfinger, 2017-04-24:
That means that the driver is never selecting an antenna. I think that allows the firmware to choose it depending on the encoding of the EFUSE.

I was trying to implement CONFIG_ANTENNA_DIVERSITY, but the code does not yet build. I'll let you know when that is available.

sancloud, 2017-04-24:
That would make sense as to why the performance is poor.  We can't use antenna diversity due to only having a single antenna and need to use coexistence.  If we need to patch the driver to force the desired antenna then that's OK, just need to know which register to poke.

lwfinger, 2017-04-24:
I know you cannot use antenna diversity, but I hoped that implementing that option would give us more clues as to how they utilize the antennas.

Register 0860 & 0x300 shows the RX setting. The current code is setting 1 into that field.
Register 0x860 & 0x7000 masks the TX setting. The current code sets 0 into that field.

One easy way to test is to go to hal/HalHWImg8723B_BB.c, line 347, which now contains:             0x860, 0x66F60110, and change the 110 part to 010 to select antenna 0, 210 for antenna 2, of 310 for antenna 3.  See if any of those improve the RX signal.

mansr, 2017-04-25:
I've tried all values for both rx and tx setting. Nothing makes a difference on reported received signal strength or measured output signal.

lwfinger, 2017-04-25:
Unfortunately, we have exhausted my knowledge. Sorry.

sancloud, 2017-04-25:
Can you recommend anyone else who has experience of the firmware in this module that may be able to help please ?
Thank you for all your assistance.