lwfinger/rtl8723bu #63: incomplete type error during make

ljishen, 2017-04-29:
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.5.0-00185-g3bb556b/build M=/home/nano/Desktop/rtl8723bu  modules
make[1]: Entering directory '/root/de10_nano/linux-socfpga'
  Building modules, stage 2.
  MODPOST 1 modules
WARNING: "register_netdevice" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "system_wq" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "kmalloc_caches" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "__kmalloc" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "ktime_get_with_offset" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "msleep" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "complete_and_exit" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "wiphy_free" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "cfg80211_unlink_bss" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "vmalloc" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "single_open" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "param_ops_int" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "del_timer" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "strlen" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "ieee80211_frequency_to_channel" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "single_release" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "find_vpid" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "__tasklet_hi_schedule" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "netif_carrier_on" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "skb_clone" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "dev_get_by_name" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "simple_strtoul" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "skb_copy" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "down_interruptible" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "seq_printf" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "register_netdevice_notifier" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "netif_carrier_off" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "cancel_work_sync" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "usb_kill_urb" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "remove_proc_entry" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "cfg80211_rx_mgmt" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "filp_close" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "__dev_kfree_skb_any" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "init_timer_key" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "arm_copy_from_user" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "mutex_unlock" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "usb_autopm_get_interface" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "strsep" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "vfree" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "sprintf" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "ns_to_timespec" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "seq_read" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "kthread_create_on_node" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "jiffies" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "unregister_netdevice_notifier" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "arm_copy_to_user" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "skb_trim" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "strcmp" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "cfg80211_mgmt_tx_status" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "__divsi3" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "__netdev_alloc_skb" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "netif_rx" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "__pskb_pull_tail" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "__init_waitqueue_head" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "PDE_DATA" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "param_ops_charp" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "__memzero" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "memset" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "__ieee80211_get_channel" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "netif_tx_wake_queue" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "netif_tx_stop_all_queues" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "cfg80211_get_bss" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "usb_deregister" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "mutex_lock_interruptible" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "__mutex_init" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
WARNING: "printk" [/home/nano/Desktop/rtl8723bu/8723bu.ko] undefined!
[... 103 lines cut ...]
          ^
/home/nano/Desktop/rtl8723bu/8723bu.mod.c:10:2: error: unknown field 'init' specified in initializer
  .init = init_module,
  ^
/home/nano/Desktop/rtl8723bu/8723bu.mod.c:10:10: warning: excess elements in struct initializer
  .init = init_module,
          ^
/home/nano/Desktop/rtl8723bu/8723bu.mod.c:10:10: note: (near initialization for '__this_module')
/home/nano/Desktop/rtl8723bu/8723bu.mod.c:14:2: error: unknown field 'arch' specified in initializer
  .arch = MODULE_ARCH_INIT,
  ^
/home/nano/Desktop/rtl8723bu/8723bu.mod.c:14:10: error: 'MODULE_ARCH_INIT' undeclared here (not in a function)
  .arch = MODULE_ARCH_INIT,
          ^
/home/nano/Desktop/rtl8723bu/8723bu.mod.c:14:10: warning: excess elements in struct initializer
/home/nano/Desktop/rtl8723bu/8723bu.mod.c:14:10: note: (near initialization for '__this_module')
scripts/Makefile.modpost:114: recipe for target '/home/nano/Desktop/rtl8723bu/8723bu.mod.o' failed
make[2]: *** [/home/nano/Desktop/rtl8723bu/8723bu.mod.o] Error 1
Makefile:1394: recipe for target 'modules' failed
make[1]: *** [modules] Error 2
make[1]: Leaving directory '/root/de10_nano/linux-socfpga'
Makefile:321: recipe for target 'modules' failed
make: *** [modules] Error 2

Thanks in advance for any response!

lwfinger, 2017-04-29:
I'm not sure if this explains all the problems, but it looks as if the system symbol table in "System.map" is missing, or not current/correct.

ljishen, 2017-05-06:
Thanks. I solved this problem by installing some modules. Now the `make` process didn't complain any undefined WARNINGs.

However, when I run `sudo modprobe -v 8723bu`, it gives `Exec format error`.

The related message from dmesg is:
cfg80211: exports duplicate symbol __cfg80211_alloc_event_skb (owned by kernel)

lwfinger, 2017-05-07:
That message seems to indicate that cfg80211 is built in AND a module. Check your .config.

inklit, 2017-08-13:
@ljishen Can you tell me what you did to solve the problem? What modules did you install? I am having the same issue but nobody has stated what fixes it.

ljishen, 2017-08-14:
@inklit  My solution was that I got the source tree of the running system and use `cat /proc/config.gz | gunzip > .config` to obtain a "correct" configuration file for the kernel compiling.