lwfinger/rtl8723bu #68: Direct firmware load for rtl_bt/rtl8723b_config.bin failed with error -2

hotxboy, 2017-06-09:
I installed today. And Wifi seems work perfectly but I found some errors in dmesg, says 
`Direct firmware load for rtl_bt/rtl8723b_config.bin failed with error -2`. Is it normal?
`[    0.000000] Linux version 4.12.0-041200rc3-generic (kernel@gomeisa) (gcc version 6.3.0 20170519 (Ubuntu/Linaro 6.3.0-18ubuntu2) ) #201705282131 SMP Mon May 29 01:32:35 UTC 2017
[    0.000000] Command line: BOOT_IMAGE=/boot/vmlinuz-4.12.0-041200rc3-generic root=UUID=883b511c-a36b-4c81-abe7-467e4c5421e6 ro quiet splash vt.handoff=7
[    0.000000] KERNEL supported cpus:
[    0.000000]   Intel GenuineIntel
[    0.000000]   AMD AuthenticAMD
[    0.000000]   Centaur CentaurHauls
[    0.000000] x86/fpu: x87 FPU will use FXSAVE
[    0.000000] e820: BIOS-provided physical RAM map:
[    0.000000] BIOS-e820: [mem 0x0000000000000000-0x000000000006efff] usable
[    0.000000] BIOS-e820: [mem 0x000000000006f000-0x000000000006ffff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x0000000000070000-0x0000000000087fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000000088000-0x000000000009ffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000000100000-0x000000001fffffff] usable
[    0.000000] BIOS-e820: [mem 0x0000000020000000-0x00000000201fffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000020200000-0x0000000073fa9fff] usable
[    0.000000] BIOS-e820: [mem 0x0000000073faa000-0x00000000741a9fff] type 20
[    0.000000] BIOS-e820: [mem 0x00000000741aa000-0x00000000749a9fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000749aa000-0x00000000764a9fff] ACPI NVS
[    0.000000] BIOS-e820: [mem 0x00000000764aa000-0x00000000764e9fff] ACPI data
[    0.000000] BIOS-e820: [mem 0x00000000764ea000-0x0000000077ffffff] usable
[    0.000000] BIOS-e820: [mem 0x00000000e00f8000-0x00000000e00f8fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000fed01000-0x00000000fed01fff] reserved
[    0.000000] BIOS-e820: [mem 0x00000000ffc00000-0x00000000ffffffff] reserved
[    0.000000] BIOS-e820: [mem 0x0000000100000000-0x000000017fffffff] usable
[    0.000000] NX (Execute Disable) protection: active
[    0.000000] efi: EFI v2.40 by INSYDE Corp.
[    0.000000] efi:  ACPI 2.0=0x764e9014  SMBIOS=0x7433a000  ESRT=0x7433d718 
[    0.000000] SMBIOS 2.8 present.
[    0.000000] DMI: LYZPIN SL/SL, BIOS 22.16 07/29/2015
[    0.000000] e820: update [mem 0x00000000-0x00000fff] usable ==> reserved
[    0.000000] e820: remove [mem 0x000a0000-0x000fffff] usable
[    0.000000] e820: last_pfn = 0x180000 max_arch_pfn = 0x400000000
[    0.000000] MTRR default type: uncachable
[    0.000000] MTRR fixed ranges enabled:
[    0.000000]   00000-9FFFF write-back
[    0.000000]   A0000-FFFFF write-protect
[    0.000000] MTRR variable ranges enabled:
[    0.000000]   0 base 0FFC00000 mask FFFC00000 write-protect
[    0.000000]   1 base 000000000 mask F80000000 write-back
[    0.000000]   2 base 07C000000 mask FFC000000 uncachable
[    0.000000]   3 base 07BC00000 mask FFFC00000 uncachable
[    0.000000]   4 base 100000000 mask F80000000 write-back
[    0.000000]   5 disabled
[    0.000000]   6 disabled
[    0.000000]   7 disabled
[    0.000000] x86/PAT: Configuration [0-7]: WB  WC  UC- UC  WB  WC  UC- WT  
[    0.000000] e820: last_pfn = 0x78000 max_arch_pfn = 0x400000000
[    0.000000] esrt: Reserving ESRT space from 0x000000007433d718 to 0x000000007433d750.
[    0.000000] Scanning 1 areas for low memory corruption
[    0.000000] Base memory trampoline at [ffff9c82c007e000] 7e000 size 24576
[    0.000000] BRK [0x24d15000, 0x24d15fff] PGTABLE
[    0.000000] BRK [0x24d16000, 0x24d16fff] PGTABLE
[    0.000000] BRK [0x24d17000, 0x24d17fff] PGTABLE
[    0.000000] BRK [0x24d18000, 0x24d18fff] PGTABLE
[    0.000000] BRK [0x24d19000, 0x24d19fff] PGTABLE
[    0.000000] BRK [0x24d1a000, 0x24d1afff] PGTABLE
[    0.000000] BRK [0x24d1b000, 0x24d1bfff] PGTABLE
[    0.000000] Secure boot could not be determined
[    0.000000] RAMDISK: [mem 0x32e5f000-0x35726fff]
[    0.000000] ACPI: Early table checksum verification disabled
[    0.000000] ACPI: RSDP 0x00000000764E9014 000024 (v02 INSYDE)
[    0.000000] ACPI: XSDT 0x00000000764E9120 0000A4 (v01 INSYDE INSYDE   00000003      01000013)
[    0.000000] ACPI: FACP 0x00000000764E4000 00010C (v05 INSYDE INSYDE   00000003 ACPI 00040000)
[    0.000000] ACPI: DSDT 0x00000000764D2000 00BB30 (v02 INSYDE INSYDE   00000003 ACPI 00040000)
[    0.000000] ACPI: FACS 0x0000000076496000 000040
[    0.000000] ACPI: UEFI 0x00000000764E8000 000236 (v01 INSYDE INSYDE   00000001 ACPI 00040000)
[    0.000000] ACPI: TCPA 0x00000000764E7000 000032 (v02 INSYDE INSYDE   00000000 ACPI 00040000)
[    0.000000] ACPI: UEFI 0x00000000764E5000 000042 (v01 INSYDE INSYDE   00000000 ACPI 00040000)
[... 671 lines cut ...]
[   22.939295] RTL871X: nolinked power save leave
[   23.264014] IPv6: ADDRCONF(NETDEV_UP): wlx7cc709c9957e: link is not ready
[   23.294857] input: CyPS/2 Cypress Trackpad as /devices/platform/i8042/serio2/input/input20
[   25.260873] RTL871X: nolinked power save enter
[   45.474269] input: ImPS/2 Generic Wheel Mouse as /devices/platform/i8042/serio2/input/input23
[   45.485814] atkbd serio0: Unknown key released (translated set 2, code 0x7c on isa0060/serio0).
[   45.485835] atkbd serio0: Use 'setkeycodes 7c <keycode>' to make it known.
[   45.688264] atkbd serio0: Unknown key released (translated set 2, code 0x7c on isa0060/serio0).
[   45.688284] atkbd serio0: Use 'setkeycodes 7c <keycode>' to make it known.
[   57.051964] RTL871X: RTW_ADAPTIVITY_EN_
[   57.051970] AUTO, chplan:0x20, Regulation:0,0
[   57.051972] RTL871X: RTW_ADAPTIVITY_MODE_
[   57.051973] NORMAL
[   57.582653] RTL871X: nolinked power save leave
[   59.755134] RTL871X: rtw_set_802_11_connect(wlx7cc709c9957e)  fw_state=0x00000008
[   59.897508] RTL871X: start auth
[   59.901410] RTL871X: auth success, start assoc
[   59.910504] RTL871X: rtw_cfg80211_indicate_connect(wlx7cc709c9957e) BSS not found !!
[   59.910531] RTL871X: assoc success
[   59.910666] IPv6: ADDRCONF(NETDEV_CHANGE): wlx7cc709c9957e: link becomes ready
[   60.267749] RTL871X: send eapol packet
[   60.321588] RTL871X: send eapol packet
[   60.322367] RTL871X: set pairwise key camid:4, addr:xx:xx:xx:xx:xx:xx, kid:0, type:AES
[   60.324216] RTL871X: set group key camid:5, addr:xx:xx:xx:xx:xx:xx, kid:1, type:AES
`

lwfinger, 2017-06-09:
Yes. The "config.bin" files are for modifying the BlueTooth firmware, if necessary. They have absolutely no effect on wifi. In addition, the BT part of RTL8723BE does not need such a file. The message has been removed in later versions.

closed 2017-06-09 by lwfinger

liar666, 2019-06-02:
Hi,

FWIW, I have the exact same problem (error appears in logs, but wifi works / can't test BT) in Xubuntu 19.04...

lwfinger, 2019-06-02:
BT messages have NO effect on WIFI!!!! In addition, the BT part of RT:8723BU does not need the kind of patch foune in the xxx_config.bin. Thus your situation is normal.

Jose-MPM, 2023-12-14:
And if we have:
[   29.088020] bluetooth hci0: firmware: failed to load rtl_bt/rtl8723b_config.bin (-2)
[   29.088073] firmware_class: See https://wiki.debian.org/Firmware for information about missing firmware
[   29.088126] bluetooth hci0: firmware: failed to load rtl_bt/rtl8723b_config.bin (-2)

is it possible that this generates an error with the NetworkManager? But the main error is that I cant use BT, in this cases what do u recommend?

lwfinger, 2023-12-15:
No, it generates an error in the BT driver. You NEED that firmware file if you want to use BT.

gachezra, 2024-06-05:
> And if we have:
>
> [   29.088020] bluetooth hci0: firmware: failed to load rtl_bt/rtl8723b_config.bin (-2)
>
> [   29.088073] firmware_class: See https://wiki.debian.org/Firmware for information about [...]

I'm having the exact same problem. In the event I'm successfull fixing it I'll post the solution here👍🏽