lwfinger/rtl8723bu #69: What is the license for this module?

laneroberts, 2017-06-27:
The sources appear to be GPLv2 from Realtek but there doesn't appear to be a top-level license. I'm particularly worried about the firmware .bin files.

lwfinger, 2017-06-27:
In the original driver as it came from Realtek, the firmware came in the form of huge data statements. For those drivers that had been incorporated into staging, the firmware was extracted and written as the .bin files to reduce the size of the source. The same action was done (by me) on this driver.

The source has a GPLv2 license while ALL Realtek firmware is covered by a license to redistribute that is covered by file LICENCE.rtlwifi_firmware.txt  in the linux-firmware repo.

If using this firmware license bothers you, you can always go back to the source as it came from Realtek.

closed 2017-06-27 by lwfinger

laneroberts, 2017-06-27:
Okay, cool, sounds good. Many thanks for the information (and for a driver that actually works!)