lwfinger/rtl8723bu #71: Doesn't compile when enabling MP_DRIVER

alkino, 2017-07-31:
Hi,
In autoconf.h we can see `#define MP_DRIVER 0`. If we enable it, it doesn't compile due to your change in 7b704bc0ff847f9c3c389b48e52d0ee4c9f7cba2. (Remove of `mppriv` for example). My company need this to enable some strange feature of wifi for CEM compatibility of our devices.
Is there something to do to get those features?
Regards,
Nicolas

blpanadero, 2017-09-22:
Hi Nicolas,

I have the same problem right now, I need testing my device por CEM compatibility, and I can't compile the driver with "CONFIG_MP_INCLUDED = y" option enabled in the Makefile.

Did you get compile the driver with these opcion? 

Regards.

alkino, 2017-09-22:
I endly use the original tar file in this repo. It works with MP_DRIVER.

lwfinger, 2017-09-23:
Why do you want to use MP_DRIVER? The MP stands for Manufacturing Process. That version includes lots of tests employed when checking if the chip is any good. As there is no documentation for interpreting the results, the code does no good in the field. It does NOT mean multi-processing. ALL Linux drivers are capable of handling SMP.

blpanadero, 2017-09-25:
I need to use MP_DRIVER because I have to pass a EMC (electromagnetic compability) test for our device. The MP_DRIVER can help me to select the channel, frequency, modulation... and others features which are required for pass the test.

lwfinger, 2017-09-25:
OK, then you will need to use the original driver without any of the simplifications that have been made. Of course, there are no guarantees that it will do what you want.

closed 2017-09-25 by lwfinger