lwfinger/rtl8723bu #74: Failure to report access point vanishing

mansr, 2017-08-04:
If the associated access point goes away, such as by being turned off, the rtl8723bu driver fails to report this the way other wifi drivers do, and wpa_supplicant never notices the lost association. The end result is that if an access point is power cycled, the rtl8723bu device never reconnects after the AP comes back up. The kernel log contains a single message related to the lost connection:
` RTL871X: linked_status_chk(wlan0) disconnect or roaming`

With other wifi adapters and drivers, the userspace tools notice the disconnect and start scanning, eventually reconnecting when the network becomes available again.

mansr, 2017-08-04:
Disabling CONFIG_LAYER2_ROAMING makes it behave as expected.

mdtek, 2017-12-19:
I confirm the issue.
In Raspbian Jessie (4.4.37) the driver works as expected.
In Raspbian Jessie (4.9.35) the wifi adapter won't reconnect if AP is power cycled.
I tried to edit all power-related parameters, but wpa is not notified of AP disconnection. You can check it by powering down AP and then typing wpa_cli -i wlan0 status. You will see that wpa is not aware of AP disconnection.

The only way to make it reconnect is by calling "wpa_cli -i wlan0 reconfigure"

Commenting out CONFIG_LAYER2_ROAMING in include/autoconf.h and re-build, makes it work again.

Thank you mansr!

Egor-G, 2023-07-22:
Thank you, mansr so much! I spend a lot of hours to fix same behavior for rtl8188eu until I found you message

closed 2023-07-22 by lwfinger