lwfinger/rtl8723bu #76: Dongle usb bluetooth + wifi 0bda:b720 realtek : no bluetooth ?

Coeur-Noir, 2017-08-28:
Hi,

first sorry for my poor english, and my low skills about drivers.

The whole story is here https://forum.ubuntu-fr.org/viewtopic.php?id=2013553 and there https://forum.ubuntu-fr.org/viewtopic.php?id=2013991 , in french.

To sum it up : 
- I've bought this usb dongle ( no ad intended ) https://www.amazon.fr/gp/product/B01IT7GJRI/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1
- out of the box on linux ubuntu 16.04 unity, bluetooth works but not wifi.
- I've ended up to compiling rtl8723bu, blacklist rtl8xxxu
- then wifi works ( but can't change its parameters, interface unknown wlx0013efd002c3 ) and then also I can no longer connect my bluetooth device ( headphone ).

What informations may I provide to help diagnose ?
( you may find dmesg, hcidump, iwconfig, and so on along the threads linked above )

Best regards.

lwfinger, 2017-08-28:
I do not read French, thus I was only able to read the computer output in your links. From those, I have the following comments:

1. The iwconfig commands are not implemented. You need to supply the equivalent commands using iw.

2. The driver in this repo is provided "as is". If you have a problem building it, or if it crashes with a kernel oops, I will try to fix it. If it does not work the way you want, then you will need to fix that.  However, if I build this driver and load it with no other driver loaded, it works for my device with ID obda:b720.

3. The device also works with rtl8xxxu, but not as well as with 8273bu. Note that rtl8xxxu is being actively worked on. If you have complaints about it, send those to the maintainer shown in the MAINTAINERS file in your source tree.

4. The version of udev in Ubuntu is really bad! Why should you rename a wireless device from wlan0 to wlx0013efd002c3? I know that it needs to be renamed, but my openSUSE Leap 42.2 system renames it to wlan24. The only reason it gets a number as large as 24 is that I have a large number of wifi cards that get tested.

Coeur-Noir, 2017-08-29:
Hello and thank you for explanations.

No problem for building, no kernel oops. Only no bluetooth ( but wifi ok ) on 0bda:b720 if used with 8723bu. And no wifi ( but bluetooth ok ) on a vanilla ubuntu.

The interface name seemed also very strange to me. And I expected something like you describe, or close to wlp1s0 ( this might be of interest https://askubuntu.com/questions/767786/changing-network-interfaces-name-ubuntu-16-04 )

Last question : can you confirm your 0bda:b720 device has bluetooth+wifi working together ok ?

Coeur-Noir, 2017-09-01:
Ok ! I've cleaned my system of previous driver installation or compilation. And then have re-compiled rtl8723bu. And wifi + bluetoooth work both as expected !

xieyanlin, 2018-01-23:
Hi,Can your RTL8723BU BT be connected to other devices and communicate with the data?

Coeur-Noir, 2018-07-23:
I succesfully transfered some photos fromo phone Lenovo K6 note to computer,

and some music files from computer to phone…

…so yes it seems to work for me.