lwfinger/rtl8723bu #77: NO wifi, sound, and bluetooth in my NEWLY installed LINUX PC

nareshnulu, 2017-09-30:
Good morning all. I've changed from windows 10 to linux though i have license due to f...ing updates and i want to become a programmer as well. Watched the installation video in youtube and installed os even though it didn't recognize my network adapter. I googled to find a driver for wifi and bluetooth. while installing i'm facing lot of errors. When i'm using windows i know the driver is Realtek8723 for wifi. don't find any sound card or wifi card when entered lsusb or lspci. please help me with this issue. Thanks for your time.

lwfinger, 2017-09-30:
This is not the place for newbies to learn simple things. You should consult the Forums at whatever distro you have selected.

Qammarbhat, 2020-08-09:
> This is not the place for newbies to learn simple things. You should consult the Forums at whatever distro you have selected.

Hi I used your repos of rtlwifi_new earlier now I formatted my system and when I tried to install the same I couldn't find you repo. I cloned the link but it says it doesn't exist. Can you give me the link tonyour rtlwifi_new8723be driver

lwfinger, 2020-08-09:
No. It was deleted because the driver in the kernel was the same as in the repo for rtl8723be. If you are using a kernel older than 5.1, then get the driver from the Backports project.

closed 2020-08-09 by lwfinger

Qammarbhat, 2020-08-09:
Can you send me the exact link . I clone and tried one from backport it didn't work but I thing I got ut wrong. I have rtl8723be wifi drivers on HP 15 model. I hope you can send me the link.

Qammarbhat, 2020-08-10:
Hi I tried the backports project but It was empty repo.

lwfinger, 2020-08-10:
https://backports.wiki.kernel.org/index.php/Releases

Qammarbhat, 2020-08-10:
There is no documentation on how to install rtl8723be drivers. Can you
please help me one last time I am really much thankful for what you did
already.  I need help here I tried make command it didn't work. Please help
me ! I need to install realtek drivers on hp 15. I need rtl8723be drivers.
Please send documentation or anything of sort. Thank You ins advance.

lwfinger, 2020-08-10:
What kernel are you running? Until you tell me that, I cannot help you at all.

Qammarbhat, 2020-08-10:
Linux 5.5.0-kali2-amd64 #1 SMP Debian 5.5.17-1kali (2020-04-21) x86_64
GNU/Linux.

lwfinger, 2020-08-10:
With 5.5.0, you should just run the built-in kernel driver.

Qammarbhat, 2020-08-10:
When I installed the kali linux there was no wifi and there is no wifi. But
I have the rtl8723be wifi chipset and if I install window the wifi is pre
installed.  Previously on kali 2019.4 I used your repo to install wifi and
it worked really well.  I wasn't worried about wifi untill I realized you
deleted it . Can you send me your repo files via anything you like. I tried
everything in 2019 only yours worked.  I know today only your repo will
work. Please share the older repo of rtlwifi_new or please do tell me the
steps I need to get wifi on hp15

lwfinger, 2020-08-11:
You should only have to boot as kernel 5.5 has the driver built in - it is the same driver that you used before.

Check the output of dmesg to see if you are missing the firmware. If not, then post the output of 'lspci -nn'.

Qammarbhat, 2020-08-12:
I installed wifi in kali linux from https://git.kernel.org/pub/scm/linux/kernel/git/iwlwifi/linux-firmware.git/
but everytime I try to use monitor mode it freezes my whole computer . I tried airmon-ng start wlan0, wifite and wifite --kill but everytime my whole laptop freezes. 
the output of dmesg is :
`[   39.062098] rtw_8723de 0000:03:00.0: stop vif 0e:bf:b6:7f:fa:ca on port 0
[   39.211084] rtw_8723de 0000:03:00.0: start vif 9c:30:5b:12:4f:7b on port 0
[   40.533693] wlan0: authenticate with 1c:18:4a:6a:ca:90
[   40.792039] wlan0: send auth to 1c:18:4a:6a:ca:90 (try 1/3)
[   40.830805] wlan0: authenticated
[   40.830978] wlan0: associating with AP with corrupt probe response
[   40.831554] wlan0: associate with 1c:18:4a:6a:ca:90 (try 1/3)
[   40.893702] wlan0: RX AssocResp from 1c:18:4a:6a:ca:90 (capab=0x411 status=0 aid=5)
[   40.893734] rtw_8723de 0000:03:00.0: sta 1c:18:4a:6a:ca:90 joined with macid 0
[   40.894114] wlan0: associated
[   41.570385] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   49.915319] fuse: init (API version 7.31)
[ 1200.307388] rfkill: input handler disabled
   `

and output of  lspci -nn is :
`root@batm:~# lspci -nn
00:00.0 Host bridge [0600]: Intel Corporation Xeon E3-1200 v6/7th Gen Core Processor Host Bridge/DRAM Registers [8086:5914] (rev 08)
00:02.0 VGA compatible controller [0300]: Intel Corporation UHD Graphics 620 [8086:5917] (rev 07)
00:04.0 Signal processing controller [1180]: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem [8086:1903] (rev 08)
00:08.0 System peripheral [0880]: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model [8086:1911]
00:14.0 USB controller [0c03]: Intel Corporation Sunrise Point-LP USB 3.0 xHCI Controller [8086:9d2f] (rev 21)
00:14.2 Signal processing controller [1180]: Intel Corporation Sunrise Point-LP Thermal subsystem [8086:9d31] (rev 21)
00:16.0 Communication controller [0780]: Intel Corporation Sunrise Point-LP CSME HECI #1 [8086:9d3a] (rev 21)
00:17.0 SATA controller [0106]: Intel Corporation Sunrise Point-LP SATA Controller [AHCI mode] [8086:9d03] (rev 21)
00:1c.0 PCI bridge [0604]: Intel Corporation Sunrise Point-LP PCI Express Root Port #1 [8086:9d10] (rev f1)
00:1c.4 PCI bridge [0604]: Intel Corporation Sunrise Point-LP PCI Express Root Port #5 [8086:9d14] (rev f1)
00:1c.5 PCI bridge [0604]: Intel Corporation Sunrise Point-LP PCI Express Root Port #6 [8086:9d15] (rev f1)
00:1f.0 ISA bridge [0601]: Intel Corporation Sunrise Point LPC Controller/eSPI Controller [8086:9d4e] (rev 21)
00:1f.2 Memory controller [0580]: Intel Corporation Sunrise Point-LP PMC [8086:9d21] (rev 21)
00:1f.3 Audio device [0403]: Intel Corporation Sunrise Point-LP HD Audio [8086:9d71] (rev 21)
00:1f.4 SMBus [0c05]: Intel Corporation Sunrise Point-LP SMBus [8086:9d23] (rev 21)
01:00.0 Display controller [0380]: Advanced Micro Devices, Inc. [AMD/ATI] Sun XT [Radeon HD 8670A/8670M/8690M / R5 M330 / M430 / Radeon 520 Mobile] [1002:6660] (rev 83)
02:00.0 Ethernet controller [0200]: Realtek Semiconductor Co., Ltd. RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller [10ec:8168] (rev 15)
03:00.0 Network controller [0280]: Realtek Semiconductor Co., Ltd. RTL8723DE 802.11b/g/n PCIe Adapter [10ec:d723]`

also when I check what wifi card I have in linux using lspci it shows realtek8723de but when I check using my laptop serial number and model number hp.com it shows I have rtl8723be.

lwfinger, 2020-08-12:
I have a question for you. Do you think that your computer uses the on-line HP literature to determine what type of wireless interface you have? Of course not - it scans the PCI bus using the same code as lspci. Believe me, you have an RTL8723DE.

That reference you provided is for Intel wifi firmware. That would do nothing for a Realtek card.

Now, since you told me incorrect information earlier, my info was equally bad. This is a new driver. I have no idea why your computer freezes when you go into monitor mode. You will need to communicate with the people that wrote the driver. I did not.

To get the latest driver, you need to 
git clone https://github.com/lwfinger/rtw88.git
That is the latest driver for RTL8723DE.

If that still fails to go into monitor mode, then you need to follow the instructions concerning reporting problems in README.md of that new repo.