lwfinger/rtl8723bu #81: Can not build on Ubuntu 16.04 kernel 4.10.0-35

vuchl, 2017-10-14:
Hey there!

I try to compile and install the driver like stated in the docs:
```bash
sebastian@localhost  ~/Projects  git clone https://github.com/lwfinger/rtl8723bu.git
Cloning into 'rtl8723bu'...
remote: Counting objects: 2361, done.
remote: Total 2361 (delta 0), reused 1 (delta 0), pack-reused 2360
Receiving objects: 100% (2361/2361), 5.67 MiB | 691.00 KiB/s, done.
Resolving deltas: 100% (1901/1901), done.
Checking connectivity... done.

sebastian@localhost  ~/Projects  cd rtl8723bu

sebastian@localhost  ~/Projects/rtl8723bu   master  make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.10.0-35-generic/build M=/home/sebastian/Projects/rtl8723bu  modules
make[1]: Entering directory '/usr/src/linux-headers-4.10.0-35-generic'
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_cmd.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_security.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_debug.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_io.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_ioctl_query.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_ioctl_set.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_ieee80211.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_mlme.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_mlme_ext.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_wlan_util.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_vht.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_pwrctrl.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_rf.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_recv.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_sta_mgt.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_ap.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_xmit.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_p2p.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_tdls.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_br_ext.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_iol.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_sreset.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_btcoex.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_beamforming.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_odm.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/core/rtw_efuse.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/os_dep/osdep_service.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/os_dep/os_intfs.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/os_dep/usb_intf.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/os_dep/usb_ops_linux.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/os_dep/ioctl_linux.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/os_dep/xmit_linux.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/os_dep/mlme_linux.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/os_dep/recv_linux.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/os_dep/ioctl_cfg80211.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/os_dep/wifi_regd.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/os_dep/rtw_android.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/os_dep/rtw_proc.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/hal_intf.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/hal_com.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/hal_com_phycfg.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/hal_phy.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/hal_btcoex.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/hal_usb.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/hal_usb_led.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/HalPwrSeqCmd.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/Hal8723BPwrSeq.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/rtl8723b_sreset.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/rtl8723b_hal_init.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/rtl8723b_phycfg.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/rtl8723b_rf6052.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/rtl8723b_dm.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/rtl8723b_rxdesc.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/rtl8723b_cmd.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/usb_halinit.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/rtl8723bu_led.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/rtl8723bu_xmit.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/rtl8723bu_recv.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/usb_ops.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/odm_debug.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/odm_AntDiv.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/odm_interface.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/odm_HWConfig.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/odm.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/HalPhyRf.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/odm_EdcaTurboCheck.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/odm_DIG.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/odm_PathDiv.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/odm_RaInfo.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/odm_DynamicBBPowerSaving.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/odm_DynamicTxPower.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/odm_CfoTracking.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/odm_NoiseMonitor.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/HalBtc8723b1Ant.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/HalBtc8723b2Ant.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/HalHWImg8723B_BB.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/HalHWImg8723B_MAC.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/HalHWImg8723B_RF.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/HalHWImg8723B_FW.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/odm_RegConfig8723B.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/HalPhyRf_8723B.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/hal/odm_RTL8723B.o
  CC [M]  /home/sebastian/Projects/rtl8723bu/platform/platform_ops.o
  LD [M]  /home/sebastian/Projects/rtl8723bu/8723bu.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/sebastian/Projects/rtl8723bu/8723bu.mod.o
  LD [M]  /home/sebastian/Projects/rtl8723bu/8723bu.ko
make[1]: Leaving directory '/usr/src/linux-headers-4.10.0-35-generic'

sebastian@localhost  ~/Projects/rtl8723bu   master  sudo make install
install -p -m 644 -D 8723bu.ko /lib/modules/4.10.0-35-generic/kernel/drivers/net/wireless/8723bu.ko
/sbin/depmod -b "" -a 4.10.0-35-generic
depmod: ERROR: could not open directory /home/sebastian/Projects/rtl8723bu//lib/modules/4.10.0-35-generic: No such file or directory
depmod: FATAL: could not search modules: No such file or directory
Makefile:330: recipe for target 'install' failed
make: *** [install] Error 1
```
can some one help me out here?

vuchl, 2017-10-14:
I tampered a bit with the Makefile. In the install section the depmod command is rendered to this
```bash
/sbin/depmod -b "" -a 4.10.0-35-generic
```
I manually edited it to 
```bash
/sbin/depmod  -a 4.10.0-35-generic
```
and re ran 
```bash
sudo make install
```

Now it compiled after completing th steps from the README the device is started and listed in the network manager

lwfinger, 2017-10-14:
Your system has a bad value for the symbol KVER. Whatever the problem is, no one else has reported it, and my system certainly does not fail.

closed 2017-10-14 by lwfinger