lwfinger/rtl8723bu #83: MP test driver (CONFIG_MP_INCLUDED = y)

blpanadero, 2017-10-17:
Has anyone compiled rtl8723bu driver with CONFIG_MP_INCLUDED option enabled? I can't do it, its appears a lot of errors in several files.

lwfinger, 2017-10-17:
That MP code was stripped out. If you want that "feature", you will need to start with Realtek's tarball.

closed 2017-10-17 by lwfinger

blpanadero, 2017-10-17:
I have the rtl8723bu and rtl8723au_bt compiled on my target and I trying recompiling it with this option enabled. How I can start with Realtek's tarball? What I have to do?

Regards.

lwfinger, 2017-10-17:
You will need to find the driver on some website, download it, unpack the tar file, and build it. You will also need to handle any missing updates for kernel API changes.

The MP "feature" is for the manufacturing process. With it enabled, there is a lot of overhead and log spamming, which is why I disabled it.

blpanadero, 2017-10-18:
Hi again,

I have searching the driver on a lot of websites, but I could not find it. Could you say me where did you get it?  

Thanks and regards.