lwfinger/rtl8723bu #84: make install issues on raspberry pi

zinake, 2017-10-18:
I get through the make section with no problems.  When I attempt to `sudo make install`, I get the following.

> install -p -m 644 -D 8723bu.ko /lib/modules/4.9.35-v7+/kernel/drivers/net/wireless/8723bu.ko /sbin/depmod -b "" -a 4.9.35-v7+
> depmod: Error: could not open directory /home/pi/rtl18723bu//lib/modules/4.9.35-v7+: No such file or directory
> depmod: Fatal: could not search modules: No such file or directory
> Makefile:330: recipe for taret 'install' failed
> make: *** [install] Error 1

When looking at some of the other issues, it seems that that there may be problems with the headers.  I have verified that I have the headers installed

>raspberrypi-kernel-headers is already the newest version.

Forgive me as my knowledge in this is lacking... is there supposed to be a space in line 3 of the above output?  There seems to be the appropriate folder in "/lib", but not in "/home"

> depmod: Error: could not open directory /home/pi/rtl18723bu/ "SHOULD THERE BE A SPACE HERE?" /lib/modules/4.9.35-v7+: No such file or directory

If there should be a space there, how do I fix this?

Thanks in advance for your assistance.

lwfinger, 2017-10-18:
A 3rd party made some changes to Makefile that I inadvertently accepted.

Try 'git checkout 842f0dc709a659267' and see if that helps.

lwfinger, 2017-10-18:
I just pushed a fix for this issue. If you already git the 'git checkout' proposed above, do a 'git checkout master', 'git pull', and do the build again.

zinake, 2017-10-18:
That seems to have fixed it.  Thank you so much for your fast reply.

closed 2017-10-18 by zinake