lwfinger/rtl8723bu #85: Ubuntu 17.10

axalox, 2017-10-21:
I can't use my wifi with Ubuntu 17.10

`make

 ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.13.0-16-generic/build M=/home/papa/rtl8723bu  modules
make[1] : on entre dans le répertoire « /usr/src/linux-headers-4.13.0-16-generic »
  CC [M]  /home/papa/rtl8723bu/core/rtw_cmd.o
  CC [M]  /home/papa/rtl8723bu/core/rtw_security.o
  CC [M]  /home/papa/rtl8723bu/core/rtw_debug.o
  CC [M]  /home/papa/rtl8723bu/core/rtw_io.o
  CC [M]  /home/papa/rtl8723bu/core/rtw_ioctl_query.o
  CC [M]  /home/papa/rtl8723bu/core/rtw_ioctl_set.o
  CC [M]  /home/papa/rtl8723bu/core/rtw_ieee80211.o
  CC [M]  /home/papa/rtl8723bu/core/rtw_mlme.o
  **CC [M]  /home/papa/rtl8723bu/core/rtw_mlme_ext.o
/rtl8723bu/core/rtw_mlme.c: In function ‘rtw_restructure_ht_ie’:
/rtl8723bu/core/rtw_mlme.c:3663:3: warning: this ‘if’ clause does not guard... [-Wmisleading-indentation]
   if (stbc_rx_enable)
   ^~
/home/papa/rtl8723bu/core/rtw_mlme.c:3666:18: note: ...this statement, but the latter is misleadingly indented as if it were guarded by the ‘if’
                  set_mcs_rate_by_mask(ht_capie.supp_mcs_set, MCS_RATE_1R);
                  ^~~~~~~~~~~~~~~~~~~~**
  CC [M]  /home/papa/rtl8723bu/core/rtw_wlan_util.o
  CC [M]  /home/papa/rtl8723bu/core/rtw_vht.o
  CC [M]  /home/papa/rtl8723bu/core/rtw_pwrctrl.o
  CC [M]  /home/papa/rtl8723bu/core/rtw_rf.o
  CC [M]  /home/papa/rtl8723bu/core/rtw_recv.o
  CC [M]  /home/papa/rtl8723bu/core/rtw_sta_mgt.o
  CC [M]  /home/papa/rtl8723bu/core/rtw_ap.o
  CC [M]  /home/papa/rtl8723bu/core/rtw_xmit.o
`

lwfinger, 2017-10-21:
If your compiler is going to quit after every warning, you had better learn to fix these things yourself in the future. On my system, those warnings do not occur.

I have fixed this one, but I will not promise to do so in the future.

closed 2017-10-21 by lwfinger

axalox, 2017-10-22:
Thank's
No error on compilation now

But no wifi with my usb wifi, works on ubuntu 16.04 , not with my Ubuntu 17.10