lwfinger/rtl8723bu #88: Very slow connection with weak signal compared to windows

hellslinger, 2017-10-29:
Not sure how to go about providing more information, but am glad to follow any steps. Here is dmesg and kernel versions for Ubuntu 17.10
```
[    4.031513] Bluetooth: hci0: rtl: examining hci_ver=06 hci_rev=000b lmp_ver=06 lmp_subver=8723
[    4.031515] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_config.bin
[    4.032132] bluetooth hci0: Direct firmware load for rtl_bt/rtl8723b_config.bin failed with error -2
[    4.032136] Bluetooth: hci0: rtl: loading rtl_bt/rtl8723b_fw.bin
[    4.160515] usb 1-6: rtl8723bu_parse_efuse: dumping efuse (0x200 bytes):
[    4.160667] usb 1-6: RTL8723BU rev E (SMIC) 1T1R, TX queues 3, WiFi=1, BT=1, GPS=0, HI PA=0
[    4.160670] usb 1-6: RTL8723BU MAC: 7c:c7:09:69:b6:3a
[    4.160674] usb 1-6: rtl8xxxu: Loading firmware rtlwifi/rtl8723bu_nic.bin
[    5.038493] 8723bu: loading out-of-tree module taints kernel.
[    5.039092] 8723bu: module verification failed: signature and/or required key missing - tainting kernel
[    5.042286] RTL871X: module init start
[    5.042289] RTL871X: rtl8723bu v4.3.6.11_12942.20141204_BTCOEX20140507-4E40
[    5.042290] RTL871X: rtl8723bu BT-Coex version = BTCOEX20140507-4E40
[    5.042331] usbcore: registered new interface driver rtl8723bu
[    5.042332] RTL871X: module init ret=0
[  188.690578] Modules linked in: rfcomm ccm cmac bnep 8723bu(OE) arc4 snd_hda_codec_hdmi cmdlinepart intel_spi_platform intel_spi rtl8xxxu spi_nor nls_iso8859_1 mtd rtsx_usb_ms mac80211 memstick btusb btrtl btbcm cfg80211 btintel uvcvideo bluetooth videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_core videodev media ecdh_generic intel_rapl x86_pkg_temp_thermal intel_powerclamp coretemp kvm_intel kvm irqbypass crct10dif_pclmul crc32_pclmul ghash_clmulni_intel pcbc aesni_intel aes_x86_64 crypto_simd glue_helper cryptd intel_cstate intel_rapl_perf input_leds serio_raw hid_multitouch snd_hda_codec_realtek snd_hda_codec_generic joydev snd_soc_rt5640 snd_hda_intel intel_pch_thermal snd_hda_codec mei_me mei lpc_ich snd_hda_core snd_soc_rl6231 processor_thermal_device snd_hwdep snd_soc_ssm4567 intel_soc_dts_iosf
Linux butterscotch 4.13.0-16-generic #19-Ubuntu SMP Wed Oct 11 18:35:14 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

Bus 001 Device 004: ID 0bda:b720 Realtek Semiconductor Corp. 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.10
  bDeviceClass          239 Miscellaneous Device
  bDeviceSubClass         2 ?
  bDeviceProtocol         1 Interface Association
  bMaxPacketSize0        64
  idVendor           0x0bda Realtek Semiconductor Corp.
  idProduct          0xb720 
  bcdDevice            2.00
  iManufacturer           1 
  iProduct                2 
  iSerial                 3 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength          236
    bNumInterfaces          3
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower              500mA
    Interface Association:
      bLength                 8
      bDescriptorType        11
      bFirstInterface         0
      bInterfaceCount         2
      bFunctionClass        224 Wireless
      bFunctionSubClass       1 Radio Frequency
      bFunctionProtocol       1 Bluetooth
      iFunction               4 
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass       224 Wireless
      bInterfaceSubClass      1 Radio Frequency
      bInterfaceProtocol      1 Bluetooth
      iInterface              4 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
[... 253 lines cut ...]
          Usage Type               Data
        wMaxPacketSize     0x0040  1x 64 bytes
        bInterval               3
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x08  EP 8 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x09  EP 9 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               0

```

lwfinger, 2017-10-29:
This is not the correct place to report problems with rtl8xxxu. Send your complaints on that driver to linux-wireless@vger.kernel.org so that the maintainer of that driver will see it.

Note that even if you were using this driver, there is little I can do to address issues of speed. I know nothing of the internals of the device!!!!!!!!!

closed 2017-10-29 by lwfinger

hellslinger, 2017-10-30:
Understood. Thank you for responding and pointing me in the right direction! Also thank you for your great work on this project.