lwfinger/rtl8723bu #91: Fails to associate with AP after resume

dmig, 2017-11-11:
First noticed this problem after kernel update 4.13.10 -> 4.13.11, also reproduced on 4.13.12.
After resuming networkmanager shows available networks, but can't connect to any. No wifi problems before suspend.

Log for kernel 4.13.12:
```
Nov 11 15:15:48 dmig-tablet kernel: [14954.123849] PM: Syncing filesystems ... done.
Nov 11 15:15:48 dmig-tablet kernel: [14954.165874] PM: Preparing system for sleep (mem)
Nov 11 15:15:48 dmig-tablet kernel: [14954.166974] Freezing user space processes ... (elapsed 0.003 seconds) done.
Nov 11 15:15:48 dmig-tablet kernel: [14954.170526] OOM killer disabled.
Nov 11 15:15:48 dmig-tablet kernel: [14954.170528] Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
Nov 11 15:15:48 dmig-tablet kernel: [14954.172213] PM: Suspending system (mem)
Nov 11 15:15:48 dmig-tablet kernel: [14954.172271] Suspending console(s) (use no_console_suspend to debug)
Nov 11 15:15:48 dmig-tablet kernel: [14954.174118] sd 3:0:0:0: [sda] Synchronizing SCSI cache
Nov 11 15:15:48 dmig-tablet kernel: [14954.174159] sd 3:0:0:0: [sda] Stopping disk
........................................ driver problem? ......................................
Nov 11 15:15:48 dmig-tablet kernel: [14955.829644] RTL871X: ERROR rtw_cmd_thread(wlx7cc70926347e) stop_req:1, break
.........................................................................................................
Nov 11 15:15:48 dmig-tablet kernel: [14956.041437] PM: suspend of devices complete after 1868.690 msecs
Nov 11 15:15:48 dmig-tablet kernel: [14956.061014] PM: late suspend of devices complete after 19.568 msecs
Nov 11 15:15:48 dmig-tablet kernel: [14956.081044] PM: noirq suspend of devices complete after 20.020 msecs
Nov 11 15:15:48 dmig-tablet kernel: [14956.082035] ACPI: Preparing to enter system sleep state S3
Nov 11 15:15:48 dmig-tablet kernel: [14956.121794] ACPI: EC: event blocked
Nov 11 15:15:48 dmig-tablet kernel: [14956.121796] ACPI: EC: EC stopped
Nov 11 15:15:48 dmig-tablet kernel: [14956.121797] PM: Saving platform NVS memory
Nov 11 15:15:48 dmig-tablet kernel: [14956.121972] Disabling non-boot CPUs ...
Nov 11 15:15:48 dmig-tablet kernel: [14956.137746] IRQ 43: no longer affine to CPU1
Nov 11 15:15:48 dmig-tablet kernel: [14956.138773] smpboot: CPU 1 is now offline
Nov 11 15:15:48 dmig-tablet kernel: [14956.157671] IRQ 42: no longer affine to CPU2
Nov 11 15:15:48 dmig-tablet kernel: [14956.159698] smpboot: CPU 2 is now offline
Nov 11 15:15:48 dmig-tablet kernel: [14956.181511] IRQ 1: no longer affine to CPU3
Nov 11 15:15:48 dmig-tablet kernel: [14956.181517] IRQ 6: no longer affine to CPU3
Nov 11 15:15:48 dmig-tablet kernel: [14956.181521] IRQ 7: no longer affine to CPU3
Nov 11 15:15:48 dmig-tablet kernel: [14956.181526] IRQ 8: no longer affine to CPU3
Nov 11 15:15:48 dmig-tablet kernel: [14956.181530] IRQ 9: no longer affine to CPU3
Nov 11 15:15:48 dmig-tablet kernel: [14956.181536] IRQ 27: no longer affine to CPU3
Nov 11 15:15:48 dmig-tablet kernel: [14956.181542] IRQ 33: no longer affine to CPU3
Nov 11 15:15:48 dmig-tablet kernel: [14956.181555] IRQ 44: no longer affine to CPU3
Nov 11 15:15:48 dmig-tablet kernel: [14956.182569] smpboot: CPU 3 is now offline
Nov 11 15:15:48 dmig-tablet kernel: [14956.188239] ACPI: Low-level resume complete
Nov 11 15:15:48 dmig-tablet kernel: [14956.188378] ACPI: EC: EC started
Nov 11 15:15:48 dmig-tablet kernel: [14956.188380] PM: Restoring platform NVS memory
Nov 11 15:15:48 dmig-tablet kernel: [14956.189020] Suspended for 160.291 seconds
Nov 11 15:15:48 dmig-tablet kernel: [14956.200766] Enabling non-boot CPUs ...
Nov 11 15:15:48 dmig-tablet kernel: [14956.200852] x86: Booting SMP configuration:
Nov 11 15:15:48 dmig-tablet kernel: [14956.200856] smpboot: Booting Node 0 Processor 1 APIC 0x2
Nov 11 15:15:48 dmig-tablet kernel: [14956.205550]  cache: parent cpu1 should not be sleeping
Nov 11 15:15:48 dmig-tablet kernel: [14956.206125] CPU1 is up
Nov 11 15:15:48 dmig-tablet kernel: [14956.206200] smpboot: Booting Node 0 Processor 2 APIC 0x1
Nov 11 15:15:48 dmig-tablet kernel: [14956.210378]  cache: parent cpu2 should not be sleeping
Nov 11 15:15:48 dmig-tablet kernel: [14956.210818] CPU2 is up
Nov 11 15:15:48 dmig-tablet kernel: [14956.210880] smpboot: Booting Node 0 Processor 3 APIC 0x3
Nov 11 15:15:48 dmig-tablet kernel: [14956.214837]  cache: parent cpu3 should not be sleeping
Nov 11 15:15:48 dmig-tablet kernel: [14956.215494] CPU3 is up
Nov 11 15:15:48 dmig-tablet kernel: [14956.223968] ACPI: Waking up from system sleep state S3
Nov 11 15:15:48 dmig-tablet kernel: [14957.303503] PM: noirq resume of devices complete after 18.509 msecs
Nov 11 15:15:48 dmig-tablet kernel: [14957.859390] PM: early resume of devices complete after 555.812 msecs
Nov 11 15:15:48 dmig-tablet kernel: [14957.859714] ACPI: EC: event unblocked
Nov 11 15:15:48 dmig-tablet kernel: [14957.860051] ACPI: button: The lid device is not compliant to SW_LID.
Nov 11 15:15:48 dmig-tablet kernel: [14957.861037] rtc_cmos 00:01: Alarms can be up to one month in the future
Nov 11 15:15:48 dmig-tablet kernel: [14957.862386] sd 3:0:0:0: [sda] Starting disk
Nov 11 15:15:48 dmig-tablet kernel: [14957.924008] xhci_hcd 0000:00:14.0: port 7 resume PLC timeout
Nov 11 15:15:48 dmig-tablet kernel: [14957.940864] xhci_hcd 0000:00:14.0: port 4 resume PLC timeout
Nov 11 15:15:48 dmig-tablet kernel: [14957.957876] xhci_hcd 0000:00:14.0: port 2 resume PLC timeout
Nov 11 15:15:48 dmig-tablet kernel: [14958.175247] ata4: SATA link up 6.0 Gbps (SStatus 133 SControl 300)
Nov 11 15:15:48 dmig-tablet kernel: [14958.177323] ata4.00: configured for UDMA/133
Nov 11 15:15:48 dmig-tablet kernel: [14958.195369] usb 1-6: reset high-speed USB device number 4 using xhci_hcd
Nov 11 15:15:48 dmig-tablet kernel: [14958.455416] usb 1-5: reset high-speed USB device number 3 using xhci_hcd
Nov 11 15:15:48 dmig-tablet kernel: [14958.715555] usb 1-3: reset high-speed USB device number 2 using xhci_hcd
Nov 11 15:15:48 dmig-tablet kernel: [14958.975358] usb 1-8: reset high-speed USB device number 6 using xhci_hcd
Nov 11 15:15:48 dmig-tablet kernel: [14959.793919] PM: resume of devices complete after 1934.412 msecs
[... 71 lines cut ...]
Nov 11 15:16:03 dmig-tablet gnome-software[9368]: no app for changed refresh-wifi@kgshank.net
Nov 11 15:16:03 dmig-tablet gnome-shell[9030]: loading default theme (Adwaita)
Nov 11 15:16:03 dmig-tablet gnome-software[9368]: no app for changed user-theme@gnome-shell-extensions.gcampax.github.com
Nov 11 15:16:03 dmig-tablet gnome-software[9368]: no app for changed hibernate-status@dromi
Nov 11 15:16:03 dmig-tablet gnome-software[9368]: no app for changed removeaccesibility@lomegor
Nov 11 15:16:03 dmig-tablet gnome-software[9368]: no app for changed windowsNavigator@gnome-shell-extensions.gcampax.github.com
Nov 11 15:16:03 dmig-tablet gnome-software[9368]: no app for changed location-icon-autohide@darednaxella.name
Nov 11 15:16:03 dmig-tablet gnome-shell[9030]: [pixel-saver]: 'appmenu' is not a valid button.
Nov 11 15:16:03 dmig-tablet gnome-software[9368]: no app for changed pixel-saver@deadalnix.me
Nov 11 15:16:03 dmig-tablet gnome-software[9368]: no app for changed clipboard-indicator@tudmotu.com
Nov 11 15:16:03 dmig-tablet org.gnome.Shell.desktop[9030]: BingWallpaper extension: next check in 60 seconds @ local time 2017-11-11 15:17:03
Nov 11 15:16:03 dmig-tablet gnome-software[9368]: no app for changed BingWallpaper@ineffable-gmail.com
Nov 11 15:16:03 dmig-tablet gnome-software[9368]: no app for changed ubuntu-appindicators@ubuntu.com
Nov 11 15:16:04 dmig-tablet wpa_supplicant[1058]: wlx7cc70926347e: Trying to associate with c8:3a:35:1e:c2:60 (SSID='TheBase_5c' freq=2412 MHz)
Nov 11 15:16:05 dmig-tablet wpa_supplicant[1058]: wlx7cc70926347e: Association request to the driver failed
Nov 11 15:16:05 dmig-tablet NetworkManager[879]: <info>  [1510388165.8975] device (wlx7cc70926347e): supplicant interface state: scanning -> disconnected
Nov 11 15:16:06 dmig-tablet NetworkManager[879]: <info>  [1510388166.9037] device (wlx7cc70926347e): supplicant interface state: disconnected -> scanning
Nov 11 15:16:08 dmig-tablet wpa_supplicant[1058]: wlx7cc70926347e: Trying to associate with c8:3a:35:1e:c2:60 (SSID='TheBase_5c' freq=2412 MHz)
Nov 11 15:16:09 dmig-tablet wpa_supplicant[1058]: wlx7cc70926347e: Association request to the driver failed
Nov 11 15:16:09 dmig-tablet wpa_supplicant[1058]: wlx7cc70926347e: CTRL-EVENT-SSID-TEMP-DISABLED id=0 ssid="TheBase_5c" auth_failures=1 duration=10 reason=CONN_FAILED
Nov 11 15:16:09 dmig-tablet NetworkManager[879]: <info>  [1510388169.8627] device (wlx7cc70926347e): supplicant interface state: scanning -> disconnected
Nov 11 15:16:15 dmig-tablet NetworkManager[879]: <info>  [1510388175.6345] device (wlx7cc70926347e): supplicant interface state: disconnected -> scanning
...
```
Log for kernel 4.13.10 shows same error messages, but no problems with wifi connection after resume.

lwfinger, 2017-11-12:
As you are the one with the problem, you need to bisect between 4.13.11 and 4.13.10 to find the bad commit.

dmig, 2017-11-13:
Took a quick look over changelog before starting bisect, there is only one commit related to wireless or usb: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git/commit/?id=f2aa694b7459f20d37199651bc2a4495a5559502

Maybe this will give you some clue

dmig, 2017-11-27:
Same regression between stock Ubuntu kernels 4.13.0-17-generic and 4.13.0-18-generic, changelog shows, that later one is based on 4.13.13 mainline kernel.

lwfinger, 2017-11-27:
I have the same answer as the one I stated above: You need to bisect the problem between those two Ubuntu kernels. I cannot do that as your machine is the one with the problem.

dmig, 2017-11-27:
Unfortunately I'm unable to build kernels right now -- I don't have enough space.

ruoso, 2017-12-31:
I can confirm the same behavior on 4.14.0, the good news is that doing

```
#  rmmod 8723bu
#  modprobe rtl8723bu
```

After waking up makes it go again.

lwfinger, 2017-12-31:
Then investigate adding the appropriate scripts to unload on going to sleep and reloading on waking up.

ruoso, 2017-12-31:
Yes, I should have said that I also did that, but adding systemd sleep
hooks worked perfectly, and wifi is now fully functional.

dpriedel, 2018-01-13:
i have this same problem on my Alpha litebook running Manjaro.
  
Could you please explain what you did to solve this problem?

Thannks !!

carlocaione, 2018-01-23:
Yeah, hitting the same problem here also with on 4.14.12.

carlocaione, 2018-01-23:
@lwfinger output of bisection:
```
f2aa694b7459f20d37199651bc2a4495a5559502 is the first bad commit                                                                                                                                                                              
commit f2aa694b7459f20d37199651bc2a4495a5559502                                                                                                                                                                                               
Author: Johannes Berg <johannes.berg@intel.com>                                                                                                                                                                                               
Date:   Tue Oct 17 21:56:20 2017 +0200

    cfg80211: fix connect/disconnect edge cases
    
    commit 51e13359cd5ea34acc62c90627603352956380af upstream.
    
    If we try to connect while already connected/connecting, but
    this fails, we set ssid_len=0 but leave current_bss hanging,
    leading to errors.
    
    Check all of this better, first of all ensuring that we can't
    try to connect to a different SSID while connected/ing; ensure
    that prev_bssid is set for re-association attempts even in the
    case of the driver supporting the connect() method, and don't
    reset ssid_len in the failure cases.
    
    While at it, also reset ssid_len while disconnecting unless we
    were connected and expect a disconnected event, and warn on a
    successful connection without ssid_len being set.
    
    Signed-off-by: Johannes Berg <johannes.berg@intel.com>
    Signed-off-by: Greg Kroah-Hartman <gregkh@linuxfoundation.org>

:040000 040000 f7f2d9fdd7e3a8ee7ffc82047957be9e519f8eb8 f686d94cb8f9a0401bab281225b4e9dda6e7c815 M      net
```

carlocaione, 2018-01-23:
Some info about this (just a braindump).

The problem seems to be originated in https://github.com/torvalds/linux/blob/51e13359cd5ea34acc62c90627603352956380af/net/wireless/sme.c#L1077 since `prev_bssid` is not being set by https://github.com/torvalds/linux/blob/51e13359cd5ea34acc62c90627603352956380af/net/wireless/nl80211.c#L8942

karora, 2018-01-24:
So is info->attrs[NL80211_ATTR_PREV_BSSID] supposed to be set by the driver during suspend?

carlocaione, 2018-01-25:
@karora no, that's set by the userspace when using `nl80211` on reconnection. I checked `wpa_supplicant` and it appears that it should set it correctly (see https://w1.fi/cgit/hostap/plain/wpa_supplicant/ChangeLog). I'm still looking into it.

Still somehow this is mismanaged by the driver causing this problem.

carlocaione, 2018-01-25:
Just to update this on resume `NL80211_ATTR_PREV_BSSID` is not set by `wpa_supplicant`. This is only triggering the problem but probably it's correct and not a problem in itself.

carlocaione, 2018-02-01:
Just opened a PR for this issue

dpriedel, 2018-02-02:
I installed the updated driver on my Alpha Litebook running kernel 4.15.  All is well so far.

Thanks !!!

pcav, 2019-01-08:
Sorry, unclear to me: is this fixed now?
Thanks

cropinghigh, 2021-09-27:
Having same issue on Manjaro 5.13 now

lwfinger, 2021-09-27:
It works on my system, thus there is a problem with your user-space components that control network connections.

Do some research on how to set up a script that will unload the driver on hibernate or suspend, and reload it on resume.

cropinghigh, 2021-09-28:
After some debugging of wpa_supplicant, found that it's error is "nl80211: MLME connect failed: ret=-114(Operation already in progress)", returned, as i can see, directly from driver, but can't find where

cropinghigh, 2021-09-29:
This code is returned from kernel/net/wireless/sme.c, line 1200
"If connected, reject (re-)association unless prev_bssid matches the current BSSID."