lwfinger/rtl8723bu #92: Can not connect to Wi-Fi

bxwllzz, 2017-12-22:


lwfinger, 2017-12-22:
What make/model of access point are you using? Does it have the latest firmware?

Please post the output of 'sudo iw phy'.

bxwllzz, 2017-12-22:
`sudo iw phy` on Ubuntu 16.04.
```
Wiphy phy2
	max # scan SSIDs: 9
	max scan IEs length: 2304 bytes
	Retry short limit: 7
	Retry long limit: 4
	Coverage class: 0 (up to 0m)
	Supported Ciphers:
		* WEP40 (00-0f-ac:1)
		* WEP104 (00-0f-ac:5)
		* TKIP (00-0f-ac:2)
		* CCMP (00-0f-ac:4)
	Available Antennas: TX 0 RX 0
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x1862
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			No RX STBC
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
		Maximum RX AMPDU length 65535 bytes (exponent: 0x003)
		Minimum RX AMPDU time spacing: 16 usec (0x07)
		HT TX/RX MCS rate indexes supported: 0-7, 32
		Bitrates (non-HT):
			* 1.0 Mbps
			* 2.0 Mbps
			* 5.5 Mbps
			* 11.0 Mbps
			* 6.0 Mbps
			* 9.0 Mbps
			* 12.0 Mbps
			* 18.0 Mbps
			* 24.0 Mbps
			* 36.0 Mbps
			* 48.0 Mbps
			* 54.0 Mbps
		Frequencies:
			* 2412 MHz [1] (20.0 dBm)
			* 2417 MHz [2] (20.0 dBm)
			* 2422 MHz [3] (20.0 dBm)
			* 2427 MHz [4] (20.0 dBm)
			* 2432 MHz [5] (20.0 dBm)
			* 2437 MHz [6] (20.0 dBm)
			* 2442 MHz [7] (20.0 dBm)
			* 2447 MHz [8] (20.0 dBm)
			* 2452 MHz [9] (20.0 dBm)
			* 2457 MHz [10] (20.0 dBm)
			* 2462 MHz [11] (20.0 dBm)
			* 2467 MHz [12] (20.0 dBm)
			* 2472 MHz [13] (20.0 dBm) (no IR)
			* 2484 MHz [14] (disabled)
	Band 2:
		Capabilities: 0x1862
			HT20/HT40
			Static SM Power Save
			RX HT20 SGI
			RX HT40 SGI
			No RX STBC
			Max AMSDU length: 7935 bytes
			DSSS/CCK HT40
[... 216 lines cut ...]
		 * frame
		 * set_channel
		 * connect
		 * disconnect
	Supported TX frame types:
		 * IBSS: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * managed: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * AP/VLAN: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-client: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
		 * P2P-GO: 0x00 0x10 0x20 0x30 0x40 0x50 0x60 0x70 0x80 0x90 0xa0 0xb0 0xc0 0xd0 0xe0 0xf0
	Supported RX frame types:
		 * IBSS: 0xd0
		 * managed: 0x40 0xd0
		 * AP: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * AP/VLAN: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
		 * P2P-client: 0x40 0xd0
		 * P2P-GO: 0x00 0x20 0x40 0xa0 0xb0 0xc0 0xd0
	WoWLAN support:
		 * wake up on anything (device continues operating normally)
	software interface modes (can always be added):
		 * monitor
	interface combinations are not supported
	Device supports scan flush.
```

bxwllzz, 2017-12-22:
# More Test
- `B Only` is OK. 11 Mb/s
- `G Only` is OK. 54 Mb/s
- `N Only` fails. 
dmesg:
```
...
[ 2257.911267] RTL871X: start auth
[ 2257.913165] RTL871X: auth success, start assoc
[ 2261.614071] RTL871X: nolinked power save enter
[ 2264.697080] RTL871X: RTW_ADAPTIVITY_EN_AUTO, chplan:0x20, Regulation:3,3
[ 2264.697082] RTL871X: RTW_ADAPTIVITY_MODE_NORMAL
[ 2265.219571] RTL871X: nolinked power save leave
[ 2269.428722] RTL871X: rtw_set_802_11_connect(wlx40a5ef430adc)  fw_state=0x00000008
[ 2269.685924] RTL871X: start auth
...
```
- `B/G mixed` is OK. 54 Mb/s
- `G/N mixed` is OK. 72.2 Mb/s
- `B/G/N mixed` is OK. 72.2 Mb/s
# AP Settings
[image]
[image]

maurerr, 2019-03-02:
the same issue in my case -  tp-link mr200 4g router with padavan-ng firmware (ralink/mediatek wisoc) .
switched from N-only to NG-mixed fixed the issue

BrunoKlu, 2026-10-23:
I hit what looks like the same problem (RTL8723BU, association timeout) and
tracked it down with an over-the-air capture.

Setup: RTL8723BU in a Xiegu X6100 radio, this driver (v4.3.6.11_12942.20141204),
wpa_supplicant 2.10. APs: GL.iNet GL-MT3000 and GL-MT3600BE (both MediaTek).

What happens: auth succeeds, then every Association Request is ACKed by the AP
but never answered. The driver retries a few times and reports
CTRL-EVENT-ASSOC-REJECT status_code=1, which it generates locally.

Cause: in all Association Requests sent by this driver, the HT Capabilities
element (ID 45) is 25 octets long instead of 26. In include/wifi.h:

    struct rtw_ieee80211_ht_cap {
        __le16 cap_info;
        unsigned char ampdu_params_info;
        unsigned char supp_mcs_set[16];
        __le16 extended_ht_cap_info;
        __le16 tx_BF_cap_info;          /* should be 32 bits */
        __le16 antenna_selection_info;  /* should be 8 bits  */
    } __attribute__ ((packed));         /* 25 bytes, standard is 26 */

rtw_restructure_ht_ie() writes the IE with sizeof() of this struct. Newer
Realtek drivers (e.g. rtl8812au 5.6.4.2, 8821cu) use
`unsigned int tx_BF_cap_info; unsigned char antenna_selection_info;`.

MediaTek APs drop the malformed frame silently. Qualcomm-based APs and an ISP
box tolerate it. With the in-kernel rtl8xxxu driver (26-octet HT Capabilities),
the same radio associates with the same AP in 27 ms. An AP set to N-only, as
in this issue, probably checks the HT element strictly in the same way.

Untested fix: declare those two fields as __le32 and u8, after checking the
SET_HT_CAP_TXBF_* macros that write into them.